tsduck/tsduck #615: Tsswitch - didn't switch to backup stream

mks75, 2020-07-09:
Hi,
I am using tsduck version 3.22-1866 and observed some issues with tsswitch, that I am using as failover switch between two MPEG-TS streams.
On the loss of primary stream, it didn't switch to backup stream. Although the tsswitch process was running well, and backup stream was also present. Initially everything was working well when I started this process couple of days back, but recently it didn't work - not sure what happened and when it stopped working? Only after restarting the process it worked.

Any advice please?

Thanks
MKS

lelegard, 2020-07-10:
What is your command? Specifically, do you use  options `--delayed-switch` and `--receive-timeout` at the same time?

There is a long standing interaction bug between the two (see #324). This is a bit tricky to solve and I did not find the time yet to fix it.

mks75, 2020-07-10:
Hi, I am using fast switch option as my streams are LIVE. The command that I am using is
/tsswitch -i -f --receive-timeout 1000 -p 0 -I ip 239.1.1.1:1234 -I ip 239.2.2.2:1234 -O ip 239.3.3.3:1234

Thanks - MKS