tsduck/tsduck #616: MPE: Can you suppress the filling of the last fragment with 0xFF?

VDelport, 2020-07-21:
I have a question about MPE from UDP segments in a transport stream. I embed the UDP segments in the transport stream via netcat. When embedding, a UDP segment is divided in to fragments of 184 bytes. If the last bytes of the UDP payload are less than 184 bytes, a corresponding number of 0xFFs are appended to the last fragment (see Wireshark screenshot)

[image]

Is there an option that prevents this filling with 0xFF?

lelegard, 2020-07-23:
These FF are not specific to MPE. They are artifacts of section packetization. 

UDP datagrams are encapsulated into DSM-CC sections. Those sections are packetized into TS packets. Those TS packets have a fixed size, hence the padding to fill the last TS packet of a given DSM-CC section.

VDelport, 2020-07-23:
Thank you for your response. What do you mean by "artifacts of section packetization"? So is it up to netcat? If so, which tool is suitable?

lelegard, 2020-07-23:
A transport stream is a suite of TS packets. Each TS packet is 188 bytes long. This is the standard. No tool will change this. 

The UDP datagram is inside a section. The section is split over several TS packets using the 184-byte payload of each TS packet (except in case of adaptation field - rare in section PIDs - or pointer field - as in the first packet). The UDP datagram (green in your screenshot) is followed by 4 bytes of CRC32 at the end of the DSM-CC section. At this point, we are in the middle of a TS packet. Then what? You cannot truncate a TS packet. A TS packet is _always_ 188 bytes long. You need to fill the rest of it. So you can either start a new section right after the previous one (typically not the case with DSM-CC) or you fill with FF, indicating that this is just stuffing. 

Why would you want to remove those FF? You can't, simply. This is not a matter of tool. You would get the same result with the `mpeinject` plugin of TSDuck.

VDelport, 2020-07-24:
Thank you for your detailed answer and your patience. Of course, you are absolutely right. You can't just remove the filling. The last TS packet of the UDP segment must of course also have the full 184 byte payload, and filling up shouldn't be a problem.

Just to answer your question:
My problem is that I hand over the UDP blocks of a software from a third party. This software obviously needs the blocks in such a way that they are embedded in the TS packets as efficiently as possible by combining the data of the last fragment of a UDP segment and the data of the first fragment of the subsequent UDP segment into one TS packet . The whole thing is organized by a 1-byte pointer that is inserted after the header or that extends the header. In the example you see two consecutive UDP segments.

UDP segment 1:
[image]

UDP segment 2:
[image]

In the TS packet for the last fragment of UDP segment 1, only 34 bytes are used as payload. The remaining 184-1-34 = 149 bytes are already used for the first fragment of UDP segment 2.

I assume that I have to implement such packaging myself, or do you know a tool that can do this?

lelegard, 2020-07-24:
Ok, so you want section packing with MPE. This could be possible through an enhancement of the `mpeinject` plugin. I will have a look at it. 

However, this will be a specific option, not the default. We may anticipate that some MPE extractors may have a problem with that (although this is a valid section packetization). 

Usually, section packing is used with heterogeneous streams of short sections such as EMM's, rarely with cycled signalisation or DSM-CC carousels. But, although unusual, this is valid.

lelegard, 2020-08-12:
I have added the option `--pack-sections` to plugin `mpeinject` so that DSM-CC sections are packed, potentially starting in the middle of a TS packet, right after the previous section. Be sure to test that your MPE receiver does not have any issue with this.

closed 2022-09-23 by lelegard