tsduck/tsduck #617: Compiling TSDuck on Debian Jessie fails (chnge suggested)

bertothunder, 2020-07-23:
TSDuck is not available for debian jessie (stable), only for testing (bullseye).

Trying to build it on Debian Jessie (with SRT available), using g++ 4.9.x as default compiler, it fails building it up because of a warning converted to error:

```
make NOPCSC=1 NODTAPI=1 NOTEST=1 NOTELETEXT=1 

  [CXX] dtv/descriptors/tsAACDescriptor.cpp
In file included from /root/tsduck-master/src/libtsduck/dtv/tsStandards.h:37:0,
                 from /root/tsduck-master/src/libtsduck/dtv/tsAbstractDefinedByStandards.h:36,
                 from /root/tsduck-master/src/libtsduck/dtv/tsAbstractSignalization.h:36,
                 from dtv/descriptors/tsAbstractDescriptor.h:36,
                 from dtv/descriptors/tsAACDescriptor.h:36,
                 from dtv/descriptors/tsAACDescriptor.cpp:30:
/root/tsduck-master/src/libtsduck/base/tsEnumUtils.h:73:16: error: specialization of 'template<class T> struct ts::EnableBitMaskOperators' in different namespace [-fpermissive]
     struct ts::EnableBitMaskOperators<T>     \
                ^
/root/tsduck-master/src/libtsduck/dtv/tsStandards.h:61:1: note: in expansion of macro 'TS_ENABLE_BITMASK_OPERATORS'
 TS_ENABLE_BITMASK_OPERATORS(ts::Standards);
 ^
/root/tsduck-master/src/libtsduck/base/tsEnumUtils.h:49:12: error:   from definition of 'template<class T> struct ts::EnableBitMaskOperators' [-fpermissive]
     struct EnableBitMaskOperators
            ^
/root/tsduck-master/src/libtsduck/base/tsEnumUtils.h:73:16: error: specialization of 'template<class T> struct ts::EnableBitMaskOperators' in different namespace [-fpermissive]
     struct ts::EnableBitMaskOperators<T>     \
                ^
/root/tsduck-master/src/libtsduck/dtv/tsNames.h:47:1: note: in expansion of macro 'TS_ENABLE_BITMASK_OPERATORS'
 TS_ENABLE_BITMASK_OPERATORS(ts::names::Flags);
 ^
/root/tsduck-master/src/libtsduck/base/tsEnumUtils.h:49:12: error:   from definition of 'template<class T> struct ts::EnableBitMaskOperators' [-fpermissive]
     struct EnableBitMaskOperators
            ^
../../Makefile.common:958: recipe for target '/root/tsduck-master/bin/release-x86_64-266ed4a5eaa8-static/objs-libtsduck/tsAACDescriptor.o' failed
make[2]: *** [/root/tsduck-master/bin/release-x86_64-266ed4a5eaa8-static/objs-libtsduck/tsAACDescriptor.o] Error 1
Makefile:43: recipe for target 'default' failed
make[1]: *** [default] Error 2
Makefile:59: recipe for target 'default' failed
make: *** [default] Error 2
```

To fix it, just edit `src/libtsduck/base/tsEnumUtils.h` and do change the definition for `TS_ENABLE_BITMASK_OPERATORS(T)`:
```
//!
//! @hideinitializer
//! This macro enables bitmask operators on an enumeration type.
//! @param T An enumeration type or enumeration class.
//!
//! Example:
//! @code
//! enum E {A = 0x01, B = 0x02, C = 0x04};
//! TS_ENABLE_BITMASK_OPERATORS(E)
//!
//! E e = A | B | C;
//! e ^= B | C;
//! @endcode
//!
#define TS_ENABLE_BITMASK_OPERATORS(T)       \
namespace ts {                               \
    /** Template specialization on type T */ \
    template<>                               \
    struct EnableBitMaskOperators<T>         \
    {                                        \
        /** Enable bitmask operators on T */ \
        static constexpr bool value = true;  \
    };                                       \
}
```

This way, it works and compiles with no problem.

lelegard, 2020-07-23:
Thanks for reporting this. I am currently away and cannot fix this right now but I will when I get back.

lelegard, 2020-08-10:
This is now fixed in commit 465051103229adee3661b0c7b3a17ee52dfd2041

Note that Debian Jessie is 5 years old and GCC 4.x is no longer used to test TSDuck. All recent distros use GCC version 8.x to 10.x. So, I cannot guarantee that such mishaps won't happen again with old compilers.

bertothunder, 2020-10-30:
Thanks @lelegard. I can confirm 3.23-1965 compiles nicely on Debian Jessie.

closed 2020-10-30 by bertothunder