tsduck/tsduck #619: tsecmg does not appear to handle rollover of CP_number mid CW_provision

djsdevel, 2020-07-24:
If multiple control words are used for the tsecmg and the CP_number rolls over from 65535 to 0 in one CW_provision message a stream_error gets sent back - see the following debug:

* Debug: received message from 192.168.3.248:51994
    CW_provision (ECMG<=>SCS)
    protocol_version = 0x02
    message_type = 0x0201
    ECM_channel_id = 0x0000
    ECM_stream_id = 0x0001
    CP_number = 65532
    access_criteria (6 bytes) =
        AA AA BB BB CC CC
    CP = 65532
    CW (8 bytes) = 90 4C 25 01 BF 8B 4E 98
    CP = 65533
    CW (8 bytes) = 07 0A C1 D2 0F 50 B5 14
    CP = 65534
    CW (8 bytes) = 3A 8B 41 06 33 70 F6 99
    CP = 65535
    CW (8 bytes) = 79 18 B9 4A 8C 81 71 7E
    CP = 0
    CW (8 bytes) = 3D E0 45 62 BE F5 9A 4D

* Debug: sending message to 192.168.3.248:51994
    stream_error (ECMG<=>SCS)
    protocol_version = 0x02
    message_type = 0x0106
    ECM_channel_id = 0x0000
    ECM_stream_id = 0x0001
    error_status = 0x000B (not enough control words to compute ECM)

Brilliant work though I've been extensively using tsecmg and tsemmg today.

djsdevel, 2020-07-24:
I've seen this issue in 3.19 and just re-tested in 3.21-1819 and it still occurs.

lelegard, 2020-07-24:
Thank you for reporting this. I am away for a couple of weeks and will have a look at it when I am back.

lelegard, 2020-08-10:
Fixed in commit b1b304f5b7702ab7fd1cc020ea284da61293b863, with a non-regression test [here](https://github.com/tsduck/tsduck-test/blob/master/tests/test-069.sh).

djsdevel, 2020-08-11:
Thanks for the update and resolving the issue

closed 2020-08-11 by djsdevel