tsduck/tsduck #620: tsduck_3.21-1819_amd64.deb - tsp: error while loading shared libraries: libsrt.so.1

themoog, 2020-07-26:
Hi I have the error;
 
tsp: error while loading shared libraries: libsrt.so.1: cannot open shared object file: No such file or directory

when installing on ubuntu.....clearly missing some sort of dependency  Any thoughts?

thanks in advance

themoog, 2020-07-26:
Ahh i solved that and then found a load of other dependencies - is there a guide to install tsduck in linux?

thanghl, 2020-07-27:
I have problem distro ubuntu how to fix it ??? Thanks a lot

trollcop, 2020-07-28:
apt-get install libsrt-dev
that's the only missing dependency if you install from -releases deb on ubuntu 20.04

lelegard, 2020-08-09:
Hi, sorry about that. This is a known issue that was fixed in commit 6acb734c6fedc564ec79da6bd8f7b7325bf095ba, on June 15 this year.

The actual dependency for `libsrt.so.1` is `libsrt1` not `libsrt-dev`. But since `libsrt1` is a dependency for `libsrt-dev`, installing the latter forces the installation of the former.

As of version 3.21, `libsrt1` was missing from the list of dependencies for package `tsduck`. This was fixed in the above mentioned commit. Version 3.22 will no longer have this issue.

gkeiser, 2020-08-26:
Do you have an anticpiated release date for 3.22?

lelegard, 2020-08-26:
> Do you have an anticipated release date for 3.22?

Not really, I have no product management team at home 😄 

But since I just completed a cycle of internal refactoring, I may consider release it one of these week-ends.

In the meantime, you may get an Ubuntu binary in the [pre-release / nightly builds area](https://tsduck.io/download/prerelease/). Please note that the nightly builds are automatically built on GitHub runner hosts which use Ubuntu 18.04. The official release will be built for Ubuntu 20.04 (latest one).

gkeiser, 2020-08-26:
Thanks! And thanks for this project.

lelegard, 2020-08-29:
Version 3.22 is out.

closed 2020-08-29 by lelegard

picatchoof, 2021-07-26:
I'm running Ubuntu 18.04 and try to install TSDuck 3.27 and get errors.

lelegard, 2021-07-31:
Hi @picatchoof, is this related to #827 or something else ?

andimik, 2022-12-31:
I have the same error under Ubuntu 22.04

It worked before

```
INFO:  Using pid: 0x03e8 (1000)
WARN:  Forward error correction (FEC) enabled
^C0+0 Datensätze ein
0+0 Datensätze aus
0 Bytes kopiert, 50,5089 s, 0,0 kB/s
* tsp: user interrupt, terminating...
* new TSDuck version 3.32-2983 is available (yours is 3.30-2710), use 'tsversion --upgrade' or see https://tsduck.io/
```
but then I got a message that there is a newer version out. Now each `tsp` results in an error.

```
$ tsversion --upgrade
tsversion: error while loading shared libraries: libsrt.so.1.5: cannot open shared object file: No such file or directory
```

lelegard, 2022-12-31:
What did you do exactly? You mentioned "it worked before". So, what did you do exactly to come to a state where it no longer works? 

What are the installed srt packages? On an Ubuntu 22.10, I see this:
~~~
$ dpkg -l '*srt*'
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                     Version      Architecture Description
+++-========================-============-============-=============================================================================
un  libsrt                   <none>       <none>       (no description available)
un  libsrt-dev               <none>       <none>       (no description available)
un  libsrt-doc               <none>       <none>       (no description available)
un  libsrt-gnutls-dev        <none>       <none>       (no description available)
ii  libsrt-openssl-dev:amd64 1.5.0-2      amd64        Secure Reliable Transport UDP streaming library (OpenSSL flavour development)
un  libsrt1                  <none>       <none>       (no description available)
un  libsrt1-openssl          <none>       <none>       (no description available)
ii  libsrt1.4-openssl:amd64  1.4.4-4      amd64        Secure Reliable Transport UDP streaming library (OpenSSL flavour)
ii  libsrt1.5-openssl:amd64  1.5.0-2      amd64        Secure Reliable Transport UDP streaming library (OpenSSL flavour)
~~~

andimik, 2023-01-02:
I have successfully downgraded to [Version 3.30-2710](https://github.com/tsduck/tsduck/releases/tag/v3.30-2710) again, which works under Ubuntu 22.04.

The newer version shows this error.

lelegard, 2023-01-02:
The prebuilt binaries are built on the latest version of each distro. I assume that 3.32 was built on Ubuntu 20.10 and that Ubuntu 20.10 contains SRT 1.5 while your Ubuntu 20.04 contained SRT 1.4, hence the missing dependency

This is an eternal problem with Linux distros. Nothing is compatible with anything that was built in a slightly different environment (see my comment to [this post](https://www.linkedin.com/posts/hackingarticles_infosec-cybersecurity-cybersecuritytips-activity-7010494195085815808-1YnV)). The prebuilt binaries are here to help in case you have the exact same OS version as used in the build. Otherwise, you need to rebuild yourself.

I would recommend to rebuild the latest version for you exact distro. This is [explained here](https://tsduck.io/doxy/building.html).

Or simply:
~~~
git clone https://github.com/tsduck/tsduck.git
cd tsduck
scripts/install-prerequisites.sh
make installer
~~~
And then collect the `.deb` packages in the `installers` subdirectory.

andimik, 2023-01-02:
No, even newer. Latest Ubuntu 22.04 LTS. 

I don't find that user-friendly because one would not expect such troubles resp. be warned before updating.

Any chance to build new tsduck versions only with Ubuntu LTS versions?

lelegard, 2023-01-02:
> No, even newer. Latest Ubuntu 22.04 LTS.

What do you mean by "no"? The latest Ubuntu is 22.10.

> I don't find that user-friendly because one would not expect such troubles resp. be warned before updating.

Being user-friendly would mean that `tsversion --upgrade` would know exactly how each distro works and what are the exact dependencies. This would be yet another package management tool. `tsversion --upgrade` is just a helper for people using the latest version of the few distros for which a TSDuck binary package is available.

> Any chance to build new tsduck versions only with Ubuntu LTS versions?

No, sorry. TSDuck is a free open-source software which is managed by one single person (me). I do not have the resources to build for all versions of all distros. The concept of LTS is Ubuntu only, each distro has its own support policy.

To give you an idea of the nightmare it would be, I have 10 different Linux virtual machines running 10 different distros. Each VM runs the latest version of the corresponding distro. Assuming I would support, for each distro, the latest version, the latest "long support" (or whatever exists for that distro), that would make 20 virtual machines. Resulting in 40 different binary packages (there are 2 packages per version, base and dev).

In practice, I generate the binaries for 4 distros (Ubuntu, Debian, Fedora, Redhat), latest version only.

Everybody using any other distro or version of distro has to rebuild. Each operating system implements its own version of hell. This is Linux hell...

If you want a user-friendly environment, use a Mac. My main machine is a Mac (where I run those 10 Linux VM, plus 2 Windows VM). A macOS system is a Unix system with bash, emacs, clang, sudo, wireshark, gimp, all usual free software. And if you want to install TSDuck, just type `brew install tsduck` and it works at once, for all versions of macOS, Intel or Arm CPU.

More seriously, I invest a lot of personal time in this project, for no return. If someone does not want to invest the time to rebuild in order to use that project for free, he would better move to some commercial product (or buy a Mac...)

TL713, 2023-01-21:
Hi Lelegard,

I met a similar error when installing `tsduck_3.32-2983.debian11_amd64.deb` on Debian. 
```
tsversion: error while loading shared libraries: libsrt.so.1.4: cannot open shared object file: No such file or directory
```
On my Debian, I can only install `libsrt1.5-openssl`, which provides `/usr/lib/x86_64-linux-gnu/libsrt.so.1.5` and `/usr/lib/x86_64-linux-gnu/libsrt.so.1.5.1`. Is there a TSDuck version I can use using `libsrt.so.1.5`? 

Thank you for help!

lelegard, 2023-01-21:
Hi @TL713 

You did not mention which Debian you use. On my standard Debian 11 bullseye with up-to-date packages, I have `libsrt1.4-openssl`. This was present when the TSDuck package was built, it is still present, and the package installs without problem.

If you have a slightly different version, as indicated in my previous post, you need to rebuild. Packages are rebuilt using `make installer`. See [here](https://tsduck.io/doxy/building.html) for details.

TL713, 2023-01-25:
Thx for help! It fixed by rebuilt.

null-coffee, 2023-02-24:
I'm just going to say two things:
First a big thank you for creating this software, I know that's no small or trivial task particularly when you get deep into the weeds of the DVB standards.

Second, I also hit this issue on Ubuntu LTS 22.04 but your steps (and scripts) for (re)building worked perfectly, much obliged

Saentist, 2023-06-18:
@lelegard 
is it possible `tsduck` to check somehow is `libsrt` exist before show error

```make
......./srt-1.5.0# make install
[ 67%] Built target srt_virtual
[ 70%] Built target srt_static
[ 85%] Built target srtsupport_virtual
[ 89%] Built target srt-tunnel
[ 92%] Built target srt-file-transmit
[ 96%] Built target srt-live-transmit
[100%] Built target srt_shared
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib/libsrt.so.1.5.0
-- Installing: /usr/local/lib/libsrt.so.1.5
-- Installing: /usr/local/lib/libsrt.so
-- Installing: /usr/local/lib/libsrt.a
-- Installing: /usr/local/include/srt/version.h
-- Installing: /usr/local/include/srt/srt.h
-- Installing: /usr/local/include/srt/logging_api.h
-- Installing: /usr/local/include/srt/access_control.h
-- Installing: /usr/local/include/srt/platform_sys.h
-- Installing: /usr/local/include/srt/udt.h
-- Installing: /usr/local/lib/pkgconfig/haisrt.pc
-- Installing: /usr/local/lib/pkgconfig/srt.pc
-- Installing: /usr/local/bin/srt-live-transmit
-- Up-to-date: /usr/local/bin/srt-live-transmit
-- Installing: /usr/local/bin/srt-file-transmit
-- Up-to-date: /usr/local/bin/srt-file-transmit
-- Installing: /usr/local/bin/srt-tunnel
-- Up-to-date: /usr/local/bin/srt-tunnel
-- Installing: /usr/local/bin/srt-ffplay
......./srt-1.5.0#
.......# tsversion --upgrade
tsversion: error while loading shared libraries: libsrt.so.1.5: cannot open shared object file: No such file or directory

```

I'm using Ubuntu 20.04 with come with libsrt 1.4, but after as seen above installed 1.5 result is same.

more strange is this dependency 

```
ldd /usr/bin/tsanalyze
/usr/bin/tsanalyze: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /usr/bin/tsanalyze)
/usr/bin/tsanalyze: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /lib/libtsduck.so)
/usr/bin/tsanalyze: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /lib/libtsduck.so)
/usr/bin/tsanalyze: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /lib/libtsduck.so)
/usr/bin/tsanalyze: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/libtsduck.so)
/usr/bin/tsanalyze: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /lib/libtsduck.so)
        linux-vdso.so.1 (0x00007ffd2e9e8000)
        libtsduck.so => /lib/libtsduck.so (0x00007f1979c99000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f1979ab7000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f1979968000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f197994d000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f197975b000)
        libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f197973f000)
        libpcsclite.so.1 => /lib/x86_64-linux-gnu/libpcsclite.so.1 (0x00007f1979730000)
        libedit.so.2 => /lib/x86_64-linux-gnu/libedit.so.2 (0x00007f19796f8000)
        libcurl.so.4 => /lib/x86_64-linux-gnu/libcurl.so.4 (0x00007f1979665000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f197b243000)
        libsrt.so.1.5 => not found
        libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f1979638000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1979613000)
        libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f19795e3000)
        libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f19795c9000)
        libnghttp2.so.14 => /lib/x86_64-linux-gnu/libnghttp2.so.14 (0x00007f19795a0000)
        libidn2.so.0 => /lib/x86_64-linux-gnu/libidn2.so.0 (0x00007f197957f000)
        librtmp.so.1 => /lib/x86_64-linux-gnu/librtmp.so.1 (0x00007f197955f000)
[... 13 lines cut ...]
        libgmp.so.10 => /lib/x86_64-linux-gnu/libgmp.so.10 (0x00007f1978c44000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1978c3e000)
        libkrb5.so.3 => /lib/x86_64-linux-gnu/libkrb5.so.3 (0x00007f1978b61000)
        libk5crypto.so.3 => /lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007f1978b2e000)
        libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007f1978b27000)
        libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007f1978b18000)
        libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f1978afc000)
        libsasl2.so.2 => /lib/x86_64-linux-gnu/libsasl2.so.2 (0x00007f1978adf000)
        libgssapi.so.3 => /lib/x86_64-linux-gnu/libgssapi.so.3 (0x00007f1978a9a000)
        libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x00007f1978a77000)
        libp11-kit.so.0 => /lib/x86_64-linux-gnu/libp11-kit.so.0 (0x00007f197893f000)
        libtasn1.so.6 => /lib/x86_64-linux-gnu/libtasn1.so.6 (0x00007f1978929000)
        libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007f1978922000)
        libheimntlm.so.0 => /lib/x86_64-linux-gnu/libheimntlm.so.0 (0x00007f1978916000)
        libkrb5.so.26 => /lib/x86_64-linux-gnu/libkrb5.so.26 (0x00007f1978883000)
        libasn1.so.8 => /lib/x86_64-linux-gnu/libasn1.so.8 (0x00007f19787db000)
        libhcrypto.so.4 => /lib/x86_64-linux-gnu/libhcrypto.so.4 (0x00007f19787a3000)
        libroken.so.18 => /lib/x86_64-linux-gnu/libroken.so.18 (0x00007f197878a000)
        libffi.so.7 => /lib/x86_64-linux-gnu/libffi.so.7 (0x00007f197877e000)
        libwind.so.0 => /lib/x86_64-linux-gnu/libwind.so.0 (0x00007f1978754000)
        libheimbase.so.1 => /lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007f1978740000)
        libhx509.so.5 => /lib/x86_64-linux-gnu/libhx509.so.5 (0x00007f19786f2000)
        libsqlite3.so.0 => /lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007f19785c9000)
        libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f197858e000)
```