tsduck/tsduck #621: Question: How to get the length in TS packets stored in a Cycling Packetizer?

charluy, 2020-07-27:
Hello everyone,

I been trying to implement an other multiplexing logic of PSI/SI tables for my project using TSDuck library. For that purpose i create a CyclingPacketizer without specifying the bitrate, and then getting the packet when is necessary. The only problem is that i need to know the length in TS packets stored in the CyclingPacketizer to send it in a constant manner, any ideas? 

So far i been counting the amount of TS packets between end of cycles but doesn't seems like a good idea.

Best regards.
Diego Sánchez

lelegard, 2020-08-10:
Hi @charluy,

There is no simple answer to that question. Merely, "it depends".

In the general case, a `CyclingPacketizer` contains a set of _scheduled_ sections (with a given repetition rate and a known PID bitrate) and a set of _unscheduled_ sections. The concept of "cycle" is then undefined in some cases when a mixture of both exists. In your case, you only have unscheduled sections only, but `CyclingPacketizer` has to manage the most general case.

Additionally, the section stuffing option has an impact on the number of packets.
- When it is specified as `ALWAYS`, then you can simply add the result of `packetCount()` on all sections before adding them.
- When the stuffing option is `AT_END`, the number of packets may be smaller. But it is hard to predict the number of packets since it depends on the size and order of sections, as well as the `headerSplitAllowed()` option. 
- When the stuffing option is `NEVER`, then a cycle may start in the same packet as the end of the previous cycle. So, there is not even a specific number of packets per cycle.

Depending on what your application wants to do, if you never change the content of the packetizer after filling it with sections and you want to have full control over the corresponding TS packets, then the class `OneShotPacketizer` may be better for you.

charluy, 2020-09-28:
Hello @lelegard,

I apology for the late reply.

I will follow your suggestion and use the class `OneShotPacketizer`. 
Thank you very much for the recommendation.

Best regards.
Diego Sánchez

closed 2020-09-28 by charluy