tsduck/tsduck #622: tsscan nit scan for Japan DVB-S/S2 (124E/128E)

trollcop, 2020-07-28:
Trying to make this work.
First of all problem 1: There's no lnb config for SkyPerfect LNB, which is operated like this
124/128 switch: 124: 40kHz tone 128: no tone. (22kHz is enough from foreign DVB card).
There's no band switch.
LO is 11200MHz.

So ignoring satellite switch stuff for now.
Their NIT is broken, so when --nit-scan retunes, I think it chooses wrong delivery system (S instead of S2)

```
tsscan -n --debug=10 --japan -f 12613000000 -s 23303000 --lnb L11200 --polarity horizontal -m "8-PSK" --delivery-system "DVB-S2" --fec-inner "3/5"
tsscan: Debug[10]: debug level set to 10
tsscan: Debug: ====> tsscan -n --debug=10 --japan -f 12613000000 -s 23303000 --lnb L11200 --polarity horizontal -m 8-PSK --delivery-system DVB-S2 --fec-inner 3/5
tsscan: Debug: loading XML file /usr/bin/tsduck.lnbs.xml
tsscan: Debug: loading XML file /usr/bin/tsduck.lnbs.model.xml
tsscan: Debug: starting discarding frontend events
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: starting discarding frontend events
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 9 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 6 (0x00000006)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 1413000 (0x00158F88)
tsscan: Debug: [2] cmd = 4 (DTV_MODULATION), data = 9 (0x00000009)
tsscan: Debug: [3] cmd = 8 (DTV_SYMBOL_RATE), data = 23303000 (0x01639358)
tsscan: Debug: [4] cmd = 9 (DTV_INNER_FEC), data = 10 (0x0000000A)
tsscan: Debug: [5] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [6] cmd = 13 (DTV_ROLLOFF), data = 0 (0x00000000)
tsscan: Debug: [7] cmd = 12 (DTV_PILOT), data = 1 (0x00000001)
tsscan: Debug: [8] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x00 on PID 0x0000
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x40 on PID 0x0010
tsscan: Debug: got 10000 packets
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got table id 0x46 on PID 0x0011
tsscan: Debug: got 10000 packets
tsscan: Debug: got 10000 packets
tsscan: Debug: got 10000 packets
tsscan: Debug: got table id 0x42 on PID 0x0011
tsscan: Debug: * tuning to --delivery-system DVB-S --frequency 12,523,000,000 --symbol-rate 23,303,700 --fec-inner auto --polarity horizontal --modulation 8-PSK
tsscan: Debug: starting discarding frontend events
tsscan: Debug: one frontend event discarded
tsscan: Debug: one frontend event discarded
tsscan: Debug: one frontend event discarded
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 1 DTVProperties:
tsscan: Debug: [0] cmd = 2 (DTV_CLEAR), data = 4294967295 (0xFFFFFFFF)
tsscan: Debug: starting discarding frontend events
tsscan: Debug: finished discarding frontend events
tsscan: Debug: tuning on /dev/dvb/adapter0/frontend0
tsscan: Debug: 7 DTVProperties:
tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 5 (0x00000005)
tsscan: Debug: [1] cmd = 3 (DTV_FREQUENCY), data = 1923000 (0x001D57B8)
tsscan: Debug: [2] cmd = 4 (DTV_MODULATION), data = 9 (0x00000009)
tsscan: Debug: [3] cmd = 8 (DTV_SYMBOL_RATE), data = 23303700 (0x01639614)
tsscan: Debug: [4] cmd = 9 (DTV_INNER_FEC), data = 9 (0x00000009)
tsscan: Debug: [5] cmd = 6 (DTV_INVERSION), data = 2 (0x00000002)
tsscan: Debug: [6] cmd = 1 (DTV_TUNE), data = 4294967295 (0xFFFFFFFF)
* Frequency: 12,523,000,000 Hz H, strength: 39%
```

So it will keep failing.
If you tell me how I can capture some sample TS from above tuning line (using tsdump or whatever?) I can provide a sample file with NIT.

trollcop, 2020-07-28:
I was able to capture some TS with PSI data in it.
[test.zip](https://github.com/tsduck/tsduck/files/4988419/test.zip)

The problem is in satellite delivery system descriptor in NIT:

When run with --japan, it shows the delivery system as ISDB-S

```
* NIT Actual, TID 64 (0x40), PID 16 (0x0010)
  Version: 8, sections: 1, total size: 945 bytes
  - Section 0:
    Network Id: 10 (0x000A)
    Network information:
    - Descriptor 0: Network Name (0x40, 64), 8 bytes
      Name: "スカパー!"
    - Descriptor 1: ISDB System Management (0xFE, 254), 2 bytes
      Broadcasting flag: 0 (Broadcasting)
      Broadcasting identifier: 7 (Satellite advanced narrow-band using 27 MHz bandwidth in 12.2 to 12.75 GHz band)
      Additional broadcasting id: 0x01 (1)
    Transport Stream Id: 12308 (0x3014), Original Network Id: 10 (0x000A)
    - Descriptor 0: Service List (0x41, 65), 21 bytes
      Service id: 33376 (0x8260), Type: 0x01 (Digital television service)
      Service id: 33399 (0x8277), Type: 0x01 (Digital television service)
      Service id: 33409 (0x8281), Type: 0x01 (Digital television service)
      Service id: 33415 (0x8287), Type: 0x01 (Digital television service)
      Service id: 33428 (0x8294), Type: 0x01 (Digital television service)
      Service id: 33442 (0x82A2), Type: 0x01 (Digital television service)
      Service id: 33711 (0x83AF), Type: 0x01 (Digital television service)
    - Descriptor 1: Satellite Delivery System (0x43, 67), 11 bytes
      Orbital position: 128.0 degree, east
      Frequency: 12.52300 GHz
      Symbol rate: 23.3037 Msymbol/s
      Polarization: linear - horizontal (0)
      Delivery system: ISDB-S
      Modulation: Advanced narrow-band CS digital broadcasting (10)
      Inner FEC: Advanced narrow-band CS digital broadcasting (10)
```

When run without, it shows up as DVB-S:

```
* NIT Actual, TID 64 (0x40), PID 16 (0x0010)
  Version: 8, sections: 1, total size: 945 bytes
  - Section 0:
    Network Id: 10 (0x000A)
    Network information:
    - Descriptor 0: Network Name (0x40, 64), 8 bytes
      Name: "|.....!"
    - Descriptor 1: unknown (0xFE, 254), 2 bytes
      0000:  07 01                                      ..
    Transport Stream Id: 12308 (0x3014), Original Network Id: 10 (0x000A)
    - Descriptor 0: Service List (0x41, 65), 21 bytes
      Service id: 33376 (0x8260), Type: 0x01 (Digital television service)
      Service id: 33399 (0x8277), Type: 0x01 (Digital television service)
      Service id: 33409 (0x8281), Type: 0x01 (Digital television service)
      Service id: 33415 (0x8287), Type: 0x01 (Digital television service)
      Service id: 33428 (0x8294), Type: 0x01 (Digital television service)
      Service id: 33442 (0x82A2), Type: 0x01 (Digital television service)
      Service id: 33711 (0x83AF), Type: 0x01 (Digital television service)
    - Descriptor 1: Satellite Delivery System (0x43, 67), 11 bytes
      Orbital position: 128.0 degree, east
      Frequency: 12.52300 GHz
      Symbol rate: 23.3037 Msymbol/s
      Polarization: linear - horizontal (0)
      Delivery system: DVB-S
      Modulation: 8-PSK (2)
      Inner FEC: unknown (10)
```

but when tsscan is invoked with --japan, it tries to retune using DVB-S delivery system instead of DVB-S2 (and fails).
It's not "really" ISDB-S in the sense that 124/128 is closer to normal DVB-S2 and using standard 8PSK and not TC8PSK like ISDB-S (BS110) does.

Anyway, I suspect this should be fixed somehow in tsscan by forcing S2 when modulation is 8PSK and --japan is specified.
... Yeah

lelegard, 2020-07-28:
Hi,

The option `--japan` is a "synthetic" option, ie. a shortcut for several individual options (including `--isdb`) which can be useful to process Japanese streams. Please RTFM for more details. 

The problem with ISDB is that this is a fundamentally broken standard. Unlike ATSC, it is neither independent nor consistent. It reuses part - but not all - of DVB. Worse, in some cases such as the satellite delivery descriptor, it reuses the same descriptor as DVB with slightly different layouts and semantics. This difference of interpretation of the same structure is the reason for your problem. 

When you analyze an ATSC stream, you find ATSC-defined tables and you know for sure that this is an ATSC stream. When analyzing an ISDB stream, it looks like a DVB stream at first. Some specific descriptors may indicate at a later time evidences of ISDB but usually too late. This is why, DVB being the default, you need to specify `--isdb ` (or some synthetic option which includes it) to tell TSDuck to use ISDB interpretations of ambiguous structures. 

I know that some Japanese satellites are not ISDB-S. If you want an accurate interpretation of their signalisation, you need to specify the right set of options, excluding `--isdb`.

trollcop, 2020-07-28:
I think you missed the part where the NIT on that satellite is broken, and specifies DVB-S delivery system with 8PSK modulation, which is not a possible combination.

Initial tune with tsscan succeeds, when it goes to next channel following NIT, there's no way to force it to use S2.

tsscan -a 1 -n --debug=10 --default-charset=ARIB-STD-B24 -f 12613000000 -s 23303000 --lnb L11200 --polarity horizontal -m "8-PSK" --delivery-system "DVB-S2" --fec-inner "3/5"

1st run:
`tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 6 (0x00000006)`

once it discovers NIT:
`tsscan: Debug: * tuning to --delivery-system DVB-S --frequency 12,523,000,000 --symbol-rate 23,303,700 --fec-inner auto --polarity horizontal --modulation 8-PSK`

result:
`tsscan: Debug: [0] cmd = 17 (DTV_DELIVERY_SYSTEM), data = 5 (0x00000005)`
^ this is *incorrect*.

lelegard, 2020-07-28:
Ok, so you confirm that this is a DVB and not an ISDB network. 

If the NIT is broken, ie. inconsistent delivery system and modulation, there is nothing we can do. We cannot do any automatic fix.

You wrote that the delivery system is incorrect _because_ the modulation is 8PSK and that TSDuck should fix the delivery system. But you write this because you know that this should be the right fix in that specific situation. But it could also be possible that the delivery system is actually DVB-S and the modulation is incorrect. After all, DVB-S is always QPSK and a careless operator may not care to set it. So, in practice, there is no absolute reason to decide in favor of fixing the delivery system or fixing the modulation. 

The NIT is broken. There are several possible ways to "fix" it. So, you just can't use it to scan the network. Sorry, but the operator is to blame, not TSDuck.

trollcop, 2020-07-28:
Yep, right.

Network is not isdb but ARIB STD-B10 tho i suppose that doesn't matter at
this point. There's a bunch of isdb-like descriptors and it really WAS
DVB-S at first, then all the channels moved to S2 but i guess they never
updated the NIT for hysterical reasons.

The modulation is actually correct and I can tune all the transponders with
TSReader in S2/8PSK mode.

Which also had the same bug with retune, but after I provided the nit
samples a year or two ago, he also made an exception for this particular
network.

Anyway, the only reason i wanted to scan because I didn't want to manually
map SID -> channel number +av pid for a dvr service app, but i found out
all i needed to provide was the freq/polarization and sid, which was easy
enough to massage from tspsi output.

If you don't think it's worth making an exception/something for this
particular config (like if --Japan and (optional?) orbital == 124 or 128
and modulation = 8psk tune it as s2, then never mind and close this bug.

Since i solved my original requirement for scanning anyway, i don't care if
it's fixed or not, and it's unlikely anyone else will come across this
problem again.

trollcop, 2020-08-01:
Closing, thanks for feedback.

closed 2020-08-01 by trollcop