tsduck/tsduck #625: Extract all information about Video PID from multicast input

Anas92230, 2020-08-13:
Hello,

I try to get all es_info about Video PID coming from multicast.
Now I use this command : tsp -I ip ip_source@multicast -l ip_interface -P until -s 5 -P analyze --normalized
I want for example extract color_space;frame_rate;interlaced_frame etc..

Thank you for your help

BR

Anas

lelegard, 2020-08-14:
Hi,

TSDuck essentially manipulates the system structure, not the video or audio content. Tools such as ffmpeg/ffprobe or MediaInfo are more appropriate to analyze video and audio content.

With TSDuck, the plugin PES reports a minimal set of information, e.g.:
~~~
$ tsp -I ip ... -P until -s 5 -P pes --pid 0x0140 --video-attributes -O drop
* PID 0x0140, stream_id 0xE0 (Video 0), AVC video attributes:
  1920x1080, high profile, level 4.0, 4:2:0
~~~

For more details, use `ffprobe` (comes with `ffmpeg`):
~~~
tsp -I ip ... -P until -s 5 -P filter --pid ... | ffprobe -i -
...
Input #0, mpegts, from 'pipe:':
  Duration: N/A, start: 65940.740278, bitrate: N/A
    Stream #0:0[0x140]: Video: h264 (High), yuv420p(tv, bt709, top first), 1920x1080 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 90k tbn, 50 tbc
~~~

For even more details, see MediaInfo.

Anas92230, 2020-08-15:
Thank you Thierry.
Excactly i already use ffprobe, but i' ll hope to use only your tool :-)

BR

Anas

lelegard, 2020-08-15:
TSDuck and ffmpeg are complementary and are often used together. TSDuck handles the TS systems and ffmpeg handles the video and audio. TSDuck handles very few things about video and does not do it well. Conversely, ffmpeg can generate a TS but does it notoriously badly. In short, you need both tools to build a system.

Anas92230, 2020-08-16:
Thank you for your help and your advice.

Anas

closed 2020-08-16 by Anas92230