tsduck/tsduck #626: Add Spain DVB-T Stream Repository

heroku-miraheze, 2020-08-16:
Add Spain DVB-T Stream Repository

lelegard, 2020-08-16:
> Add Spain DVB-T Stream Repository

What do you mean ? Please be specific and comprehensive.

heroku-miraheze, 2020-08-16:
Add all Spain Channels for Stream Repository, and also, Portugal.

lelegard, 2020-08-16:
> Add all Spain Channels for Stream Repository, and also, Portugal.

Hum... This sounds like a really strong injunction. No need for some complicated politeness but "please" is a cheap word that is always welcome.

Apart from that, the stream repository is populated from real streams that were captured live. Since I am physically located neither in Spain nor Portugal, I cannot capture DVB-T streams from those countries. In the stream repository, streams from France were captured by me and all other streams were captured and provided by TSDuck users with reception facilities in their respective countries. So, if any TSDuck user who is located in Spain or Portugal can capture local DVB-T streams, I will be pleased to publish them. In the meantime, I have no plan to travel in Spain or Portugal (especially in a Covid-19 era...)