tsduck/tsduck #627: Support SMPTE 2022-4

samarasteph, 2020-08-18:
Hi,

In our company, we have integrated tsduck in our automatic tests environment/framework to extract information from TS packet streams. 

We are currently testing a new feature implementing standard **RTP SMPTE 2022-4** (NULL packet stripping)  and tsduck raises lot of Continuity Counter errors when RTP is 2022-4 whereas no error with standard RTP stream. 

My question is, **does tsduck currently support SMPTE 2022-4 standard**? Knowing that would greatly help us to move forward to choose a new method to validate the stream.

Thank you for your answer

Kind regards,

S. SAMARA
Software engineer - ATEME corp.

lelegard, 2020-08-18:
Hi @samarasteph 

Currently, there is no support for SMPTE 2022-4 in TSDuck because no developer needed it so far.

Like most open source projects, TSDuck is developed for free by volunteers (mostly only one, sadly) for the benefit of all users. Since Ateme is a well-known commercial company with a lot of developers, if you get some benefit using TSDuck for free, I would suggest that you implement an SMPTE 2022-4 plugin. This is how open source works: get a lot for free and contribute a bit in return (especially when _you_ need this contribution).

Writing a UDP-based input plugin for TSDuck is quite easy: Simply write a C++ class which derives from `ts::AbstractDatagramInputPlugin`. Basically, the only code to write is the processing of the UDP packet content. An an example, the basic `ip` input plugin is made of 66 actual lines of C++ code (195 with blanks and comments) and the `srt` input plugin (Secure Reliable Transport) - which was developed by a contributor who apparently needed it - has 84 loc (210 with blanks and comments). I just checked with `cloc`. Quite easy, isn'it?

samarasteph, 2020-08-19:
Thank you for quick response and explanation. I agree, DIY is good idea ;).

Issue closed.

closed 2020-08-19 by samarasteph