tsduck/tsduck #629: Question: How to save a file broadcast over satellite?

Pazzeo, 2020-08-26:
Hi all,

I have a question. I would like to save a file which is broadcasted over satellite in a specific file. It should be a zip file.
I've tried the following command:
tsp -v -I dvb --device-name /dev/dvb/adapter1 --delivery-system DVB-S --fec-inner 3/4 --frequency XYZ --polarity vertical --modulation QPSK --roll-off auto -s 27500000 -P tables -a -d -p 2015  -b test.zip -O drop

but this command never ending and it continues to download the data. 
How can I do?

thanks
Matteo

lelegard, 2020-08-26:
> a file which is broadcasted over satellite

Hum, what do you mean by "file which is broadcasted"? A satellite broadcast is a continuous and endless stream of TS packets, not a "file" per se.

If you want to save the complete TS, use `-O file filename.ts`. Note that the file will be a TS file, not zipped.

In your command `-P tables -a -d -p 2015 -b test.zip`, you save all sections from PID 2015 in a binary file (this won't be a zipped file, even though you name it `.zip`). And indeed, it will never end because a satellite transmission never ends by definition (there is no "end of file" on input). If you want to stop the processing on some condition, have a look at the plugin `until`.

What I suspect is that the sections in your PID 2015 contain some encapsulation protocol which in turn contains a "file" which is endlessly cycled. There are many possible encapsulation protocols in a TS. You have to first understand which encapsulation protocol is used and then find a software that implements this decapsulation protocol to extract the file from the binary sections. TSDuck can capture the sections from the satellite. Then you have to find a way to implement the target decapsulation protocol (I have no idea what it is).

Pazzeo, 2020-08-26:
Hi lelegard,

thanks a lot for the clear answer. I mean that the "file" is encapsulated and then cycled in the TS. So you are right, I need first to understand which is the encapsulation protocol.

thanks