tsduck/tsduck #630: error: unknown type name 'clockid_t'

ryandesign, 2020-08-30:
Hi, tsduck 3.22-1953 fails to build on slightly older versions of macOS (OS X 10.11 with Xcode 7.3.1)

```
/opt/local/var/macports/build/_Users_travis_build_macports_macports-ports_multimedia_tsduck/tsduck/work/tsduck-tsduck-d1e6e0d/src/libtsduck/base/tsTime.h:716:48: error: unknown type name 'clockid_t'; did you mean 'clock_t'?
        static NanoSecond UnixClockNanoSeconds(clockid_t clock, const MilliSecond& delay = 0);
                                               ^~~~~~~~~
                                               clock_t
/usr/include/sys/_types/_clock_t.h:30:33: note: 'clock_t' declared here
typedef __darwin_clock_t        clock_t;
                                ^
```

Here is how another project solved this problem:

https://github.com/RIOT-OS/RIOT/pull/2811/files

lelegard, 2020-08-31:
Hi @ryandesign 

Thanks for reporting this. I started the port of TSDuck on Mac when I switched my own development environment to macOS. I think it was Sierra, ie. 10.12. Since then, the current source code compiles with clang on macOS. 

It is quite possible that at the time of 10.11, the realtime part of POSIX was not implemented on macOS.

The modification you refer to is simply:
~~~
#if defined(__MACH__)
    typedef int clockid_t;
#endif
~~~

This cannot be accepted as such. First, it may compile on macOS 10.11 but it will certainly fail on subsequent versions. For instance, on my macOS 10.15.6 system, `clockid_t` in defined as `typedef enum {...} clockid_t;` in `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/time.h`. So, the above "fix" will in fact fails on 10.15.6 because of duplicated and conflicting definitions of the same type.

First, if such fix should be applied, it would need at least to be conditioned to the platform version, assuming that we may exactly identify after which version `clockid_t` was defined on macOS.

Second, even if the proposed fix compiles on 10.11, does the code work? If `clockid_t` is not defined, this means that the corresponding POSIX features are probably not supported. I have no such old version of macOS to test. If you have one, you may try `make test` which builds and run some low-level unitary tests, including time features.

I cannot test on macOS 10.11 because I have no such old system. As maintainer of MacPorts, you probably have a wide range of macOS versions. If you can propose a fix that compiles and runs correctly on all system versions, please submit a PR. If you don't have time for this, then TSDuck should be simply marked as unsupported on macOS 10.11 and older.

As far as I am concerned, I do not have the resources to support TSDuck on old platforms, macOS, Linux or Windows.

ryandesign, 2020-09-01:
`clock_gettime` and friends were introduced in macOS 10.12. We have a [library of compatibility functions](https://github.com/macports/macports-legacy-support) including `clock_gettime` that we can try including in the MacPorts tsduck port on older systems.

lelegard, 2020-09-01:
Thanks @ryandesign 

> clock_gettime and friends were introduced in macOS 10.12.

OK, that explains the issue. This is the version I started with. So, I never encountered the problem.

> We have a library of compatibility functions including clock_gettime that we can try including in the MacPorts tsduck port on older systems.

Why not, as long as this is limited to MacPorts environment and to pre-10.12 versions only. I do not want to clutter the code of TSDuck to support old versions.

The project is pretty large (~ 350000 C++ loc) and there is only one maintainer (me). After 15 years in the life of the project, to keep the code robust across all modifications, I have a strict code cleanup policy. I regularly go through sessions of refactoring to get rid of old code and use cleaner or more robust features. The price to pay is the abandonment of the past. The gain is remaining alive and ready for the future.

At some point, for instance, I switched to C++11, a revision of the language with many interesting features such as meta-programming  which allowed the introduction of much more robust features. Some users complained that old compilers could not handle the code. Too bad. With old platforms, you must stay with old versions of the project. This may seem a bit harsh, but I do not have the resources to manage the past and the future at the same time and I choose the future.

ryandesign, 2020-09-01:
I wanted to report the issue to you since I wasn't aware of your stance on compatibility. If your goals are to move the code forward and avoid compatibility code for old systems and compilers, that's totally reasonable.

closed 2020-09-01 by ryandesign