tsduck/tsduck #631: Q: Inject teletext subtitling into MPEG-TS

rhymerjr, 2020-08-31:
Hi!

I already read that TSDuck can "Extract specific encapsulated data (Teletext, T2-MI)."

My question is that is it possible to inject teletext carousel?
We are searching a solution to inject teletext subtitling quite accurately.

Thank you!

lelegard, 2020-08-31:
> My question is that is it possible to inject teletext carousel?

There is currently no such feature in TSDuck. As a matter of fact, the same question has already been asked here several times: can we insert audio or video or subtitles (your case). 

As a test, you may try ffmpeg but I do not know if it handles teletext streams correctly.

> We are searching a solution to inject teletext subtitling quite accurately.

"Quite accurately" is precisely the issue. You need a MUX, an accurate MUX. And this is a complex software with lots of corner cases. I would have liked to have the opportunity to develop one but I never found the time to do so. This complexity of doing this "accurately" makes it difficult to find such tool in the free open source world. You can't have everything for free...

rhymerjr, 2020-08-31:
Dear Thierry,

Thank you for your answer!

I know that this is a complex thing. 
We know products that can do that ( cards of deltacast for example ) but they are very expensive.

Anyway, if TSDuck can extract teletext from the transport stream maybe we will give it a try for the purpose of checking back the subtitles.

closed 2020-09-01 by rhymerjr