tsduck/tsduck #632: tspcontrol with http input plugin causing issue for tsp

rezna11, 2020-09-02:
Hi, I wanted to try `tspcontrol` to restart the `http` plugin but I if I use `--infinite --ignore-errors` options, the `http` plugin isn't restarted and `tsp` goes into a loop and can't be terminated. Without `--infinite --ignore-errors` it works.

Here's the `tsp` command:

`tsp -v --control-port 188 -t -I http --ignore-errors --infinite http://10.0.92.244:8088/channel1 -P continuity -O ip 239.0.0.11:5000`

Then the `tspcontrol` command:

`tspcontrol -d -v --tsp 127.0.0.1:188 restart -v 0 --ignore-errors --infinite http://10.0.92.244:8088/channel2`

And `tsp` then stars looping like this (it seems to be stuck with the original options, unable to break that loop before restarting the plugin with new options):

```
* 2020/09/02 13:44:34 - http: unknown initial input bitrate
* 2020/09/02 13:44:58 - control commands: received from 127.0.0.1:56810: restart -v 0 --ignore-errors --infinite http://10.0.92.244:8088/channel2
* 2020/09/02 13:44:58 - http: restarting due to remote tspcontrol
* 2020/09/02 13:44:58 - http: downloading from http://10.0.92.244:8088/channel1
* 2020/09/02 13:44:58 - http: MIME type: video/mpeg, expected size: unknown
* 2020/09/02 13:44:58 - Warning: http: MIME type is video/mpeg, maybe not a valid transport stream
* 2020/09/02 13:44:59 - http: downloading from http://10.0.92.244:8088/channel1
* 2020/09/02 13:44:59 - http: MIME type: video/mpeg, expected size: unknown
* 2020/09/02 13:44:59 - Warning: http: MIME type is video/mpeg, maybe not a valid transport stream
* 2020/09/02 13:44:59 - http: downloading from http://10.0.92.244:8088/channel1
* 2020/09/02 13:44:59 - http: MIME type: video/mpeg, expected size: unknown
* 2020/09/02 13:44:59 - Warning: http: MIME type is video/mpeg, maybe not a valid transport stream
* 2020/09/02 13:45:00 - http: downloading from http://10.0.92.244:8088/channel1
* 2020/09/02 13:45:00 - http: MIME type: video/mpeg, expected size: unknown
* 2020/09/02 13:45:00 - Warning: http: MIME type is video/mpeg, maybe not a valid transport stream
* 2020/09/02 13:45:01 - http: downloading from http://10.0.92.244:8088/channel1
* 2020/09/02 13:45:01 - http: MIME type: video/mpeg, expected size: unknown
```

lelegard, 2020-09-02:
Hi Jan,

Sorry about this. This is a bug of course. Quite easy to fix, fortunately.
I just fixed it in commit c97d550827b4fca354fb363dfbe9625262f37e6a. Let us know if it works for you.

rezna11, 2020-09-03:
Thank you, I tried and it's fixed.

closed 2020-09-03 by rezna11