tsduck/tsduck #633: Is fork broken?

rezna11, 2020-09-03:
Hello. I've built latest TSDuck (Windows) but somehow it now crashes when `fork` plugin is used. Doesn't crash with an older version (3.21-1819).

Here's the ouptut of the latest version crashing:

```
C:\>tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.23-1962

C:\>tsp -d -v -b 10000000 -I null -P regulate -P fork "tsp -O ip 239.0.0.20:6000" -O ip 239.0.0.11:15000
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d -v -b 10000000
* Debug: debug level set to 1
* Debug: null: debug level set to 1
* Debug: null: ====> tsp -I null
* Debug: ip: debug level set to 1
* * Debug: registering processor plugin "regulate", status: ok
Debug: ip: ====> tsp -O ip 239.0.0.11:15000
* Debug: trying to load "C:\Program Files\TSDuck\bin\tsplugin_regulate.dll"
* Debug: regulate: debug level set to 1
* Debug: regulate: ====> tsp -P regulate
* Debug: fork: debug level set to 1
* Debug: fork: ====> tsp -P fork 'tsp -O ip 239.0.0.20:6000'
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: fork: creating process "tsp -O ip 239.0.0.20:6000"
* regulate: minimum packet burst duration is 2,000,000 nano-seconds
* Debug: null: initial buffer load: 44,620 packets, 8,388,560 bytes
* null: initial input bitrate is 10,000,000 b/s
* Debug: control commands: found 2 packet processor plugins
* Debug: null: input thread started
* Debug: ip: output thread started
* Debug: fork: packet processing thread started
* Debug: regulate: packet processing thread started
* regulate: regulated at bitrate 10,000,000 b/s
* Debug: regulate: new regulation, burst: 2,406,400 nano-seconds, 16 packets
```
[image]

rezna11, 2020-09-03:
Actually it seems fork plugin wasn't built at all, so when I pasted over the new dlls and exes, the fork plugin remained from previous version and that's probably why it fails. I will try building the installer.

lelegard, 2020-09-03:
Hi Jan,

Now, the plugin `fork` and a number of other simple and basic plugins are included in `tsduck.dll`. There is no longer any `tsplugin_fork.dll`. If I remember correctly, 8 plugins are now fully included in `tsduck.dll`. For technical reasons, the code of these plugins was already in `tsduck.dll` and the DDL's were just triggers to register in `tsduck.dll` plugins which were already in `tsduck.dll`. So, to remove useless complexity, the DLL's were removed.

If you build and run an installer, the installation will remove the obsolete DLL's and you will be safe. But if you build the current code of TSDuck inside a build area where a rather old built was performed, the old `tsplugin_fork.dll` remain. I need to investigate what is the impact of having a builtin plugin and an external plugin with the same name. Maybe this is the root cause of your problem.

Anyway, before building a completely new version of TSDuck, I recommend to cleanup the build area first (`make distclean` on Linux and macOS, run `cleanup.ps1` on Windows).

lelegard, 2020-09-03:
Checking the code, the presence of a plugin DLL with the same name as a builtin plugin should be completely ignored. So, the problem must be somewhere else. I will look into it.

rezna11, 2020-09-03:
I did a fresh install of Windows 10 a week ago, so this was the first time I was building TSDuck here. Even with the installer, I'm unable to run the `fork` plugin. I tried the installer I just built as well as TSDuck-Win64-3.22-1953.exe downloaded from your website. I also tried both Windows portable versions downloaded from tsduck.io. I restarted Windows. Nothing helps.

lelegard, 2020-09-03:
OK, I found the issue. This is a bug which was introduced between version 3.21 and 3.22. Easy to fix, a silly omission. But the bad news is that the official release 3.22 is affected. So, I need to release 3.23 asap.

lelegard, 2020-09-03:
I fixed the issue in the repo. You may try it. In the meantime, I start the early unexpected production of official version 3.23.

rezna11, 2020-09-03:
Thank you, `fork` is good now.

lelegard, 2020-09-09:
Version 3.23 is released with the fix.

closed 2020-09-09 by lelegard