tsduck/tsduck #634: Please, add me Spain Hispasat 30W streams

heroku-miraheze, 2020-09-05:
Please, add me Spain Hispasat 30W streams, the frecuencies is: 10917V, 10770H and 10923V. Thank you.

lelegard, 2020-09-05:
Anyone with a dish on Hispasat 30W?
I don't have one.

heroku-miraheze, 2020-09-05:
You have an dish with Hispasat 30W.

lelegard, 2020-09-05:
> You have an dish with Hispasat 30W

Who is that "you"?

heroku-miraheze, 2020-09-05:
Lelegard have an dish with Hispasat 30W, dish it and now capture me streams.

lelegard, 2020-09-05:
> Lelegard have an dish with Hispasat 30W, dish it and now capture me streams.

Lelegard is me and I do not have a dish on 30W. The fact that I once captured a stream for you on 19.2E means that I have access to a dish on 19.2E, not 30W! The former is over Eastern Europe while the latter is right in the middle of the Atlantic Ocean. Not really the same direction. Please stop assuming things about which you know nothing.

Furthermore, this is the third time you request streams captures in a rather imperious manner. People are here to offer benevolent help to TSDuck users. Please consider that we are not stream capture robots. You seem to have a strong interest in various Spanish streams. If this is for professional reasons, then you should find the right professional resources to achieve your project.