tsduck/tsduck #635: Question: can tsanalyze periodically output its analysis without stopping the stream

fatpelt, 2020-09-05:
Can tsanalyze be configured to periodically output its analysis, i don't see any option.  ideally what i'd like is something like:

`
srt -> tsp (srt) -> sanalyze -> multicast (output)
`

i could also fork and use until, but that doesn't look like it can restart the stream after it hits the end of stream?

```
                       -> until -> analyze
                       |
  srt -> tsp (srt) -> fork -> multicast (output) ->
```

lelegard, 2020-09-05:
Hi @fatpelt,

Yes, this is possible. The plugin `analyze` accepts all options from the `tsanalyze` commands, plus specific options which are useful only in the plugin: `--interval`, `--output-file`, `--multiple-files`. Please read the user's manual, the `analyze` section in chapter 4.

fatpelt, 2020-09-05:
Ah!  I was only reading in chapter 3, assuming they were the same.   My
apologies, and thanks!

closed 2020-09-05 by fatpelt