tsduck/tsduck #636: Only p/f EIT export by tstables --pid 0x12

bergerp128, 2020-09-07:
Command: tstables --pid 0x12 --xml-output filename.xml filename.ts exports only p/f EIT (actual and other) but not schedule! What is wrong: this command or tsduck? Where I am making mistake?

lelegard, 2020-09-07:
This is the "normal" behaviour for EIT's. EIT's, especially EIT schedule, are not complete tables in the MPEG sense. They are just sparse sections according to the DVB usage rules. Using `tstables --all-sections`, you can get a dump of all individual sections. But an XML file shall contain complete tables only and sparse EIT schedule sections accumulate in memory, waiting for missing sections to form complete tables. But those sections will never come.

EIT p/f are complete tables because section 0/1 contains the present event and section 1/1 contains the following one.

To manipulate EIT schedule, the `tstables` command has specific options such as `--fill-eit`, `--pack-all-sections` and `--pack-and-flush` (read the user's guide for a complete description of them). These options _modify_ the section numbering in different ways so that EIT become complete tables. But keep in mind that this is a _transformation_ of the actual EIT's. Getting all EIT schedule as "complete tables" with their original section structure is impossible by design.

bergerp128, 2020-09-07:
Thank you lelegard. 
I've tried --pack-all-sections in the meantime (I've read #537 issue). XML file is very large and I haven't been able to analyze it yet. I will try the other options mentioned.
I have another question: Why Extended Event Descriptor (or another descriptors) is not exported?

lelegard, 2020-09-07:
>  Why Extended Event Descriptor (or another descriptors) is not exported?

Please be more precise. All descriptors are exported in XML when the syntax of the table is known (which is the case for the EIT).

bergerp128, 2020-09-08:
Very sorry lelegard, I analyzed this stream in Dektec analyzer and I noticed that EIT is w/o Extended Event Descriptor also. All EIT descriptors displayed in Dektec analyzer are in exported XML file by tstables command.
Thanks for help

closed 2020-09-08 by bergerp128