tsduck/tsduck #637: Specific TSP plugin to modify H264 PES packets

sym39, 2020-09-09:
Hello,

I would like to implement a specific plugin that modify PES H264 packets by adding an extra custom SEI for each frame.
The idea would be to get the PES content, then to parse the H264 to be able to add this custom SEI after each slice.

I get a look in the [PES plugin source](https://github.com/tsduck/tsduck/blob/master/src/tsplugins/tsplugin_pes.cpp), but I am not sure it is feasible:

- `ts::PESPlugin::handlePESPacket(PESDemux&, const PESPacket& pkt)` or `ts::PESPlugin::handleAVCAccessUnit()` may not be the correct place to modify the packet (const ...)
- I read that adding new data requires to add stuffing packets in input. But I am not sure how I can access those stuffing packets to regenerate a bigger PES data, where the initial data is left untouched, but the data after each slice would be shifted.
- or should I copy this buffer, modify it (add SEIs and so on), store it on a shared object? but I have to _reset_ the TS packets belonging to the PES, then to regenerate another PES packet? and would TSP be able to remux this PES packet?

In short, do you think TSP is well suite for this kind of project?
may be could you point me to a plugin that could be used as a basis to modify/inject PES data?

Thanks a lot in advance,
--
Sylvain

lelegard, 2020-09-09:
Hi @sym39,

This is a quite interesting idea. This is certainly feasible, not the easiest plugin to develop but quite feasible.

You have identified the first difficulty: adding data, larger than original data. This is the easiest one to solve. Since tsp is designed to work also on streamed data, packets are processed one by one and there is no way to go back. There is also no way to add new TS packets. The typical use case I had in mind from day one was receiving a TS through a cheap tuner, modify it in real time and reinject it in a Dektec modulator card with the same parameters. The bitrate must be unchanged, no packet can be added and the processing must be stable.

So, all "new" data, all kinds of "insertion" are done by replacing stuffing. The idea is that you keep the TS packets you want to insert and when your processPacket() method is called with a TS packet on PID_NULL, simply replace the complete TS packet with the one you want to insert. Of course, you need at least as many stuffing packets as new packets. In CBR broadcast streams (satellite or terrestrial), there are always a few kilobytes (sometimes megabytes) of stuffing. So, this is fine for most usages like yours. In VBR IP streams, there is usually no stuffing at all. But since this is VBR, you can artifically boost the input bitrate with stuffing (see the user's guide), use your insertion plugins which replace stuffing and end with a `filter` plugin which removes the residual stuffing. In all cases, this is transparent for the plugin: it simply replaces null packets. When you write the command line, you know the context and you know if you need to add input stuffing (this is a command line option).

The second difficulty is the induced latency or "time shift". You modify existing data. So, you need to accumulate enough data to analyze them and then replace the content of the input PID with the new data. In short, the scenario is always the same:

- Identify the target PID, explicitly on command line or after analyzing the signalization, typically a PMT.
- Accumulate data from the input PID. In the startup phase, as long as you do not have enough data to create the replacement packets, replace the input PID with null packets (return TSP_NULL in processPacket() method). More generally, for each packet of the input PID, if you do not have new data immediately available, replace it with a null packet.
- When you have new data, replace all packets from the target PID with the next new packet.
- If your replacement data are larger, try to permanently evaluate how many new packets are created. When you encounter a null packet on input and your "extra packet counter" is not zero, just replace it with a new packet. This is the best way to absorb the extra output.

This scenario is typically used in all plugins which modify the signalization (plugins `pat`, `pmt`, `nit`, etc). You must do the same, except that you process PES packets instead of sections. In a way, this is easier for PES.

If the "duration of a PES packet" (which depends on the structure of the stream and GOP size) is an acceptable latency for you, then you can simply use a `PESDemux` and your plugin must implement `PESHandlerInterface` to receive demuxed PES packets. Each time your receive a PES packet, you can generate the new one, packetize it, evaluate the extra number of TS packets (for null packet replacement) and then reinject those buffered TS packets later, when your processPacket() method receives packets from the target PID or null packets when needed.

If you absolutely need to reduce the latency more than the "duration of a PES packet", it is much more complicated. Feasible but much more complicated. You need to parse TS payloads on the fly, keeping the context of the current PES packet, determining where new data are needed and regenerate the new PES packet on the fly, without waiting for the end of the PES packet. Unless you need ultra low latency, I do not recommend this approach.

One last point, in the context of the simpler method with `PESDemux`: you need to split the modified PES packet into TS packets. Unlike sections for which the `Packetizer` class and its subclasses are available, there is no PES packetizer class in TSDuck, only a PES demux. There are several reasons for that. First, and more importantly, I never needed a PES packetizer (otherwise it would already exist). Second, packetizing PES is trivial while packetizing sections can be extremely complex, especially when several tables with different cycles are muxed in the same PID. This is why section packetizers exist to hide that complexity.

If you go through that route, I can add a PES packetizer in TSDuck quite easily.

Alternatively, it could be interesting to add some generic `PESProcessor` class which hides the demux and remux of PES packets. It would call some new interface in a class of the application to "process PES": the application is invoked with a PES packet, processes it or replaces its content and that's all. I could do this. The core TSDuck library hides the mechanics of PES replacement and the application focuses on the processing of each PES packet content.

sym39, 2020-09-10:
Hi @lelegard,

First at all, I would like to thank you for this quick and exhaustive answer.

Next I will try to reformulate what should be done to check I have correctly understood. I will limit to the simpler case that gets whole PES packets as it is enough for me.

Big picture:

- Ensure the data I get at input contains stuffing TS packets. Could be done via tsp itself: `tsp --add-input-stuffing 1/10 ...`
- Also remove stuffing with the filter plugin, _at the end_: `tsp ... -P myplugin [...] -P filter --negate --pid 0x1FFF -O ip [...]`
- create a new plugin _myplugin_ inheriting `ProcessorPlugin` and `PESHandlerInterface`, like the PES plugin; it may take a `pid` option to identify the H264 stream (next it could get the information from the stream itself if _stream_type_ is easely available).

In the code, I understand that most of the work has to be done in `processPacket()` like in the PES plugin, one has to call `_demux.feedPacket(pkt)` (for the H264 pid only) to ensure the PES packet will be parsed by tsp. 
First read packets at initialization will be dropped on output, by returning `TSP_NULL` (more below).

in `handlePESPacket()`, I have to parse the H264 as described above, then regenerate it with custom SEIs in it. Having a new modified PES packet, I need to packetize it and somewhat to fill a `TSPacketPtrQueue` of `TSPacket` objects.

Back in `processPacket()`, when I receive a `PID_NULL` packet, or one for the H264 pid, I have to remove the first packet from the queue to pass it back to the caller (`pkt = *_queue.front(); [...]`)

Also probably the most difficult part of the job is to packetize a PES packet into TS packets. I guess the `Packetizer` class could be a good starting point? (of course, if a PES packetizer were present, it would be nice, but I don't feel very comfortable to ask you to do _this_ or _that_ when this project is still at early stage, and is not a personal project ...)

Many thanks again,

--
Sylvain

lelegard, 2020-09-10:
Hi @sym39,

You got it, exactly.

> Also probably the most difficult part of the job is to packetize a PES packet into TS packets. 

Not at all. It is quite easy on the contrary. And since it is generally useful anyway, I just did it. For PES packetization, I reproduced the same hierarchy of classes as for section packetization : pull mode <- push mode <- one shot. Considering that I refactored the generic properties in an abstract superclass, common to section and PES packetization, you get the following hierarchy of classes:

~~~
AbstractPacketizer <- PESPacketizer <- PESStreamPacketizer <- PESOneShotPacketizer
~~~

In you case, I recommend to use `PESStreamPacketizer`. When you build a modified (or even unmodified) PES packet, just call `addPES()`. And when you want to replace a TS packet from the original PID, just call `getNextPacket()`. There is no need to keep a list of generated TS packets, they stay inside the packetizer. And `getNextPacket()` can be used all the time, even in the startup phase. If there is a generated packet from the PES, it is returned. Otherwise, it returns a null packet. This is why you can blindly use it all the time.

The difficulty you have to solve is the whether or not you should replace null packets with packets from the packetizer. If the amount of output packets is the same as input, just replace the packets from the input PID, no need to replace null packets. If your output is _slightly_ larger than input, then you must replace _some_ null packets, but not all. Just enough to compensate the output volume increase. It you replace all null packets, then your output PES PID is likely to burst. Maybe a problem or maybe not, your choice. Or just test.

The most difficult part is now the parsing of AVC data and insertion of your custom SEI. But this is the core of you own work.

lelegard, 2020-09-10:
Tomorrow, after the nightly CI/CD process, the doxygen doc will have been updated on tsduck.io and you will see this class diagram:

[image]

sym39, 2020-09-11:
Hi @lelegard ,

I also just get the modifications from gitlab, and also see how to use the classes in [utestPESPacketizer.cpp](https://github.com/tsduck/tsduck/blob/master/src/utest/utestPESPacketizer.cpp) (difference is that the test uses the `PESOneShotPacketizer` and I agree, I should use the `PESStreamPacketizer::getNextPacket()` like you described above).

Many thanks again,

--
Sylvain

lelegard, 2020-09-11:
Don't focus too much in the unitary tests. The code is very different from what we do in applications. I just try to exercise many different features in a very short way to track potential regressions. Using `PESOneShotPacketizer` was the fastest way to exercise the complete hierarchy of classes because it is the deepest subclass (using `PESOneShotPacketizer` actually uses `PESStreamPacketizer` and `PESPacketizer`).

For higher level non-regression tests, there is the [TSDuck test suite project](https://github.com/tsduck/tsduck-test). But these tests use TSDuck commands and plugins and the PES packetizer classes are not yet used in any of them.