tsduck/tsduck #638: http plugin stops at TS sync error - fix/enhance --ignore-errors?

rezna11, 2020-09-11:
Hello.

The `--ignore-errors` option for the `http` plugin does not work the way I'd like it to. Maybe it's supposed to ignore other kind of errors, specifically connection errors, but the only error I usually get is some corrupt TS packets, like this one:

`* Error: http: synchronization lost after 97,004,198 packets, got 0xDA instead of 0x47`

But I'd like TSDuck to ignore these errors too and keep the `http` plugin run, not quitting. The loss of sync is unfortunately quite common, for example whenever I connect to the Windows PC using remote desktop - the whole TSDuck and DekTec stuff just freezes for a moment and TSDuck terminates because of the sync error.

rezna11, 2020-09-11:
But I guess this might not be the `http` plugin causing TSDuck to terminate, it's perhaps `tsp`. So I guess I need something like `--continue` option in `tsresync`, but in `tsp`...?

lelegard, 2020-09-11:
Hi Jan,

The input plugins just receive of a stream of input data. Checking the validity of TS packets is done at `tsp` level, as you noticed. `tsresync` is a different tool with a different approach: it works on files while `tsp` works on the most generic form of transient stream. Including the `tsresync` features in `tsp` would introduce too many constraints on the type of stream.

That being said, I think that your problem is different. HTTP is a TCP-based protocol. It is reliable, using retries when necessary. On the client side (as used in the `http` plugin), either you receive exactly the data which was sent by the server or the connection breaks. I have never seen (I repeat: never) corrupted TCP-based data. And not only in the context of TSDuck. In other words, if your system presents corrupted TCP data to the application, this is the symptom of a deeper system problem and you will run into many other troubles on that system. I do not believe that remote desktop - as bad as it is - is responsible for that.

So, I would rather investigate on the HTTP server side. It is probably the one that sends corrupted data in the first place. Or maybe it just sends an HTML error message after a reconnection (HTML is "corrupted TS packet"...). Try `tsp --debug` to display HTTP headers.

rezna11, 2020-09-11:
Thanks for explaining this to me. I realize now that the corrupt TS packet is not actually a corrupt TS packet but a bug in the HTTP server. Then let me ask if it’s possible to work around this bug in tsp, other than run tsp in loop and restart whenever the http server sends non-TS data.

lelegard, 2020-09-11:
> Then let me ask if it’s possible to work around this bug in tsp, other than run tsp in loop and restart whenever the http server sends non-TS data.

How would you _precisely_ specify this?

Should we limit that to the `http` plugin? Or shall we generalize this at `tsp` level? But, in the latter case, how would you define "restart"?

How would you define "non-TS data", vs. corrupted or desynchronized TS data?

When using option `--ignore-errors` as you do, how would you classify errors which should be ignored and errors which would still not be ignored, despite `--ignore-errors`?

rezna11, 2020-09-11:
Well you say that validity checking is done at `tsp` level, so this should be implemented at the `tsp` level. But then you also added that it would introduce too many constraints, which I'm not sure why. 

My thinking was based on the assumption that `tsp` expects MPEG-TS data from the input. Just like any other professional device - and in case the incoming data is not MPEG-TS, it raises an alarm but tries to resync to the MPEG-TS stream, instead of shutting itself down. So I thought this would be achievable in `tsp` too.

That would make more sense than implementing any checks in the `http` plugin. Originally I thought that the input plugin was responsible for telling `tsp` to quit but since that's not the case, I don't see any reason to involve the `http` plugin in this, other than a situation when implementing resync in `tsp` was not desired - then the `http` plugin would need a new option to resync and pause feeding `tsp` until resynced, so that `tsp` doesn't detect any problem further down the path, other than drop in bitrate and CC errors.

lelegard, 2020-09-14:
This is probably not as simple as that. Not all input plugins pass raw uninterpreted binary data to `tsp`. Some input plugins more or less parse inputs to extract packets. The plugins `file` and `fork`, for instance, try to detect the input file format (see the recently added option `--format`). Similarly, the plugin `ip` parses the UDP packet to find TS packets independently of encapsulation protocols such as RTP. So, those plugins will reject completely desynchornized packets before reaching the `tsp` layer;

lars18th, 2020-10-02:
Hi @lelegard ,

After reading this thread, I have a question... So, I have readed the documentation and I'm checking the code, but I'm not sure about the expected behaviour:

- When using `-P http --ignore-errors -i ...` if the received stream has _corrupted_ TS data, for example, because the server includes some headers in the middle of the stream, or because the source is a corrupted stream with errors from a tuner, in this precise case what is the behaviour?
  - Option A: The HTTP plugin continues reading the incomming data, trying to resync.
  - Option B: The HTTP plugin throws an error, stop the socket and reconnects.
  - Option C: The HTTP plugin restart without closing the socket.

What is the current behavior? Can you describe it, please?
Regards.

lelegard, 2020-10-02:
The structure between `tsp` and its plugins is very modular and based on the principle that each element is doing its basic job the simplest possible way.

First, input plugins just do input. So the `http` plugin just reads data from the HTTP connection.

The input thread in `tsp` gets packets from the input plugin, whichever it is, and validates all packets one by one, verifying that they start with 0x47. When the input plugin returns an incorrect packet, `tsp` reports a message such as _"Error: http: synchronization lost after XX packets, got 0xNN instead of 0x47"_ and stops.

So, it is the responsibility of each input plugin to return valid TS packets. Depending on the nature of the plugin, it can try to extract TS packets from some uncertain blob of data or simply assume that the input must be a valid stream of packets, letting `tsp` validate that the data are actual TS packets. 

Let's take some examples:
- The plugins `file` and `fork` anticipate several file formats (TS, M2TS, etc). By default, the beginning of the file is analyzed to guess the file format. When the format is determined, the plugin uses it all the time and extracts TS packets.
- The plugin `ip` anticipates all kinds of encapsulations (raw TS, RTP and other exotic protocols). Each UDP datagram is individually analyzed some basic heuristics locates the TS packets inside it. Again, only the play TS packets are returned to `tsp`
- Plugins such as `dvb`, `dektec`, `http` or `hls` always expect raw TS packets. So, they just receive and pass data to `tsp` without further analysis because some common validation occurs in `tsp` and will stop the process in case of data error. Here, a data error is a real error.

Concerning the option `--ignore-errors` in the `http` plugin, it is documented as _"With `--repeat` or `--infinite`, repeat also in case of error. By default, repetition stops on error."_ So it is designed to recover from network error, not data content error.

lars18th, 2020-10-02:
Hi @lelegard ,

Thank you for your response!

> So it is designed to recover from network error, not data content error.

Then, the expected behaviour is different from A, B and C, right? My concern is about this:
- With a Musticast DVB Tuner you can get the feed from RF and generate a multicast stream with the TS packets, but if you configure it to bypass _all_ data, including packets with errors, then the multicast stream can have errors (at TS stream level, not at UDP network level). Then using the IP input with TSP you can read the stream with errors without stops, as every incoming UDP packet is analyzed and errors are discarded.
- However, if you want to change the multicast UDP to HTTP unicast, this behaviour for the HTTP input inside TSP can't be replicated. That's because any error in the stream triggers a fatal error. And this fatal error requires a TCP reconnection.

It's my assuption true? Or the described behaviour is not the current behaviour with `-P http --ignore-errors -i ...` ?
And ff yes, then why not include some alternative to the HTTP plugin to discard erroneous packets without closing the TCP socket?

lelegard, 2020-10-02:
A DVB demodulator is designed, either in hardware or in driver, to drop invalid packets. So, they never reach the `dvb` input plugin. On Windows, I have noticed in the past that the DirectShow layer was sometimes responsible for some desynchronization and the TSDuck DirectShow specific sink filter resynchronizes the stream. But it simply does what the driver should do (or what DirectShow should not do).

And generally, `tsp` never attempts to replicate error situations from input to output.

And yes, there is a difference of behaviour in case of error between UDP and HTTP. This is inherent to their nature: unconnected vs. connected.

> why not include some alternative to the HTTP plugin to discard erroneous packets without closing the TCP socket?

It could be. But, an additional layer of verification should be added inside the plugin. And if we do this in the `http` plugin, there is no reason for not doing it in all other plugins (at least those which rely on a valid TS input). Possible, just a lot of work...

lars18th, 2020-10-02:
Hi @lelegard ,

Please, let me to say something about this sentence: 

> A DVB demodulator is designed, either in hardware or in driver, to drop invalid packets. 

Not really! From my point of view, and I know you have a lot of experience, we need to distinguish between two different cases:
- When the RF signal has errors, and the demodulator can get _some_ data. Then a "valid" TS packet is generated with a payload with errors (and the demodulator sets the error flag in the TS header).
- When the demodulator can't get any valid data. Then in this case, it's true and nothing (or NULL packets) are generated and all of the incoming data is discarded.

After saying that, actually not related with the HTTP input, I suggest to not add any additional layer to the HTTP input plugin. Because you're right and the verification of the incoming data doesn't have sense at this point.

But... I suggest to add a simple "--no-close-outsync" option to continue reading when the SYNC of TS packets are lost. In this case the plugin can continue sending the incoming data to the next plugin without closing the TCP socket. Then another different plugin can implement the same behaviour already included in the module that reads multicast streams to resynchronize when garbage data is comming.

For example, I see very interesting the request from #618 regarding the ZeroMQ support. This implies to use TCP sockets as the transport layer. But the same requirement will be true than with HTTP: if the stream has errors, we _need_ the option to continue the processing and discarding data, but not closing the connection. I feel that's the key point.

So, please consider to add some support for this. I'm sure you can provide a more versatile and simple solution to the problem. But the problem exists, and without some enhancement it's impossible to run a reliable processing when the input is TCP. Mainly because the assumption of _data always valid_ in reliable channels, what is not true if the server can send invalid data.

I hope you almost think on this.
Regards.

lars18th, 2020-10-03:
Hi @lelegard ,

After thinking more on this, please consider this example: `tsp -I http --ignore-errors -i --raw-output | tsresync | tsp ...`

As you can imagine, here the HTTP plugin with the parameter `--raw-output` will pass _any_ data, even if it's or not a valid TS packet. The behaviour is then to not close the TCP socket analyzing the incoming data.

What you think now? This have sense and it's easy to implement?

Note: I'm not sure if a new RAW output module is necessary. Perhaps an input module can't directly write to the output. Then in this case, a more global change is necessary. But you can define a new `--format` (section 2.1.2) named `raw` to add it to the current `TS`, `MPTS` and `tsduck` formats. The advantage of this is that perhaps other modules can use this `raw` format output too. For example, in my request #647 the solution can be a simple `-P pesdump [pid]` plugin that extract all content from a pid and pass as raw format. Then with the new raw output the data will be accessible to other tools.

I hope you want to consider this.
Regards.

lelegard, 2020-10-03:
> As you can imagine, here the HTTP plugin with the parameter --raw-output will pass any data, even if it's or not a valid TS packet. 

Why reinventing `curl`?

lars18th, 2020-10-03:
> > As you can imagine, here the HTTP plugin with the parameter --raw-output will pass any data, even if it's or not a valid TS packet.
> 
> Why reinventing `curl`?

You're right! And `curl URL | tsresync -c | tsp ...` does the trick. Even `wget -qO- URL | tsresync -c | tsp ...`

Perhaps too much complited to implement it inside TSDuck. However, the only missing point is that the "curl" tool is Unix native and not Windows. But we can use some port.

Regards.

lelegard, 2020-10-03:
curl exists on Windows, typically as part of Cygwin and Mingw.
Honestly, it is impossible to live with the old MSDOS console, so using Cygwin is a basic requirement of Windows.
Yes, PowerShell is great but not for everything.