tsduck/tsduck #640: Incorrect video bitrate after merging 2 SPTS files

PierreDTV, 2020-09-17:
Hi @lelegard,

When merging an SPTS file into a secondary SPTS file with enough stuffing (26Mb/s), using TSP merge plugin, the video bitrate of the merged stream is getting impacted and get the value of the stuffing bitrate of the main stream. 

So the merge result is an MPTS file without stuffing and a very high video bitrate for the merged service. 

To be noted that the service of the main SPTS is not impacted by this issue in the MPTS output stream.

Any idea?

Thank you for your help.

Pierre

lelegard, 2020-09-17:
Hi,

The plugin `merge` just merges streams. It is not a MUX. It does no bitrate synchronisation. The packets which are received from the forked process are merged immediately in available null packets. So, if the two streams are regulated real-time streams, the merge is more or less acceptable (if you receiver is tolerant). But if the forked process generates packets at CPU or disk file speed, then all null packets are replaced without regulation.

The topic of a MUX in TSDuck has been discussed many times here. A precise MUX is a lot of work, could have been done, I would love to do it, but I have no time. Strangely enough, everyone would like one but no-one has time to develop it 😉

PierreDTV, 2020-09-22:
Hi @lelegard,

Thank you very much for your feedback. So I have followed your advices by adding the `regulate` plugin in forked and main processes and I obtain pretty good results! The merge is now spread over the time with correct bitrate and the outcome stream got some stuffing. I m happy 👍 

`tsp -I file STPS_1.ts -P regulate -P merge "tsp -I file STPS_2.ts -r 2 -P continuity --fix -P zap Service_2 -P regulate" --no-pcr-restamp -O file MPTS.ts`
 
**Note:** in the merge process I have repeated twice STPS_2.ts  in order to fit at least SPTS_1 duration.

When playing MPTS.ts in VLC, Service_2 is playing well but it displays a wrong duration. Service_1 (from SPTS_1) is OK.

In order to solve this issue, I have tried to recompute the PCR either with `pcradjust` plugin or by removing --no-pcr-restamp from `merge` option, it works but on Service_2 seeking takes time and  A/V is not working in the concatenate area (-r 2 ). Service_1 is OK.

If you have again some tips, I will be very interested.

Thank you very much Thierry.

Pierre