tsduck/tsduck #641: DVB-S2: Help with Dektec DTA-107

Pazzeo, 2020-09-17:
Hi all,

I'm trying to modulate a multicast signal generated by external application with tsp using DTA-107.
The external app is generating a multicast 236.1.1.1:2000 70Mbps and using StreamXpert, it looks good.
So I configured tsp to modulate over DVB-S2 with the following commnad:

tsp -v -m --bitrate 70000000 --buffer-size-mb 12 --max-flushed-packets 500 --max-input-packets 500 -I ip 10.20.203.3@236.1.1.1:2000 --ssm -O dektec --modulation "DVB-S2-8PSK" --convolutional-rate "3/4" --lnb 9750 --satellite-frequency 11013000000 --symbol-rate 29900000 --stuffing --fifo-size 71680 --level -200

This is the log in the output

` Warning: ip: address 236.1.1.1:2000 is not an SSM address
* ip: joining multicast group 10.20.203.3@236.1.1.1:2000 from local address 10.20.201.65
* ip: joining multicast group 10.20.203.3@236.1.1.1:2000 from local address 10.20.203.65
* ip: initial input bitrate is 70,000,000 b/s
* dektec: using Dektec device 0, output channel 0 (DTA-107 port 1)
* dektec: setting output TS bitrate to 66,620,894 b/s
* dektec: setting output carrier frequency to 1,263,000,000 Hz
* dektec: setting output fifo size to 71,680 bytes
* dektec: output fifo size: 71,680 bytes, max: 8,388,608 bytes, typical: 8,388,608 bytes
* dektec: initial output bitrate: 66,620,894 b/s
* dektec: Will preload FIFO before starting transmission. Preload FIFO size: 57,344 bytes.
* [MON] 2020/09/17 16:20, resource monitoring started
* dektec: DTA-107 output FIFO load is 57,267 bytes, starting transmission
* [MON] 2020/09/17 16:21, VM:422 MB (stable), CPU:7.69% (average:7.69%)
* [MON] 2020/09/17 16:21, VM:422 MB (stable), CPU:3.30% (average:5.49%)
* [MON] 2020/09/17 16:21, VM:422 MB (stable), CPU:3.40% (average:4.79%)
* [MON] 2020/09/17 16:21, VM:422 MB (stable), CPU:3.30% (average:4.42%)
`

However, I'm not able to see anything in the output of DTA. Using an analyzer I'm seeing that the SNR is null.
Could you help me please?

Pazzeo

rezna11, 2020-09-17:
You could try DtLoop first to determine precise modulation parameters. Seems wrong to me trying to fit 70Mbps input to 8PSK-29900-3/4.

Pazzeo, 2020-09-18:
Hi rezna11,

You right and I change the input bitrate to 64900000. However, I'm not able to see the signal.
The board seems to work because using DtLoop I don't have any problem.
The problem is when I'm using tsp under linux with input ip.

Matteo

rezna11, 2020-09-18:
Why not try 66620894? I've always had the best results with modulators in TSDuck if the input bitrate exactly matched the output, then you don't need any `--stuffing`.

Anyway, what if you try this?

`tsp -v -m --bitrate 66620894 -I null -P regulate -O dektec --modulation "DVB-S2-8PSK" --convolutional-rate "3/4" --lnb 9750 --satellite-frequency 11013000000 --symbol-rate 29900000`

Then you should see signal from the modulator, just to verify if the modulator works properly on your Linux setup. Then you just replace the input and start adding your other parameters one by one to see what breaks it.

rezna11, 2020-09-18:
And also try `--level -270` instead of `--level -200`

lelegard, 2020-09-18:
>  I think it could be a problem of the driver of linux.

Be sure to use consistent Dektec API (linked into TSDuck) and Dektec device drivers. The easiest way is to get the latest versions of each. For Linux Dektec drivers, use this: https://github.com/tsduck/dektec-dkms/releases