tsduck/tsduck #642: Cannot load any plugin

sthfrnth, 2020-09-20:
Hi Thierry,

No plugin can be loaded with version 29f3548b on Linux 64:
```
* Debug: trying to load "/usr/bin/tsplugin_time.so"
* Error: processor plugin time not found
* Debug: trying to load "/usr/bin/tsplugin_timeref.so"
* Error: processor plugin timeref not found
```
It seems this piece of code has a bug:
```
    // Load a shared library if not found and allowed.
    if (it == plugin_map.end() && _sharedLibraryAllowed) {
        // Load shareable library. Use name resolution. Use permanent mapping to keep
        // the shareable image in memory after returning from this function. Also make
        // sure to include the plugin's directory in the shared library search path:
        // an extension may install a library in the same directory as the plugin.
        ApplicationSharedLibrary shlib(plugin_name, u"tsplugin_", TS_PLUGINS_PATH, SharedLibraryFlags::PERMANENT, report);
        if (shlib.isLoaded()) {
            // Search again if the shareable library was loaded.
            // The shareable library is supposed to register its plugins on initialization.
            it = plugin_map.find(plugin_name);
        }
        else {
            report.error(shlib.errorMessage());
        }
    }

    // Return the factory function if one was found.
    if (it != plugin_map.end()) {
        assert(it->second != nullptr);
        return it->second;
    }
    else {
        report.error(u"%s plugin %s not found", {plugin_type, plugin_name});
        return nullptr;
    }
```
plugin_map is not changed after loading the library. So the iterator will always be plugin_map.end() and complains plugin not found.

I still don't know what is the root cause. It takes some time to read through the code. I think you can immediately know what's wrong.

Thanks.

lelegard, 2020-09-20:
> It seems this piece of code has a bug: [...] So the iterator will always be plugin_map.end() and complains plugin not found.

The code looks good to me. You probably have missed this:
~~~
        if (shlib.isLoaded()) {
            // Search again if the shareable library was loaded.
            // The shareable library is supposed to register its plugins on initialization.
            it = plugin_map.find(plugin_name);
        }
~~~
After loading the shared library, the plugin is searched again in the repository and the iterator is indeed modified with the loaded plugin. The idea is that there is an internal repository of plugins so that you don't need to look for a shared library all the time but only once. Simple builtin plugins are self-registered from the beginning and never dynamically loaded.

To be clear, such a bug (failing to load plugins) has no chance to appear without being detected on push. The TSDuck test system ([described here](https://tsduck.io/doxy/testing.html)) is maybe not perfect but has proven to catch most potential regression. Each time a bunch of commits is pushed on GitHub, the [continuous integration](https://github.com/tsduck/tsduck/blob/master/.github/workflows/continuous-integration.yml) task is triggered. It rebuilds TSDuck and runs all test suites on Linux, macOS and Window. Since the test suite invokes `tsp` many times, such a bug would be immediately caught. You can [see here](https://github.com/tsduck/tsduck/actions/runs/263060726) the report of the continuous integration for the commit you refer to. There is no error.

Your problem is probably due to some inconsistency in your file system, a mixture of recent and old versions of TSDuck.

Since you use a fresh commit from the repo, you must have rebuilt it yourself. Then, there are two possibilities. First option, you run it directly from the build area. Using the same version as you, here is what I see on Ubuntu 64 bits.
~~~
$ . build/setenv.sh 
$ tsp -d -I null -P time -O drop
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d
* Debug: debug level set to 1
* Debug: registering processor plugin "time", status: ok
* Debug: null: debug level set to 1
* Debug: null: ====> tsp -I null 
* Debug: drop: debug level set to 1
* Debug: drop: ====> tsp -O drop 
* Debug: trying to load "/home/thierry/tsduck/bin/release-x86_64-vmubuntu/tsplugin_time.so"
* Debug: time: debug level set to 1
~~~
You can see that the plugin is immediately found in the build area `/home/thierry/tsduck/bin/release-x86_64-vmubuntu`, the same directory as the `tsp` executable.

Second option, you install your build (after `make install` or `make rpm` or `make deb`) and then you use TSDuck as usual. After installing my build, I can see this:
~~~
$ tsp -d -I null -P time -O drop
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d
* Debug: debug level set to 1
* Debug: null: debug level set to 1
* Debug: null: ====> tsp -I null 
* Debug: drop: debug level set to 1
* Debug: drop: ====> tsp -O drop 
* Debug: trying to load "/usr/bin/tsplugin_time.so"
* Debug: /usr/bin/tsplugin_time.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/bin/time.so"
* Debug: /usr/bin/time.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/tsplugin_time.so"
* Debug: /usr/lib64/tsduck/tsplugin_time.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/time.so"
* Debug: /usr/lib64/tsduck/time.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsplugin_time.so"
* Debug: /usr/lib64/tsplugin_time.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/time.so"
* Debug: /usr/lib64/time.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/local/lib64/tsduck/tsplugin_time.so"
* Debug: /usr/local/lib64/tsduck/tsplugin_time.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/local/lib64/tsduck/time.so"
* Debug: /usr/local/lib64/tsduck/time.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_time.so"
* Debug: registering processor plugin "time", status: ok
~~~
You can see that the plugin is found in the standard plugin location: `/usr/lib/tsduck/tsplugin_time.so`

There is a long list of possibilities which are searched in order to accommodate test versions, legacy installation, Windows, Linux, macOS, user plugins, varieties of 64-bit Linux (`lib64` on Fedora, `lib` on Ubuntu for instance). But the list is carefully crafted to activate the right plugin in the right configuration.

I am sorry you spent so much time on trying to find a bug in the code. But, in case of weird problem like this one, it is usually faster to cleanup everything and then rebuild (`make distclean; make -j10`). The `-j10` option is just here to speed up the build. You may adjust according to your machine resources.

sthfrnth, 2020-09-21:
Hi Thierry,

Thank you very much for the explanation.

I made the mistake:
The plugin_map (referencing to _processorPlugins) was actually modified outside the getFactory function when loading the plugin library.

And thanks to your hint, I found out the issue was indeed caused by mixed versions. Below is what happened on my machine:

```
sudo dpkg -i tsduck_3.20-1554_amd64.deb
tsp -d -I null -P time -O drop
# ok here
git reset --hard 29f3548b && git clean -fdx
CC=clang CXX=clang++ make -j16
sudo make install
tsp -d -I null -P time -O drop
# now fails
...
* Debug: trying to load "/usr/bin/tsplugin_time.so"
* Error: processor plugin time not found
...
sudo apt -y remove tsduck
# issue fixed here
sudo make install
tsp -d -I null -P time -O drop
# ok again
```

Still not very clear on the details but they are not important now.

Thank you again for your help and the decent tool.

closed 2020-09-21 by sthfrnth