tsduck/tsduck #643: UDP Double Bitrate When monitor it

Morgan0328, 2020-09-22:
Hi , i'm try to using tsp monitor , but when running the monitor , my bit rate will be double

what is the problem of my command?

[image]

lelegard, 2020-09-22:
Your command looks good. It simply displays that the TS bitrate is approximately 6.1 Mb/s all the time.

What is clear is that your command just receives multicast packets from the network. It does not send any packet. So, it will not change the stream on the network.

What you see is a obviously a measurement artifact.
- What is this diagram?
- How do you measure the bitrate? Which tool? With machine? Same as sender, receiver, other?
- What is the correlation between the start/stop of the peak and your tsp command?

Morgan0328, 2020-09-23:
> Your command looks good. It simply displays that the TS bitrate is approximately 6.1 Mb/s all the time.
>
> What is clear is that your command just receives multicast packets from the network. It does [...]

I measure the bitrate by StreamXpert , sender and receiver at same Network but different machine , both of then are Server machine and Window10 1803 . 

[image]

Not all udp source will be dobule bitrate after tsp monitor , i try three udp source (same software to send ) just one of them will be  double.

lelegard, 2020-09-23:
>  i try three udp source (same software to send ) just one of them will be double.

Do you mean that you try three different sources, that you run tsp bitrate monitor on each of them and you notice a double bitrate on only one of them?

Again, your tsp command does not send any packet, it just receives them. The only symptom that tsp exhibits on the network in your case is a multicast group add membership request for the specified group. So, you have to investigate how, in your network configuration, such an IGMP request may have the observed effect.

Do your three sources run from the same sender machine? Simultaneously or one after the other? On which machine does StreamXpert run? Same as sender? Same as tsp? How many other receivers do you have on the same LAN for each multicast group? What are the various multicast addresses and UDP ports?

Morgan0328, 2020-09-29:
> Do you mean that you try three different sources, that you run tsp bitrate monitor on each of them and you notice a double bitrate on only one of them?
>
> Again, your tsp command does not send any [...]

i use four machine  , three sender machine and one receiver (Run StreamXpert ) , just one double bitrate of them ,  

it work after i change the sender machine  , it seem to be network setting problem , thanks you

D3nis, 2020-11-02:
I was passing by here...

@Morgan0328 
Maybe You have another multicasts in the network with the same IP, but with different port.
For example You monitor 235.0.101.1:5678, but You have MC like 235.0.101.1:5679 . So in this case when You subscribe first, you will receive both streams => double bitrate.