tsduck/tsduck #644: EIT table does not inject

chetan-prime, 2020-09-28:
I have requirement to transcode through ffmpeg a stream that has EIT inside it.
Since ffmpeg destroys the EIT, I wanted to mux it back in though tsduck at the output phase-

This is how I extract the EIT tables from source in real time -
`tsp -v -I http http://source/mpeg.ts -P tables --include-next --pack-all-sections --pid 18 --rewrite-binary --binary /tmp/output-eit.bin -O drop`

This is the transcode command -
`tsp -a 1/5 -I http http://source/mpeg.ts -P continuity --fix | \
ffmpeg -i - -f mpegts .... /dev/stdout | \
tsp -v --realtime -P inject --bitrate 30000 --pid 18 --binary --poll-files /tmp/output-eit.bin=100 -O fork 'curl -X POST -T - http://127.0.0.1/publish/output'
`

I'm injecting more than enough stuffing packets when reading the source, from what I can understand of the docs, the "inject" plugin should be able to mux in the EIT tables, but when I analyze the output the EIT tables are missing.

|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C       24,530 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C       24,530 b/s  |
|  0x0000  PAT .......................................... C       20,301 b/s  |
|  0x0011  SDT/BAT ...................................... C        4,229 b/s  |

There is no entry for 0x12

I can confirm that /tmp/output-eit.bin has data and is updated every 1-2 seconds. If I use tstabdump on the file I can see EIT info on the screen.
But for some reason, the inject plugin isn't able to inject the data into the output as expected.

lelegard, 2020-09-28:
Hi @chetan-prime,

I understand the idea but the method that you are using is complicated and not accurate. It will create problems even after tuning it. If you want to process a TS through ffmpeg while preserving EIT's, there is a much simpler and straightforward method: encapsulate the EIT's as a private data PID in the service and tell ffmpeg to copy this data stream without processing. Thus, encapsulated EIT's pass through ffmpeg without modification. After ffmpeg output, decapsulate this data PID and restore EIT's.

The plugin `encap` encapsulates one or more PID's into one "tunnel" PID. There are several encapsulation modes, include a "PES mode" which builds a PES stream containing the encapsulated data (the PES format in `encap` is a contribution from @lars18th). The following complete example is adapted from the simpler example in the `encap` documentation, in the TSDuck user's guide (please RTFM).

What you want to do can be achieved in one long command below. You have to adapt it to your stream. See notes afterwards.
~~~
tsp --add-input-stuffing 1/10 \
    -I .... \
    -P encap --pes-mode fixed --pid 0x12 --output-pid 0x900 \
    -P pmt --add-pid 0x900/0x06 --add-pid-registration 0x900/0x4B4C5641 \
    -P filter --negate --pid 0x1FFF | \
ffmpeg -f mpegts -i - \
    -map 0:d -codec:d copy \
    -map 0:v -codec:v ... \
    -map 0:a -codec:a ... \
    -mpegts_start_pid 0x900 \
    -mpegts_transport_stream_id 0x0004 \
    -mpegts_service_id 0x0407 \
    -metadata service_name="Arte" \
    -f mpegts - | \
tsp -P decap --pid 0x900 \
    -P pmt --remove-pid 0x900 \
    -P filter --negate --pid 0x1FFF \
    -O file arte-out.ts
~~~

Steps:
- Initial `tsp` command:
  - Add one null packet every 10 input packets to accomodate the encapsulation overhead.
  - Encapsulate the PID 0x12 (EIT's) in PES mode, creating a "tunnel" PID 0x900. Make sure that this PID number is not already used in your stream.
  - Reference this PID in the PMT of the service with stream type 0x06 (PES private data) and registration type "KLVA" (0x4B4C5641). Thus, the encapsulated EIT's become a data component of the service.
  - Remove remaining null packets.
- `ffmpeg` command:
  - Map the first "data" stream `0:d` to output and use `copy` codec (ie. no transformation of the PES stream). Thus, the PES stream containing the encapsulated EIT's will pass unmodified.
  - Then, map video and audio PID's `0:v` and `0:a` with their respective codecs (your choice).
  - Important: Be sure to map the data stream first to enforce it as first output PID. This way, you can specify exactly the first PID using `-mpegts_start_pid` just afterwards.
  - Respecify the original TS id and service id using `-mpegts_transport_stream_id` and `-mpegts_service_id`. This is important to preserve the references of your EIT's. By default, ffmpeg would generate TS id 1 and service id 1 which would not match with the EIT's. You may also set the service name using `-metadata service_name`. Here, all values are from my stream. You must adapt to yours.
- Final `tsp` command:
  - Decapsulate the "tunnel" PID. You have forced it in ffmpeg as PID 0x900 using `-mpegts_start_pid`.  
  - Dereference the PID 0x900 from the PMT. The tunnel PID disappears after decapsulation and it is cleaner to dereference it.
  - Remove spurious null packets which were created from encapsulation overhead.

The result is a TS with transcoded audio and video and with preserved original EIT's.

chetan-prime, 2020-09-28:
Thanks for the detailed & clear answer. It looks like the right way to do it.
I had no idea this was kind of pid tunnelling was possible with tsduck. Will try on my end and confirm if it works

lelegard, 2020-09-28:
> I had no idea this was kind of pid tunnelling was possible with tsduck. 

It can be useful to browse the user's guide at least once to have an overview of the available tools and plugins.

Since this example is rather educational, I just added it in the example chapter of the user's guide.

chetan-prime, 2020-09-29:
I can confirm this works.
As you suggested, I just need to change the transport id & stream id so it maps to the same as source & I'm set.
A completely different issue - since ffmpeg seems to have no way to get the network id & service id I found myself using tsanalyze to automate this process. It would be great if analyze plugin had JSON output, but the current normalized format is also usable

closed 2020-09-29 by chetan-prime