tsduck/tsduck #645: Different Output in PESPacket from tsduck Demuxer vs ffmpeg demuxer

Reinok, 2020-09-29:
Hello,

I am sending TS Packets into the tsduck PESDemuxer and extracting the AVC video by streamId. I am sending the data in these packets into Nvidia's GPU h264 decoder. I was able to get it working with ffmpeg per Nvidia's examples, but want to use tsduck. The headers of the packets are the same, but tsduck PESDemux outputs different data. The ffmpeg data can be decoded, but the tsduck PESPacket data content cannot be. Is there some transformation I'm missing for the tsduck output?

lelegard, 2020-09-29:
Hi @Reinok 

It is hard to tell without knowing what your code is doing with the PES packets which are sent by PESDemux.

>  The headers of the packets are the same, but tsduck PESDemux outputs different data.

What is the difference between what ffmpeg is doing? Same number of PES packets with same headers but different payload? How different?

Reinok, 2020-09-29:
I have not checked the number of packets, but it was different as of 10 or so bytes in, so the first PESPacket is different.

lelegard, 2020-09-29:
> I have not checked the number of packets, but it was different as of 10 or so bytes in, so the first PESPacket is different.

If you want to understand the core issue, you should provide a piece of TS (producing at least one PES packet), the ffmpeg output and your application output. This is the only way to understand the difference.

oviano, 2020-12-14:
I’m somewhat interested in this topic. Currently I only use TSDuck for the actual grabbing of tuner TS data and I pass the data straight to a hacked/modified libavformat to demux. The sort of hack I needed was to make it possible to av_format_find_stream_info skip over certain streams that it cannot handle (EPG, MPEG Private streams, etc) so that demux is quicker and doesn’t timeout waiting to figure out the stream info for streams it can’t help.

Anyway, my point is I’d quite like to drop the avformat dependency and demux using TSDuck as the code is way nicer. The FFmpeg libraries are comprehensive but some of that code is old and it shows!

So I wonder if there real working examples of demuxing using TSDuck and then either passing the PES data to hardware decoders as @Reinok is doing, or indeed passing directly to libavcodec. 

In short, is TSDuck being used anywhere for demuxing and passing PES data to other decode libraries? Does the PESDemux gather enough data about codecs, etc. I wouldn’t want to go down this road if it then turned out that TSDuck doesn’t identify enough codec parameters to initialise a decoder, because it had never been used in this manner.

lelegard, 2020-12-15:
The class `PESPacket` provides the contextual information for a PES packet: PID, PCR, stream type, codec. This information must be set by the environment which builds the PES packet. In addition to extracting PES packets, the class `PESDemux` also analyzes the signalization (PAT and PMT's) to collect information on PES PID's, typically stream type and codec, based on the PMT content, including the descriptors in the PMT components. To do that, the `PESDemux` must receive the complete TS. All demux classes have a set of PID's to demux. They do the filtering on those PID's to demux but they also need the complete TS: the method `feedPacket()` must be called for all packets.

So, if your class implements `PESHandlerInterface`, it receives PES packets one by one and each packet carries its codec info. Just take care that the first few PES packets may not be correctly identified if received before the first PMT for their service.

In addition to getting the complete PES packets, the `PESHandlerInterface` can receive various info such as audio/video attributes or individual NALunits. You may experiment by cloning the `pes` plugin and playing with the received information.

Take care that HEVC info was very recently added. It is not on version 3.24. Use the latest repo if you need HEVC info.

oviano, 2020-12-16:
Great, thanks for the info!