tsduck/tsduck #646: Segfault when trying to parse malformed PES packet

vs-adm, 2020-10-02:
Hello,

PES packet with incorrect length set in header causig segfaults in `tsp`. Example command which is fail

```
tsp -I file slice-3.ts -P pes --packet-index --header -O drop
```

Sample fragment is in [slice-3.ts.zip](https://github.com/tsduck/tsduck/files/5315588/slice-3.ts.zip), the first packet is the problem one.

It should be unbounded PES packet with length of zero. Please consider add error message to output instead of crashing.

Thanks.

lelegard, 2020-10-02:
Hi @vs-adm,

Thank you for reporting this and taking the time to extract a short stream which reproduces the problem.

I am terribly sorry about that. I usually take great care of the handling of invalid input. Probably because of my security background, I am particularly aware of that type of danger. But this requires a constant care which may unfortunately depends on the daily mood...

In your specific case, the PES length field contained 2. With stream_id = 0xE0 (video, long PES header format), PES_packet_length = 2, PES_header_data_length = 5, the PES packet was shorter than its header. While the case of a PES packet larger than its actual size was correctly handled, the case of a PES packet shorter than its header was incorrectly handled. Later, the PES payload was computed as packet size - header size, resulting in integer overflow, using a infinitely long payload, ultimately reading invalid memory when exploring the payload.

This was just fixed in commit edc3e39abb8503398796cc47469883e5661b0b3c and a non-regression [test-072](https://github.com/tsduck/tsduck-test/blob/master/tests/test-072.sh) was added to make sure it won't reappear.

Such PES packet is now considered invalid and ignored.

You may pull the repo and rebuild. Or wait until tomorrow to get the next [nighly build](https://tsduck.io/download/prerelease/), commit -2011.

vs-adm, 2020-10-02:
Thank you for lighting fast response, @lelegard,

actually, just ignore such packets will make diagnostics of such packets harder. The only way to determine exact broken packet for me was the segfault I've mentioned in description.

Having some kind of descriptive warning in `pes` plugin or `tsdump` would be great. Please consider adding one. Thank you again.

lelegard, 2020-10-02:
Currently, TSDuck is not really designed to report incorrectly formed PES packets. Incorrect PES packets, as well as incorrect sections (incorrect size or CRC), are simply dropped. 

In `tsdump`, TS packets are dumped one by one. Since a PES packet or a section usually extends over future packets, it is not possible to analyze the PES packet or section at that time. I agree that the specific issue of a PES length shorter than the PES header can be detected in the first TS packet most of the time. But not incorrectly large length values. So, it does not make sense to report "too short" errors while never reporting "too large" errors. 

The `pes` plugin only analyzes full PES packets when available. Extracting PES packets is a lower level generic PES demux which can be used in many different applications without message reporting capability. The same is true with the generic section demux which is used virtually everywhere in TSDuck. The best they can do is to filter and safely ignore incorrect data (which failed in the case of this bug).

vs-adm, 2020-10-02:
This makes sense, thanks for detailed explanation.

closed 2020-10-24 by lelegard