tsduck/tsduck #647: Request: pesdump or ts2pes

lars18th, 2020-10-02:
Hi @lelegard ,

I know you don't want to reimplement the wheel. However, I wan to extract the PES content of some packets, and I prefer to use only tsduck. So, please can you implement some tool like a pesdump or ts2pes?
In opencaster is this https://github.com/nexgenta/opencaster/blob/master/tools/ts2pes/ts2pes.c
but this tool is now quite old, and not multiplatform.

You think it has sense?

lelegard, 2020-10-03:
The simplest way could be some `--output-pes-file` in plugin `pes` to save all selected PES packets.

What should be the output format? Just raw PES packets one after the other?

lars18th, 2020-10-03:
> The simplest way could be some `--output-pes-file` in plugin `pes` to save all selected PES packets.

OK.

> What should be the output format? Just raw PES packets one after the other?

Yes, just RAW.
But, one questions: It will be possible to pipe the output? It will be useful something like `tsp ... -P pes -p 1000 --output-pes-file stdout | pes_analyzer`. You will support this?

Anyway, perhaps you want to add  _some_ simple enhancements: For example, `--initial-offset` to cut the first N bytes of the PES packet (note the PES packet, not the TS packet). And `--pes-limit` to limit the size of the ouput to initial chunks. And perhaps `--ts-offset` to cut too the first X bytes of each TS packet (however, I feel this will be useful only for some user encapsulations).

Thank you @lelegard to consider this suggestion. :wink:

lelegard, 2020-10-04:
Added option:
~~~
--save-pes filename

Save all PES packets in the specified file. All PES packets are saved in a raw binary form without encapsulation.
When the specified file is '-', the standard output is used.
~~~

lars18th, 2020-10-05:
Hi @lelegard ,

Thank you! I'll try soon. :+1: 

But one question: When using the standard output, we need to use `tsp ... -P pes -p 1000 --save-pes - -O drop | pes_analyzer` or this will work `tsp ... -P pes -p 1000 --save-pes - | pes_analyzer` ?

lelegard, 2020-10-06:
> `tsp ... -P pes -p 1000 --save-pes - -O drop | pes_analyzer`

Here, the PES packets are written on stdout. The output plugin drops TS packets. So, if no other plugin writes on stdout, that will work. Note that log messages (those starting with `* ...`) are written on stderr and do not interfere with binary data on stdout.

> `tsp ... -P pes -p 1000 --save-pes - | pes_analyzer`

Here, the PES packets are written on stdout. The output plugin is by default ` file` which, by default, writes the TS packets on stdout. So, the output pipe is a binary mixture of PES packets and TS packets. Will not work as expected.

lars18th, 2020-10-06:
Thank you @lelegard !

I hope then that you update the documentation to list one "correct" example like
 `tsp ... -P pes -p 1000 --save-pes - -O drop | pes_analyzer`. 

Regards.

lars18th, 2020-11-25:
Hi @lelegard ,

I want to close this issue, but before it I suggest to consider to add _one_ example inside the documentation. If you search for `save-pes` inside the PDF only one ocurrence is found:
```
--save-pes filename
    Save all PES packets in the specified file. All PES packets are saved in a raw binary form without
    encapsulation.
    When the specified file is '-', the standard output is used.
```

And without an exemple like this `tsp ... -P pes -p 1000 --save-pes - -O drop | pes_analyzer` is quite difficult to see than the content of a PID (the PES bitstream) can be extracted using pipes in a chain.

I hope you want to accept this suggestion.
Thank you for this impresive project!
Regards.

lelegard, 2020-11-25:
Done, example added.

lars18th, 2020-11-25:
> Done, example added.

Checked. So I close this issue.
Thank you! :+1:

closed 2020-11-25 by lars18th