tsduck/tsduck #648: How to read video resolution from stream ES info

Jazzykool85, 2020-10-05:
Can you please help me to find a suitable plugin that can dump the stream ES information including the source stream  video resolution information

Jazzykool85, 2020-10-05:
I have found a utility that could give me the below result w.r.t to the ES inside a stream --> tspsi
Using this i got the below info, How can we identify the video resolution from this information?
    Elementary stream: type 0x1B (AVC video), PID: 401 (0x0191)
    - Descriptor 0: Stream Identifier (0x52, 82), 1 bytes
      Component tag: 1 (0x01)
    - Descriptor 1: Video Stream (0x02, 2), 3 bytes
      Multiple frame rate: false, frame rate: 0x03 (25 fps)
      MPEG-1 only: false, constained parameter: true, still picture: false
      Profile and level: 0x44 (68)
      Chroma format: 0x01 (4:2:0)
      Frame rate extension: false
    - Descriptor 2: Data Stream Alignment (0x06, 6), 1 bytes
      Alignment type: 0x02 (2, Video access unit)
    - Descriptor 3: Maximum Bitrate (0x0E, 14), 3 bytes
      Maximum bitrate: 0x000044CD (17613), 7,045,200 bits/second
    - Descriptor 4: AVC Video (0x28, 40), 4 bytes
      Profile IDC: 100, level IDC: 40
      Constraint set0: false, set1: false, set2: false, AVC compatible flags: 0x00
      Still pictures: false, 24-hour pictures: false

lelegard, 2020-10-05:
The resolution of a clear AVC / H.264 or MPEG2 video stream is reported by `tsanalyze` and its plugin counterpart `analyze`. The output can be either in a human-readable grid format or reproducible for automation (JSON or proprietary "normalized" format). 

There is currently no solution for HEVC / H.265.

If you are only interested in this information, regardless of other TSDuck features, the free software "MediaInfo" can be used.

Jazzykool85, 2020-10-05:
Thanks Lelegard,
I could do this, Currently we do not have HEVC stream .But only issue is my stream is not clear. I hope then there is no solution for encrypted i guess

lelegard, 2020-10-05:
The most reliable way to get the video resolution is to parse the video stream to fetch that information. This is what `tsanalyze` does. But the video stream must not be encrypted of course.

For encrypted streams, you may sometimes find some metadata information in some descriptors of the PMT (as you see with `tspsi`). But the video resolution is often not present.

In short and in practice, if your video streams are encrypted, there is usually no way to determine the resolution.

Jazzykool85, 2020-10-05:
Yeh thats correct ,
We will try to deduce type of video in this case from service type and the encoding format.
Thanks a lot 👍

closed 2020-10-05 by Jazzykool85