tsduck/tsduck #649: encap and subsequent decap lose all metadata from PMT for a stream

chetan-prime, 2020-10-05:
For my use case, I have a source with Video, Audio, Teletext & Subtitle streams.
Normally I am able to process this just fine with ffmpeg, but even though I use-
`tsp -I http http://192.168.0.161:8000/play/a003 -P continuity --fix`
I still get "non-monotonous" errors on the subtitle streams which are copied raw by the downstream ffmpeg. So my solution was to encap the subtitle, teletext streams -
`tsp -I http http://192.168.0.161:8000/play/a003 -P continuity --fix -P encap --pes-mode fixed --pid 0x1982 --pid 0x1983 --output-pid 0x900 -P pmt --add-pid 0x900/0x06 --add-pid-registration 0x900/0x4B4C5641 --remove-pid 0x1982 --remove-pid 0x1983 -P filter --negate --pid 0x1FFF --pid 0x1982 --pid 0x1983`
where Streams 0x1983 & 0x1982 are DVB teletext, subtitle streams
Downstream, ffmpeg transcodes the video & copies the raw KLVA tunnel stream, another copy of tsp does a "decap" filter to restore the two PID's for DVB teletext & subtitle streams.
Firstly the output is orphaned from the ffmpeg program so I added the following parameters after decap -
`-P decap --pid 0x900 -P pmt --remove-pid 0x900 --add-pid 0x1982/0x0006 --add-pid-registration 0x1982/0x0006 --add-pid 0x1983/0x0006 --add-pid-registration 0x1983/0x0006 --add-stream-identifier`

The teletext stream now detects in ffplay/ffmpeg and shows as part of the main mpegts program. VLC can't see the teletext stream. All the metadata like language code, hearing impaired is lost.
Is there any way with tsduck to bypass ffmpeg for selected streams, but without losing all the stream metadata from PMT?

lelegard, 2020-10-06:
ffmpeg only processes audio and video. When the output is mpegts, ffmpeg recreates a new PAT and PMT without remembering anything from the original input (which may not even be in TS format anyway).

> Is there any way with tsduck to bypass ffmpeg for selected streams, but without losing all the stream metadata from PMT?

You may add the PMT PID inside the encapsulated tunnel (it can contain more than one PID). Since the PMT "disappears" in the tunnel, recreate a new one using `-P pmt --create ...` with only the video and audio PID. The recreated PMT is only used by ffmpeg to locate the input audio and video. On output, just reuse the audio and video PID from ffmpeg, drop the ffmpeg-generated simple PMT (remove its PID) and restore the original PMT from the tunnel. Or course, take care of PID numbering for audio, video and PMT.

lelegard, 2020-10-06:
Also, since you encapsulate a synchronized component of the input, there is no guarantee that output time stamps are synchronized with the input ones. As a consequence, there is a risk that the output subtitles are present and correctly referenced but no longer synchronized with the audio and video. In which case they may simply not appear when selected in VLC.

chetan-prime, 2020-10-06:
Thanks, so there is no easy solution due to the nature of how mpegts works.
Is there a reason why tsduck doesn't correct the teletext stream DTS/PTS when I use the below command -
`tsp -I http http://your.ip:8000/play/stream -P continuity --fix`

Downstream ffmpeg keeps complaining about the teletext stream when muxing it into ouput with a stream copy-
> Non-monotonous DTS in output stream 0:3; previous: 11781766281, current: 7297857795; changing to 11781766282

I thought the continuity filter was supposed to take care of such discontinuity in all streams within an mpegts source

lelegard, 2020-10-06:
> Is there a reason why tsduck doesn't correct the teletext stream DTS/PTS when I use the below command -
> `tsp -I http http://your.ip:8000/play/stream -P continuity --fix`

The plugin `continuity` analyzes or fixes the _continuity counters_, a transport-level mechanism, not DTS/PTS (a content-level mechanism) which were originally generated by the encoder to synchronize with audio and video. The fact that the DTS/PTS are not monotonous is maybe not quite clean but not a problem in itself as long as the audio and video follow the same synchronization. The issue probably is what I predicted: after passing through ffmpeg, the unmodified teletext stream is no longer synchronous with the reencoded audio and video. There is no simple solution to this.

If this is a one-shot file, you may first extract the teletext in SRT format using the plugin `teletext` and then provide the SRT file as subtitles to ffmpeg. But this is not practicable in the general streaming case.

lars18th, 2020-10-07:
Hi @chetan-prime and @lelegard ,

When developing some part of the TSDuck encap/decap plugins for FFMpeg support, I see a lot of times the "non-monotonous" errors. And yes, they appears a lot of times with teletext and other non audio/video streams. So, let me to add some comments to clarify this issue:

- The FFMpeg tool supports teletext streams. Even it can't process them (as it can do with subtitles), it can detect and pass them.
- However, the problem with teletext inside mpeg-ts streams using the FFMpeg is with the TIMESTAMPS included in the TXT stream. Teletext streams are synchronized with the A/V content (for example, for TXT-subtitles). Therefore PTS marks are included in the TS header of some TXT packets. And for this reason FFMpeg supports this type of streams, as they need to take care of these timestamps. So the recomendation is to inject the TXT stream as plain in the FFMpeg to maintain the synchronization. In ideal conditions, FFMpeg can pass (copy) all teletext packets and remux them in the output with the rest of the A/V streams without troubles. So, you can process/reencode the audio/video streams and maintain the teletext.
- But, the problem with FFMpeg is a lack of a robust processing of PTS marks in data streams when using MPEG-TS containers. Based on my experience, a lot of tools that mux TXT streams in A/V services using mpeg-ts put the TXT packets very close in time to the PCR packets. And the FFmpeg doesn't take care of this fact. And generates these non-monotonous errors because the behaviour it uses to solve the timestamps troubles. When a new TXT packet is comming, if the PTS value is greater that the _internal_ PCR used by the FFMpeg, then the PTS mark of the TXT packet is updated with a small increment to compensate the difference. The problem, is when the next TXT with a PTS mark is readed. As the TXT packets are very close, the new PTS is now less than the new calculated PTS of the previous packet, and _this_ raises the error. As you can see, if someone fixes this inside FFMpeg the problem will go away. But at time, the problem is still unsolved.

To _overcome_ this problem different strategies can be used. One option is to encapsulate the teletext stream and deencapsulate it after the FFMpeg processing. The problem then is that the synchronization is lost _if_ you don't instruct the FFMpeg process to maintain the PCR values. But this, from my experience, generates other side effects; so I don't recommend it. Another alternative is to modify the PTS values of the teletext streams before injecting to the FFMpeg. With this solution the FFMpeg can pass (copy) the TXT packts from the input to the output without complaints. However, in this case you need _after_ to recompose the PTS marks of the teletext stream in the resulting TS. You can use for this task the plugin I developed: #347 . At time this plugin is not accepted because a strong reason (that I accept) inside TSDuck. But I hope @lelegard will want to provide a similar alternative to advance/regress the PTS/DTS marks of a PID.

Regards.

chetan-prime, 2020-10-07:
Thanks for the option @lars18th .
I found this patch https://ffmpeg.org/pipermail/ffmpeg-devel/2018-March/226721.html that fixes the buggy ffmpeg CLI code for DTS handling. I'm running now with the patch enabled and seeing if it can handle sudden DTS resets, so far its been running fine for a while & teletext works on the output streams.
I will confirm in day, will try the filter if that patch isn't stable

lars18th, 2020-10-07:
Hi @chetan-prime ,

> I will confirm in day,

Please, put here your results. And if this FFMpeg patch solves the problem, then please open a new issue in the FFMpeg trac service to point to the solution of this _very old_ reported problem. I hope this really solves the problem.

Anyway, I suggest to @lelegard to consider the implementation of a simple PTS/DTS advance/regress plugin. It will be very useful for several scenarios where multiple streams require to be resynchronized (aka retimestamped without remuxing).

Regards.

lars18th, 2020-10-07:
Hi @chetan-prime ,

I see that one person ( @jjustman ) working in ATSC 3.0 has created a repo of FFMpeg with only this patch applied:
https://github.com/jjustman/ffmpeg-hls-pts-discontinuity-reclock

Perhaps it will be interesting to do some push to accept this patch.
Regards.

Note: I apologize to @lelegard for this out-of-scope message.  I hope you can understand the reason for publishing it here. Thank you.

chetan-prime, 2020-10-10:
I can confirm that the ffmpeg patch seems to fix the issue. I had previously adapted the patch from github & got it to work upto vanilla ffmpeg 4.3 with no changes. There are some minor side effects of the patch, but those have no connection with the core issue.
As @lelegard indicated, it might be best to deal with this at the application level as we can have insight on DTP/PTS levels of other streams muxed in along with the failing one.

closed 2020-10-10 by chetan-prime

lars18th, 2020-10-13:
Hi @chetan-prime ,

> I can confirm that the ffmpeg patch seems to fix the issue. I had previously adapted the patch from github & got it to work upto vanilla ffmpeg 4.3 with no changes. There are some minor side effects of the patch, but those have no connection with the core issue.

Please, can you describe the side effects?

And, please @lelegard I apologize for continuing to talk about this topic here, but I found it relevant to anyone that need to use the TSDuck encapsulation with the FFmpeg tool.

lelegard, 2020-10-13:
> And, please @lelegard I apologize for continuing to talk about this topic here,

You're welcome, this is a free discussion forum about TSDuck and its use cases (and how to improve them as you do in this thread).