tsduck/tsduck #650: Missing ECMs

zaleksov-verimatrix, 2020-10-09:
Hello,
I'm trying to scramble a service using a real ECMG (input ts file with clear service, output ts file). I can see that the ECMG connection is ok and when I enable logging in the scrambler I can see that ECMs are being received, but they are not present in the output stream. I used tsanalyze to check the output stream, I can see that the ECM PID  is there but with 0 b/s:

> b/s:``|     PID  Usage                                     Access          Bitrate  |
|   Total  Digital television service ................... S    3,633,186 b/s  |
|  0x0061  PMT .......................................... C       15,027 b/s  |
|  0x0062  ECM ............................... C            0 b/s  |
|  0x01FF  MPEG-2 Video ................................. S    3,419,966 b/s  |
|  0x0200  MPEG-1 Audio (deu, clean effects) ............ S      198,192 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                 

I also tried to do a binary search in the output ts file for the ECMs (printed in the log) but no luck. 
Does somebody has an idea why the ECMs are not included in to the output stream?

Thanks in advance.
BR
Zoran

lelegard, 2020-10-10:
Hi @zaleksov-verimatrix 

The most common cause for requested data not being inserted is a lack of stuffing or "full TS". In your TS analysis, have a look at the "Global PID's" section, looking for stuffing, e.g.:
~~~
===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 68,430, PID's: 8 (clear: 8, scrambled: 0)                      |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C    1,677,034 b/s  |
|  0x0000  PAT .......................................... C        3,774 b/s  |
|  0x0001  CAT .......................................... C        7,523 b/s  |
   ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~
|  0x1FFF  Stuffing ..................................... C    1,348,097 b/s  |
|=============================================================================|
~~~

If there is no stuffing, then your TS is "full", just like a full file system where nothing can be inserted. If your stream comes from DVB broadcast, you will most likely get some stuffing. But if your stream is an IP-TV one (likely at VMX...), there is probably none.

Be sure to read the reference section of `tsp` in the [TSDuck user's guide](https://tsduck.io/download/docs/tsduck.pdf). The "Troubleshooting" section 6.2, reads as follow:
> __A plugin which is supposed to insert some data in the stream does not insert anything__
>
> The tsp command processes a transport stream. It can modify or delete existing packets but it cannot create new packets. “Inserting packets” essentially means replacing null packets (aka. stuffing) from the transport stream with new data. When there is not enough or no null packets, no data can be inserted.
>
> Solution: Make sure that enough input stuffing exist. The more appropriate way to do this depends on the context. See the tsp reference documentation in chapter 3, especially the section named “modifying, inserting and deleting packets”.

The easiest way to check if this is the cause for your problem is to initially "inflate" your TS with artificial stuffing (10% in the following example), let `tsp` plugins insert what they have to insert and remove the extra unused stuffing just before output using the following command skeleton:
~~~
tsp --add-input-stuffing 1/10 \
    -I ... \
    .... \
    -P scrambler ... \
    .... \
    -P filter --negate --pid 0x1FFF \
    -O ...
~~~

zaleksov-verimatrix, 2020-10-11:
Hi @lelegard,

Thanks a lot for the prompt and detailed answer.

Makes sense what you say, but on the other hand the same stream worked with the tsecmg... That made me believe that it should also work with the real ECMG. Regarding stuffing it is true, there is not much:
 0x1FFF  Stuffing ..................................... C        3,492 b/s

The stream is indeed DVB, we also have DVB in VMX :).

I'll try to play with the plugin and see how far I get.

BR
Zoran

lelegard, 2020-10-11:
Well, 3 kb/s stuffing is not much since traditional ECM streams use 15 or 30 kb/s depending on ECM size (at 10 ECM/s). But since there is some stuffing, ECM's should at least eat it and replace it with ECM's, maybe not as fast as intended but at least up to the available stuffing.

Do you still have stuffing on output (ie. after scrambling and potential ECM insertion)?

Could you post your complete `tsp` command? 

And BTW, if you work on DVB at VMX, please pass my best wishes to Mulham :)

zaleksov-verimatrix, 2020-10-11:
This stuffing is after the scrambling, also the report above is from the same scrambled stream. 

tsp -v -I file -i  /mnt/shared/recorded-200.ts -P scrambler 200 -e 10.28.0.45:11600 -s 0xAAAA0000 -a 0009000200cf -i 0x1200 --log-data=verbose --log-protocol=verbose  -O file test.ts

zaleksov-verimatrix, 2020-10-11:
One question: IS there a log that I can enable to see when an ECM is dropped because of full ts?

zaleksov-verimatrix, 2020-10-11:
I think that I know where the problem is... I originally had a recorded stream with 6 services, 1 clear and 5 scrambled. Since I was only interested in the clear one I've decided to filter it by using this command: 
tsp -v -I file /mnt/shared/recorded.ts -P zap -n 200 -O file /mnt/shared/recorded-200.ts

With this the stuffing was gone completely :) 

I've just tried to scramble the clear service from the original stream and I can see that there are some ECMs added.

lelegard, 2020-10-11:
> With this the stuffing was gone completely :)

Definitely, the `zap` plugin removes everything (including stuffing) but the target service. On DVB streams where you want to insert something but there is not enough stuffing, I usually replace a service with stuffing : `-P svremove --stuffing`. Thus, the original input bitrate is preserved (useful to reinject the modified stream in a modulator) and the TS has more free space.

> tsp -v -I file 

Use option `-P scrambler --synchronous` (RTFM). By default, the `scrambler` plugin assumes a live stream and continues to process packets in real time while an ECM is being generated. This is fine for real time streams to avoid being blocked during the generation of an ECM. But, for offline sources, a huge amount of packets are processed at input speed (disk speed) while the ECM is being generated. Usually, more than one crypto-period is processed while the ECM is generated and the ECM is then useless when it comes back. So, with offline and unregulated sources (ie. "infinite" input speed), the option `--synchronous` suspends the processing while the ECM is generated to make sure the ECM is ready for insertion before the start of the crypto-period.

fcartegnie, 2025-08-22:
I've noticed that generated ECM sections are cleared out when filtering out artificial stuffing

> -P filter --negate --input-stuffing

Seems the clearing the stuffing metadata flag isn't done when replacing null packet

`ts::ScramblerPlugin::CryptoPeriod::getNextECMPacket`

lelegard, 2025-08-22:
@fcartegnie, this is not very clear. 

> I've noticed that generated ECM sections are cleared out when filtering out artificial stuffing
~~~
  --input-stuffing
      Select packets which were articially inserted as stuffing before the
      input plugin (using tsp options --add-start-stuffing,
      --add-input-stuffing and --add-stop-stuffing). Be aware that these
      packets may no longer be null packets if some previous plugin injected
      data, replacing stuffing.
~~~

This means that `--input-stuffing` filters packets which were originally inserted as artificial stuffing. This marks a position in the stream, not a packet content. If the packet at that position has been replaced by something else, it will still be filtered by this option.

Generally speaking, if you use artificial stuffing _and_ data insertion, using this command may remove part or all of what you inserted.

> Seems clearing the stuffing metadata flag isn't done when replacing null packet

The method `ts::ScramblerPlugin::CryptoPeriod::getNextECMPacket` replaces the content of a packet with some generated ECM content, if available, it does not remove the input stuffing meta-data flag.

This "flag" is set in the metadata of the packet. It is not based on the content of the packet. This flag is updated by the method `ts::TSPacketMetadata::setInputStuffing()`. Looking at all references to this method, you will see that this flag is "set" by a few low-level stuff which generate stuffing (additional packet positions which were not part of the original input stream). However, there not a single line of code which "unsets" this flag. Once the "artificial input stuffing" flag is set, it stays set until the packet is spit out by the output plugin. It is technically possible to reset it using `ts::TSPacketMetadata::setInputStuffing()` but no current code in TSDuck does this.

lelegard, 2025-08-22:
@fcartegnie, if you want to remove stuffing, whether it was artificially inserted or part of the original input stream, use `-P file --negate --pid 0x1FFF`.

fcartegnie, 2025-08-22:
> This "flag" is set in the metadata of the packet. It is not based on the content of the packet. This flag is updated by the method `ts::TSPacketMetadata::setInputStuffing()`. Looking at all [...]

Yes, but clearly that's something that is not wanted. There should be a way to remove all input stuffing that has not been used, and keep intact the existing stuffing. On custom plugins I already clear the flag.
I believe that's a minor change on the filter plugin as additional or combining option for --stuffing. (I did not find a way to perform full boolean logic on the filter to remove --padding only when --pid isn't null)

lelegard, 2025-08-22:
> Yes, but clearly that's something that is not wanted. There should be a way to remove all input stuffing that has not been used, and keep intact the existing stuffing.

Do you have a precise use case to support this?

In my view, stuffing is only about CBR vs. VBR. Stuffing exists to fill CBR streams (typically broadcast). So, unless I miss a specific use case, you have two main cases:

1. CBR: You must keep the bitrate intact. So, you shall never add artificial stuffing. The stream must have some initial stuffing. If it is insufficient, then you need to remove something, a service, a PID, whatever, an replace it with stuffing (all plugins which remove things have a `--stuffing` option to do this).
2. VBR: You don't care about bitrate, the stream has probably no initial stuffing, and if it has some, it is useless. You have the freedom to add a lot of artificial stuffing. A lot means more opportunity to insert packets at the right time. And at the end of the processing, just remove all remaining stuffing.

Your use case is neither CBR nor VBR.

1. It is not CBR because the bitrate is changed. You want to keep some artificial stuffing that was replaced but keep the original one.
2. It is not VBR because stuffing is useless and expendable in VBR. In that case, you can remove all remaining stuffing regardless of its origin, artificial or original. And if the stream needs to be reprocessed in some downstream equipment, you can use the same method again: add artificial stuffing, insert data in stuffing, remove all stuffing.

> I did not find a way to perform full boolean logic on the filter to remove --padding only when --pid isn't null

You may try this (the label number is arbitrary):
~~~
-P filter --input-stuffing --set-label 27 -P filter --only-label 27 --negate --pid 0x1FFF
~~~
I haven't tried, but it should work.

fcartegnie, 2025-08-22:
> Your use case is neither CBR nor VBR.

The use case is in having a generic pipeline to perform encryption but with minimal impact on the bitrate shape.
Sources have unknown padding until ingestion starts.

> ```
> -P filter --input-stuffing --set-label 27 -P filter --only-label 27 --negate --pid 0x1FFF
> ```
> 
> I haven't tried, but it should work.

okay, thanks will try that workaround

lelegard, 2025-08-22:
> okay, thanks will try that workaround

That is not a workaround. This is the intended way it should work. There are zillions of possible specific use cases. It is not possible to add a specific option for each of them. TSDuck was designed in the original UNIX philosophy (or ffmpeg): provide a given set of basic tools and the possibility to combine them in any possible way. We add features only when a given scenario cannot be achieved with the current set of tools. And even in that case, we don't necessarily add the implementation of that scenario, we just add the basic tools which were missing to implement it.