tsduck/tsduck #651: Glitches in audio playblack after some hours of merging streams.

salvaaod, 2020-10-13:
Hi, first of all thanks for this great software, I have been using it over the last few months and is fantastic.

The issue I have  is as follows, I am merging two DVB ts udp streams, one 1.7 Mbps MPTS with 6 audio (radio) programs of around 256Kbps each and another SPTS 5Mbps with one video and audio program. The output is fed to a IP to ASI converter that in turn is fed to a hardware ASI multiplexer that do all the MUX CBR and DVB tables insertion, the output of the ASI multiplexer is fed to a DTT transmiter.

The command I use is as follows: **tsp --bitrate 7500000 -I null -P regulate -P merge "tsp  -I ip 238.0.0.1:2381"  -P merge  "tsp -I ip 238.0.0.2:2382" -O ip 192.168.3.204:3001 -e** being the first ts the audio MPTS and the second the video SPTS, 

Once I start the process all works ok and the radios and the video channel is received correctly by all DTT receivers, tested with a TR101 290 analyzer there are no errors.

After a few hours all the audio radio channels from the first ts have small interruptions with the audio being squeezed in the playback spaces, there is no audio lost, only small interruptions of around 100mS every 500mS, a command restart fixes everything. When that happens there are no TR101 290 errors, so, the muxing process seems to be fine as far as packets is concerned.

The thing that is leaving me more perplexed is that this issue happens only in a few receivers, I have tested around 20 different receivers and all are ok except 5, so, whatever happens seems to be a combination of an issue within my setup and particular receivers.

I have tried most options in within the merge plugin the same issue happens regardless.

Any idea what can be happening and how to fix it?  I am not sure whether this is a bug  a tsduck limitation or something odd in my setup.

I can provide ts samples or log outputs if that can help.

Thanks for your help.

rezna11, 2020-10-13:
Did you also check TR101 290 priority 2 - PCR repetition and PCR accuracy?
You can also try DVB Inspector's PCR/PTS analysis to see if there's a problem.

salvaaod, 2020-10-13:
> Did you also check TR101 290 priority 2 - PCR repetition and PCR accuracy?
> You can also try DVB Inspector's PCR/PTS analysis to see if there's a problem.

Thanks for your suggestions, Yes, PCR is within 40-50ms for all the PIDs with the audio programs, PCR accuracy does not seems to be an issue either, actually I don't see any difference for PCR or PCR accuracy either when the muxing is ok or when is generating this issue.

I have not checked about the PCR/PTS , I will have a go, again what's baffling me is that some receivers (most) are perfectly ok and a few not.

salvaaod, 2020-10-13:
> You can also try DVB Inspector's PCR/PTS analysis to see if there's a problem.

Thanks, your suggestion seems to have been spot on, so, what seems the issue is as follows, when the muxing command is started all PTS timestamps happens before the PCR time, so, the audio is correctly played, after a few hours the PCR slips and the PTS timestamps happen after the PCR time hence the glitches in audio play.

Seems that me merge plugin drifts the PCR after a few hours, any  suggestion to improve this ?

Attached to pictures of the DVB Inspector analysys:

Mux Command just started with PTS happening before PCR
[image]

Mux Command after a few hours with PTS happening after PCR with glitches in audio play.
[image]

Thanks for your help,

rezna11, 2020-10-13:
Sorry I don't have any suggestion to fix this. I myself wonder if this would be considered a bug in TSDuck or falls into "TSDuck is not a mux" category :) Let's wait for @lelegard's opinion...

salvaaod, 2020-10-14:
> Sorry I don't have any suggestion to fix this. I myself wonder if this would be considered a bug in TSDuck or falls into "TSDuck is not a mux" category :) Let's wait for @lelegard's opinion...

Thanks, at least I am happy now since I have understood the core issue here and I can do some different configuration options in  the mux/merging order to see if I can overcome the issue. My feeling is that there must be some timing issue at hand that accumulates overtime since it works ok for a few hours, I will do more testing in order to see if the issue happens at a moment noticed or is just a slight drift that accumulates over time. I can confirm that the issue not only happens with Audio programs but with video as well, seems that the video decoders are a bit more tolerant. I put an image below of the video issue, in the same way restarting the process all is ok. Let's wait for lelegard comments.

On another hand, I am testing the PCR to PTS/DTS offset with DVBInspector but this is an offline tool and is a bit time consuming, any suggestion on tools to do this that can receive a udp/rtp stream ?

Issue with video to PCR offset:
[image]

lelegard, 2020-10-14:
>  I myself wonder if this would be considered a bug in TSDuck or falls into "TSDuck is not a mux" category :) Let's wait for @lelegard's opinion...

Well, it is quite tempting to send this to the "TSDuck is not a mux" category :)

But I can try to improve the PCR fixing a little bit. I took this opportunity to do some cleanup on the `merge` plugin code but the real "fix" [is here](https://github.com/tsduck/tsduck/blob/master/src/tsplugins/tsplugin_merge.cpp#L542). Please let us know if this improves the PCR drift.

salvaaod, 2020-10-14:
> Well, it is quite tempting to send this to the "TSDuck is not a mux" category :)
>
> But I can try to improve the PCR fixing a little bit. I took this opportunity to do some cleanup on the `merge` [...]

Thanks @lelegard , I will give a try and report, Is fix this included in the last nightly  Pre-release version 3.24-2027 build ?

lelegard, 2020-10-14:
You need version -2028 or higher, so you will have to wait until tomorrow. Or you can recompile it yourself. If you are using Linux or macOS, this is quite straightforward. On Windows, the compilation takes a lot more time but is equally easy once you have setup the proper environment (just double-click on a PowerShell script).

lelegard, 2020-10-14:
I also applied some modifications on the plugin `merge` to smoothen the insertion of the merged packet. It has been reported for a long time that merged packets had a tendency to burst. A few options have been added to tune this.

salvaaod, 2020-10-16:
> But I can try to improve the PCR fixing a little bit. I took this opportunity to do some cleanup on the `merge` plugin code but the real "fix" [is here](https://github.com/tsduck/tsduck/blob/master/src/tsplugins/tsplugin_merge.cpp#L542). Please let us know if this improves the PCR drift.

Thanks @lelegard I tried for a couple of days now, it has improved somewhat but the issue is still there, with very precise CBR streams, MPTS 2Mbps audio programs and 5Mbps video program it still dritts, after something like 7 or 8 hours the PTS falls below the PCR with the audio programs, I understand the difficulties of having a precise PCR adjustment.

Will it be possible to have a option in the merge plugin like "pcr_adjust_restart_time" so that after a set number of minutes or hours the PCR adjust will be reset ? This will somewhat circumvent the issue. Right now I am restarting the merging process every few hours to achieve a similar result but much less clean.

Other than being sure that the input streams are CBR suggestion  to avoid PCR drift ? I don't know if I can do something in the underlying host system to improve.

Thanks for your help.

salvaaod, 2020-10-16:
> Will it be possible to have a option in the merge plugin like "pcr_adjust_restart_time" so that after a set number of minutes or hours the PCR adjust will be reset ? This will somewhat circumvent the issue. 
> 

May be another option will be just to reset the PCR adjust for a particular PID if the PTS/DTS falls behind PCR.

lelegard, 2020-10-17:
> May be another option will be just to have the option to reset the PCR adjust for a particular PID if the PTS/DTS falls behind PCR. like --pcr-reset-backwards

Probably the best (or "least bad") solution. Will do.

lelegard, 2020-10-17:
I have added option `--pcr-reset-backwards`. Since I cannot reproduce your case, please test it. When the PCR restamping is reset, a verbose message is printed. So, using `tsp --verbose --timed-log`, you can figure out how often this happens. Please let us know the result.

salvaaod, 2020-10-18:
> I have added option `--pcr-reset-backwards`. Since I cannot reproduce your case, please test it. When the PCR restamping is reset, a verbose message is printed. So, using `tsp --verbose --timed-log`, you can figure out how often this happens. Please let us know the result.

Thanks @lelegard  it seems to to work for me, the command I am using is this:  /usr/bin/tsp  -v  --timed-log   --bitrate 8000000 -I null -P regulate -P merge --pcr-reset-backwards "/usr/bin/tsp   -v  --timed-log  -I ip 238.0.0.1:2381"   -P merge --pcr-reset-backwards  "/usr/bin/tsp   -v  --timed-log  -I ip 238.0.0.2:2382" -O ip 192.168.3.204:3001 -e  after 7 hours the PTS of some PIDs has follen behind the PCR but nothing happened, you can get a sample of the full mux here, one of the PIDs with the issue is the PID 84
https://www.dropbox.com/s/iqgb8xu2np6evsm/mux%20tsduck%20despues%207-30.ts?dl=0

The log output seems ok:

Oct 18 09:17:37 multiplexer tsp[1263]: * 2020/10/18 09:17:37 - regulate: minimum packet burst duration is 10,000,000 nano-seconds
Oct 18 09:17:37 multiplexer tsp[1263]: * 2020/10/18 09:17:37 - null: initial input bitrate is 8,000,000 b/s
Oct 18 09:17:37 multiplexer tsp[1263]: * 2020/10/18 09:17:37 - regulate: regulated at bitrate 8,000,000 b/s
Oct 18 09:17:37 multiplexer tsp[1283]: * 2020/10/18 09:17:37
Oct 18 09:17:37 multiplexer tsp[1284]: *
Oct 18 09:17:37 multiplexer tsp[1283]:  - ip: joining multicast group 238.0.0.1:2381 from local address 192.168.3.208
Oct 18 09:17:37 multiplexer tsp[1284]: 2020/10/18 09:17:37 - ip: joining multicast group 238.0.0.2:2382 from local address 192.168.3.208
Oct 18 09:17:37 multiplexer tsp[1283]: * 2020/10/18 09:17:37 - ip: unknown initial input bitrate
Oct 18 09:17:37 multiplexer tsp[1284]: * 2020/10/18 09:17:37 - ip: unknown initial input bitrate

No further messages regarding tsp.

_Any idea why the PCR  is not resetting when PTS < PCR?_

There is the a sample of the original audio stream being merged with correct 84 PID

https://www.dropbox.com/s/azd7t9f1sysan7u/radios.ts?dl=0

Thanks for your help.

salvaaod, 2020-10-18:
Well, after a few more hours seems that it has done something, maybe it was at the TV channel PIDs, but since the logs don't say anything I am  not sure, the radio channels are all the same, all the PTS are behind the PCR and the PCR adjust reset seems not to have any effect, below are the outputs of the log:

Oct 18 09:17:37 multiplexer tsp[1263]: * 2020/10/18 09:17:37 - regulate: regulated at bitrate 8,000,000 b/s
Oct 18 09:17:37 multiplexer tsp[1283]: * 2020/10/18 09:17:37
Oct 18 09:17:37 multiplexer tsp[1284]: *
Oct 18 09:17:37 multiplexer tsp[1283]:  - ip: joining multicast group 238.0.0.1:2381 from local address 192.168.3.208
Oct 18 09:17:37 multiplexer tsp[1284]: 2020/10/18 09:17:37 - ip: joining multicast group 238.0.0.2:2382 from local address 192.168.3.208
Oct 18 09:17:37 multiplexer tsp[1283]: * 2020/10/18 09:17:37 - ip: unknown initial input bitrate
Oct 18 09:17:37 multiplexer tsp[1284]: * 2020/10/18 09:17:37 - ip: unknown initial input bitrate
Oct 18 21:07:04 multiplexer tsp[1263]: * 2020/10/18 21:07:04 - merge: resetting PCR restamping after DTS/PTS moved backwards restamped PCR
Oct 18 21:07:04 multiplexer tsp[1263]: message repeated 8 times: [ * 2020/10/18 21:07:04 - merge: resetting PCR restamping after DTS/PTS moved backwards restamped PCR]
Oct 18 21:07:05 multiplexer tsp[1263]: * 2020/10/18 21:07:05 - merge: resetting PCR restamping after DTS/PTS moved backwards restamped PCR
Oct 18 21:07:05 multiplexer tsp[1263]: message repeated 16 times: [ * 2020/10/18 21:07:05 - merge: resetting PCR restamping after DTS/PTS moved backwards restamped PCR]
Oct 18 21:07:06 multiplexer tsp[1263]: * 2020/10/18 21:07:06 - merge: resetting PCR restamping after DTS/PTS moved backwards restamped PCR
Oct 18 21:07:06 multiplexer tsp[1263]: message repeated 5 times: [ * 2020/10/18 21:07:06 - merge: resetting PCR restamping after DTS/PTS moved backwards restamped PCR]

lelegard, 2020-10-19:
Hi @salvaaod,

Thanks for providing the streams. I understand why it does not work now. This is due to some unusual (although perfectly legal) encoding in your stream.

In the last modification, I assumed that the PCR and PTS/DTS were in the same TS packet. Usually (although not required), the video PID of a TV service (and audio of a radio one) has a PCR at the beginning of each (or most) PES packet. So, in the same TS packet, you have the PCR, DTS and PTS. In my last modification, the PCR vs. PTS/DTS check was made locally only, when they are in the same single TS packet.

In your streams, the video PID's of the TV services have PCR's in random TS packets, not TS packets with a starting PES:
~~~
$ tsp -I file mux-tsduck-despues-7-30.ts -P filter -p 0x0709 -P filter --pcr -P until --packet 1 | tsdump

* Packet 0
  ---- TS Header ----
  PID: 1801 (0x0709), header size: 12, sync: 0x47
  Error: 0, unit start: 0, priority: 0
  Scrambling: 0, continuity counter: 15
  Adaptation field: yes (8 bytes), payload: yes (176 bytes)
  Discontinuity: 0, random access: 0, ES priority: 0
  PCR: 0x1F6A3983975
  ---- Full TS Packet Content ----
  47 07 09 3F 07 10 D6 75 96 48 7E B5 2B 7A F6 E7 A5 22 43 A2 72 8B 82 67
  20 38 61 FD CC 6B AE 8D BD C8 09 13 2E 01 4F AE 64 B9 5C 0F 92 03 A3 60
  85 E1 E7 BB D0 EF DB 63 67 85 B7 DF C5 1B C9 6D 2C 8D 5E C1 DE 86 09 DE
  93 87 8C F9 A3 61 4F 84 10 55 86 5B 40 5D 60 7B 94 6E 08 F2 D5 72 63 59
  80 06 E8 69 75 47 D1 81 54 5C 78 FC 02 81 71 50 9A C4 C6 44 C9 B5 34 95
  E1 A1 58 2D 84 28 64 54 72 B7 D1 63 2D A9 15 D8 B2 D0 C2 EE 4E 25 31 70
  E1 F2 F1 BD A1 9C 87 7E D4 0C D6 EE 88 B6 5F D3 90 F4 73 2E B5 F0 F8 22
  2A 21 EA 8B D7 02 95 DF 30 3C 52 F5 A1 2F DD 15 C4 E5 6D FC
~~~

Maybe, at some point, the PCR was randomly at the same TS packet as the start of a PES packet and this explains why you saw it once only.

In your radio services, the PCR are in the audio PID but in independent packets without payload. They are apparently inserted independently of the audio packetization:
~~~
$ tsp -I file mux-tsduck-despues-7-30.ts -P filter -p 0x0054 -P filter --pcr -P until --packet 1 | tsdump

* Packet 0
  ---- TS Header ----
  PID: 84 (0x0054), header size: 188, sync: 0x47
  Error: 0, unit start: 0, priority: 0
  Scrambling: 0, continuity counter: 3
  Adaptation field: yes (184 bytes), payload: no (0 bytes)
  Discontinuity: 0, random access: 0, ES priority: 0
  PCR: 0x1C538EB62E2
  ---- Full TS Packet Content ----
  47 00 54 23 B7 10 C1 5F F7 34 7F 02 FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
~~~

I will make the modification more general, keeping more context. I will also include the more complicated case where the PCR PID is completely independent from the video and audio PID (it happens).

salvaaod, 2020-10-19:
Thanks @lelegard I will test once it is done, may be you can select a minimum distance between PCR and PTS/DTS and if that distance is crossed then the PCR  adjust will reset rather than let go to 0, PCR=DTS/PTS . I am not sure if is it legal that the PCR time will be the same  than the PTS/DTS, I have looked into the TR101.290 and seen nothing obvious about minimum and maximum distances between PCR to DTS/PTS

lelegard, 2020-10-19:
@salvaaod, in commit ab130865f6e45779a7cc963e66600b54e4c92eaa, I have improved the detection of late PCR's. Now, if the PCR becomes ahead of _any_ PTS or DTS in _any_ PID which uses that PCR (according to PCR PID in PMT), the restamping is reset. If the PTS/DTS is in another previous packet, the PTS/DTS value to compare is adjusted according to the bitrate and packet distance.

I haven't included any minimum difference between the PCR and DTS for now. I just test that the extrapolated PTS/DTS is strictly greater than the PCR.

Again, this code hasn't been tested. So, I cannot guarantee it 100%. Please test it. The verbose messages are still there in case of restamping reset.

salvaaod, 2020-10-20:
Hi, I just leaved working overnight, it works, but it does something weird, after it stars resetting the PCR seems to get in to a PCR resetting loop, , it takes around 5 hours to reset the radio (audio) PCR PIDS (80, 81, 82, 83, 84, 85) for the first time, but after that it continues resetting it every few minutes or secons, I expected the reset to happen after 5 hours again but not so quickly and repeatedly. With the Video PCR PID (68) seems more normal, it resets after eigth and a half hours and it stays there.

I will copy the log output down below, Any way to improve this ? I provided the original audio encoder ts in a message before just in case you needed to check it again, if you need any other ts sample or info let me know so that you can understand the issue.

Thanks

Oct 20 01:16:21 multiplexer tsp[14481]: * 2020/10/20 01:16:21 - regulate: minimum packet burst duration is 10,000,000 nano-seconds
Oct 20 01:16:21 multiplexer tsp[14481]: * 2020/10/20 01:16:21 - null: initial input bitrate is 8,000,000 b/s
Oct 20 01:16:21 multiplexer tsp[14481]: * 2020/10/20 01:16:21 - regulate: regulated at bitrate 8,000,000 b/s
Oct 20 01:16:21 multiplexer tsp[14501]: *
Oct 20 01:16:21 multiplexer tsp[14500]: * 2020/10/20 01:16:21 - ip: joining multicast group 238.0.0.2:2382 from local address 192.168.3.208
Oct 20 01:16:21 multiplexer tsp[14501]: 2020/10/20 01:16:21 - ip: joining multicast group 238.0.0.1:2381 from local address 192.168.3.208
Oct 20 01:16:21 multiplexer tsp[14501]: * 2020/10/20 01:16:21 - ip: unknown initial input bitrate
Oct 20 01:16:21 multiplexer tsp[14500]: * 2020/10/20 01:16:21 - ip: unknown initial input bitrate
Oct 20 05:14:40 multiplexer tsp[14481]: * 2020/10/20 05:14:40 - merge: resetting PCR restamping in PID 0x0055 (85) after DTS/PTS moved backwards restamped PCR
Oct 20 05:27:12 multiplexer tsp[14481]: * 2020/10/20 05:27:12 - merge: resetting PCR restamping in PID 0x0054 (84) after DTS/PTS moved backwards restamped PCR
Oct 20 05:31:21 multiplexer tsp[14481]: * 2020/10/20 05:31:21 - merge: resetting PCR restamping in PID 0x0051 (81) after DTS/PTS moved backwards restamped PCR
Oct 20 05:31:37 multiplexer tsp[14481]: * 2020/10/20 05:31:37 - merge: resetting PCR restamping in PID 0x0050 (80) after DTS/PTS moved backwards restamped PCR
Oct 20 05:32:02 multiplexer tsp[14481]: * 2020/10/20 05:32:02 - merge: resetting PCR restamping in PID 0x0053 (83) after DTS/PTS moved backwards restamped PCR
Oct 20 05:32:17 multiplexer tsp[14481]: * 2020/10/20 05:32:17 - merge: resetting PCR restamping in PID 0x0052 (82) after DTS/PTS moved backwards restamped PCR
Oct 20 05:38:49 multiplexer tsp[14481]: * 2020/10/20 05:38:49 - merge: resetting PCR restamping in PID 0x0054 (84) after DTS/PTS moved backwards restamped PCR
Oct 20 05:40:18 multiplexer tsp[14481]: * 2020/10/20 05:40:18 - merge: resetting PCR restamping in PID 0x0050 (80) after DTS/PTS moved backwards restamped PCR
Oct 20 05:41:03 multiplexer tsp[14481]: * 2020/10/20 05:41:03 - merge: resetting PCR restamping in PID 0x0053 (83) after DTS/PTS moved backwards restamped PCR
Oct 20 05:43:33 multiplexer tsp[14481]: * 2020/10/20 05:43:33 - merge: resetting PCR restamping in PID 0x0055 (85) after DTS/PTS moved backwards restamped PCR
Oct 20 05:49:24 multiplexer tsp[14481]: * 2020/10/20 05:49:24 - merge: resetting PCR restamping in PID 0x0053 (83) after DTS/PTS moved backwards restamped PCR
Oct 20 05:54:42 multiplexer tsp[14481]: * 2020/10/20 05:54:42 - merge: resetting PCR restamping in PID 0x0051 (81) after DTS/PTS moved backwards restamped PCR
Oct 20 05:58:01 multiplexer tsp[14481]: * 2020/10/20 05:58:01 - merge: resetting PCR restamping in PID 0x0050 (80) after DTS/PTS moved backwards restamped PCR
Oct 20 06:01:57 multiplexer tsp[14481]: * 2020/10/20 06:01:57 - merge: resetting PCR restamping in PID 0x0052 (82) after DTS/PTS moved backwards restamped PCR
Oct 20 06:02:56 multiplexer tsp[14481]: * 2020/10/20 06:02:56 - merge: resetting PCR restamping in PID 0x0053 (83) after DTS/PTS moved backwards restamped PCR
Oct 20 06:06:22 multiplexer tsp[14481]: * 2020/10/20 06:06:22 - merge: resetting PCR restamping in PID 0x0050 (80) after DTS/PTS moved backwards restamped PCR
Oct 20 06:10:18 multiplexer tsp[14481]: * 2020/10/20 06:10:18 - merge: resetting PCR restamping in PID 0x0052 (82) after DTS/PTS moved backwards restamped PCR
Oct 20 06:10:31 multiplexer tsp[14481]: * 2020/10/20 06:10:31 - merge: resetting PCR restamping in PID 0x0054 (84) after DTS/PTS moved backwards restamped PCR
Oct 20 06:10:50 multiplexer tsp[14481]: * 2020/10/20 06:10:50 - merge: resetting PCR restamping in PID 0x0053 (83) after DTS/PTS moved backwards restamped PCR
Oct 20 06:17:04 multiplexer tsp[14481]: * 2020/10/20 06:17:04 - merge: resetting PCR restamping in PID 0x0052 (82) after DTS/PTS moved backwards restamped PCR
Oct 20 06:23:44 multiplexer tsp[14481]: * 2020/10/20 06:23:44 - merge: resetting PCR restamping in PID 0x0050 (80) after DTS/PTS moved backwards restamped PCR
Oct 20 06:24:37 multiplexer tsp[14481]: * 2020/10/20 06:24:37 - merge: resetting PCR restamping in PID 0x0055 (85) after DTS/PTS moved backwards restamped PCR
Oct 20 06:30:14 multiplexer tsp[14481]: * 2020/10/20 06:30:14 - merge: resetting PCR restamping in PID 0x0051 (81) after DTS/PTS moved backwards restamped PCR
Oct 20 06:56:12 multiplexer tsp[14481]: * 2020/10/20 06:56:12 - merge: resetting PCR restamping in PID 0x0055 (85) after DTS/PTS moved backwards restamped PCR
[... cut ...]
Oct 20 10:09:34 multiplexer tsp[14481]: * 2020/10/20 10:09:34 - merge: resetting PCR restamping in PID 0x0050 (80) after DTS/PTS moved backwards restamped PCR
Oct 20 10:12:29 multiplexer tsp[14481]: * 2020/10/20 10:12:29 - merge: resetting PCR restamping in PID 0x0052 (82) after DTS/PTS moved backwards restamped PCR
Oct 20 10:15:11 multiplexer tsp[14481]: * 2020/10/20 10:15:11 - merge: resetting PCR restamping in PID 0x0054 (84) after DTS/PTS moved backwards restamped PCR
Oct 20 10:23:32 multiplexer tsp[14481]: * 2020/10/20 10:23:32 - merge: resetting PCR restamping in PID 0x0055 (85) after DTS/PTS moved backwards restamped PCR

lelegard, 2020-10-20:
Hi @salvaaod 

I think I have an idea. This is maybe due to the burst effect in the `merge` plugin. This plugin is known to not smoothen the inserted packets (a recent update should have but apparently didn't). So, the inter-packet distance in a merged PID can suffer from high variations.

In the PCR restamping reset strategy, when a packet has a PCR but no PTS/DTS, we use the _previous_ PTS/DTS and extrapolate what it should be at the PCR packet, given the TS bitrate. And if the actual or extrapolated PTS/DTS is behind the PCR, we report an anomaly and reset the PCR restamping.

When the PCR and PTS/DTS are in the same packet, the PCR < PTS/DTS comparison is always valid. But in your streams the PTS/DTS are never in the same packet as PTS/DTS. Consequently, the PTS/DTS is always extrapolated. The question is: is that extrapolation valid? If the packet insertion bursts, the inter-packet distance can be much smaller or much larger than what it should be. Specifically, if the distance between the PTS/DTS packet and the next PCR packet is much smaller than expected, the extrapolated PTS/DTS at the PCR packet is incorrectly small, possibly falling behind the PCR, resetting the restamping.

To exhibit this on your streams, I quickly wrote a small plugin named `pids` to report statistics on PID's (I may include it in TSDuck afterwards with some enhancements). This plugin displays for each PID the inter-distance packet: min, max, mean and standard deviation.

Here are the results on your original radio stream and the merged stream, for the radio PID's only:
~~~
$ tsp -I file radios.ts -P pids -p 80-90 -O drop
pid: 0x0050  total:    4,274  min:   1  max:    44  mean:    6.44  std-dev:    5.03
pid: 0x0051  total:    4,271  min:   1  max:    44  mean:    6.44  std-dev:    4.95
pid: 0x0052  total:    2,408  min:   2  max:    55  mean:   11.41  std-dev:   14.21
pid: 0x0053  total:    4,272  min:   2  max:    45  mean:    6.44  std-dev:    5.13
pid: 0x0054  total:    4,274  min:   2  max:    45  mean:    6.44  std-dev:    5.15
pid: 0x0055  total:    2,399  min:   2  max:    73  mean:   11.47  std-dev:   13.52
$
$ tsp -I file mux-tsduck-despues-7-30.ts -P pids -p 80-90 -O drop
pid: 0x0050  total:    4,834  min:   1  max:  9973  mean:  113.65  std-dev:  827.80
pid: 0x0051  total:    4,835  min:   1  max:  9973  mean:  113.62  std-dev:  828.99
pid: 0x0052  total:    2,722  min:   3  max: 10084  mean:  201.81  std-dev: 1108.96
pid: 0x0053  total:    4,838  min:   3  max:  9972  mean:  113.56  std-dev:  829.05
pid: 0x0054  total:    4,835  min:   1  max: 10020  mean:  113.63  std-dev:  828.09
pid: 0x0055  total:    2,724  min:   1  max:  9991  mean:  201.69  std-dev: 1106.55
~~~

The TS birate ratio between the two streams is 17.7 (35.4 to 2 Mb/s). The mean inter-packet distances of the radio PID's in the merged stream have roughly the same ratio, meaning that the merge is "correct" but _on a gross average only_. The max and standard deviations, however, have literally exploded, showing a very poor smoothing and, as a consequence, the existence of huge bursts.

There are two fields of improvement here:

1. Understand why smoothing is so poor and how to improve it.
2. Change the PCR restamping reset criteria to mitigate the effects of bursts (but having PCR and PTS/DTS in distinct packets doesn't help really).

The good news is that it illustrates (as usual) how fast it is to extend the features of TSDuck with a new plugin (~30 mn, with the help of lots of copy/paste from the `merge` plugin itself, used as a plugin skeleton).

salvaaod, 2020-10-20:
Thanks @lelegard really the muxing process is 1,7 to 8 since the output of the muxed process is fed to a profeesional muxer, I have included the output of the tsduck muxed process without passing through the hardware muxer:

This command: tsp --bitrate 8000000 -I null -P regulate -P merge  "tsp -I ip 238.0.0.1:2381"   -P merge   "tsp   -I ip 238.0.0.2:2382" -O file muxed_only_radios_and_video.ts 

Gives this output
https://www.dropbox.com/s/4fyeye6lq7zmfwy/muxed_only_radios_and_video.ts?dl=0

I have seen that reducing the flush packets seems to reduce the bursts:  tsp --bitrate 8000000 -I null -P regulate -P merge  "tsp --max-flushed-packets 50   -I ip 238.0.0.1:2381"   -P merge   "sp   -I ip 238.0.0.2:2382" -O file muxed_only_radios_and_video_radio_flushed_packets_50.ts     

Gives this output:
https://www.dropbox.com/s/6fxoosv4xy8xym9/muxed_only_radios_and_video_radio_flushed_packets_50.ts?dl=0

In the muxed processes above there is no PSI tables since they are stripped in the muxed process by the merge plugin but in my case it does not matter since they are inserted latter by the hardware muxer along with a couple of programs more.

Just in case I will include again the radios and video program being muxed

radios stream:
https://www.dropbox.com/s/bj4qx57xtr90mnf/radios.ts?dl=0

Video stream:
https://www.dropbox.com/s/ocp47jlc2wr7kxp/video.ts?dl=0

Let's see if this gives a bit more clarity.

Thanks for your help.

lars18th, 2020-10-21:
Hi @lelegard ,

> To exhibit this on your streams, I quickly wrote a small plugin named `pids` to report statistics on PID's (I may include it in TSDuck afterwards with some enhancements). This plugin displays for each PID the inter-distance packet: min, max, mean and standard deviation.

Thank you this enhancement! I requested for something similar time ago. I'm impatient to test it and provide feedback and ideas. Good job! :+1:

> The good news is that it illustrates (as usual) how fast it is to extend the features of TSDuck with a new plugin (~30 mn, with the help of lots of copy/paste from the `merge` plugin itself, used as a plugin skeleton).

Sure! But you have a very deep knowledge of TSDuck that we don't have. :wink: 

> There are two fields of improvement here:
> 
> 1. Understand why smoothing is so poor and how to improve it.

Here is my idea, but this idea goes in confrontation with the basis of one-packet-only processing of the TSDuck. However, I want to share it:

- For a given consecutive N number of TS-packets, the muxing corresponds with:
  - Classify the P used pids in P queues.
  - For each _output_ packet in the chunk of N packets do:
    - Pop the packet from the P queues with the high priority and copy it in the output.
    - (The priority is based in the length between the PTS/DTS of the packet and the calculated PCR of the output packet).
    - Fill with a NULL packet if sufficient output space exists, and no pririoty packets are pending.

So that's the theory, rigth? Then I think the problem with the current muxing inside the TSDuck is that you're only processing the _merged_ pids. My suggestion is to "demux" _all_ pids (from the two sources: main and merged) and "remux" all packets. And of course, it will be necessary to do it in chunks, and not at 1-packet level. Perhaps you can implement this "chunk-processing" with something like a chunk-buffer-plugin. Inside this plugin you can have access to all packets in the buffer, while you maintaing the 1-input-1-output between the rest of the chain. This has sense?

I hope this helps.
Regards.

lelegard, 2020-10-21:
> Perhaps you can implement this "chunk-processing" with something like a chunk-buffer-plugin. Inside this plugin you can have access to all packets in the buffer

At best, a plugin has access to a small slice of the buffer. Each plugin executes in a separate thread and the locking mechanism assigns one slice of the buffer to each plugin. When a chunk of packets is processed, locking mechanisms are used to pass the ownership of this slice of the buffer to the next plugin. [See some details here](https://tsduck.io/doxy/devplugin.html#pdevdesign).

lars18th, 2020-10-21:
> At best, a plugin has access to a small slice of the buffer. Each plugin executes in a separate thread and the locking mechanism assigns one slice of the buffer to each plugin. When a chunk of [...]

Yes, I know it! And for this reason I suggest to add the abstraction of a "chunk-buffer-plugin". This superclass has a _fixed_ internal buffer that it's visible for only this class, and all packets inside it can be modified and altered. The connection with the rest of the TSDuck chain is that 1 incoming packet is copied to the internal buffer and 1 outcoming packet is copied to the chain. And yes, this breaks the non-copy ring-buffer of TSDuck, but I feel it will be useful for some cases where the slice processing is mandatory. So, if you don't abuse of it, only in some very specific cases it will be in use (and reducing the performance that's the cost of use it).

Anyway, the final question is: Can you get a smooth merging functionality without a slice view? If it's not possible, then what you're looking for is impossible, or not?

Regards.

lelegard, 2020-10-23:
I have released the new statistics plugin (finally named `stats` instead of `pids` because it is extended to labels, see example below). Using it is useful to get a direct measurement of the smoothing of packets during the merge.

I have also isolated the smoothing strategy in a separate C++ class to facilitate the evolutions of this specific topic outside the `merge` plugin and to be able to reuse it in other insertion modules.

The result is significantly better than previously. I tried to reproduce something that is as close as possible as your case using the following commands:

~~~
OPTS="--realtime --max-input-packets 16 --max-flushed-packets 16"

tsp -d --bitrate 8000000 $OPTS \
    -I null \
    -P pat --create \
    -P sdt --create \
    -P regulate \
    -P merge --set-label 1 --accel 0 "tsp $OPTS -I file radios.ts --infinite -P regulate" \
    -P merge --set-label 2 --accel 0 "tsp $OPTS -I file video.ts --infinite -P regulate" \
    -P skip 10000 \
    -P until --seconds 30 \
    -P stats -o test.pids.txt \
    -P stats -o test.labels.txt --label 1-2 \
    -O drop
~~~

I essentially replaced your IP input with playing your radio and video files. Everything after the two `merge` plugins is there only for the test and measurement.
- The `skip` plugin is used to exclude the initial stabilisation phase from the stats.
- The audio (lowest bitrate) is inserted first to get a more fluid smoothing (before insertion of the video which takes most of the packets).
- The first `stats` plugin evaluates the inter-packet distance by PID (direct impact on PCR adjustment).
- The second `stats` plugin evaluates the inter-packet distance by merge operation, based on labels, and evaluates the efficiency of the smoothing process.

The results are the following for the merge process (file test.labels.txt):
~~~
          Total nb  ......Inter-packet distance.......
Label   of packets     min     max      mean   std dev
------  ----------  ------  ------  --------  --------
1           39,770       3       5      4.00      0.04
2           99,425       1       5      1.60      0.76
~~~

Considering the very low standard deviation, the smoothing process seems efficient.

Now the PID's (file test.pids.txt), focusing on audio PID's:
~~~
          Total nb  ......Inter-packet distance.......
PID     of packets     min     max      mean   std dev
------  ----------  ------  ------  --------  --------
0x0050       6,176       4     176     25.75     20.08
0x0051       6,172       4     176     25.77     19.79
0x0052       3,478       8     220     45.73     56.88
0x0053       6,173       8     180     25.77     20.58
0x0054       6,176       8     180     25.76     20.68
0x0055       3,467       8     292     45.87     54.08
~~~

Compare this with the previous measurement in a previous post (initial basic version of the stats plugin):
~~~
pid: 0x0050  total:    4,834  min:   1  max:  9973  mean:  113.65  std-dev:  827.80
pid: 0x0051  total:    4,835  min:   1  max:  9973  mean:  113.62  std-dev:  828.99
pid: 0x0052  total:    2,722  min:   3  max: 10084  mean:  201.81  std-dev: 1108.96
pid: 0x0053  total:    4,838  min:   3  max:  9972  mean:  113.56  std-dev:  829.05
pid: 0x0054  total:    4,835  min:   1  max: 10020  mean:  113.63  std-dev:  828.09
pid: 0x0055  total:    2,724  min:   1  max:  9991  mean:  201.69  std-dev: 1106.55
~~~

Again, the standard deviation is much better.

Of course, this is just a test in a virtual machine and your experience may be different. Could you try the latest version?

lars18th, 2020-10-24:
Hi @lelegard ,

Thank you for this new `stats` plugin!

One suggestion: Please, can you add some statistics about PCR and PTS/DTS ? The idea is to provide, for each pid some data like:
- If the pid carries PCR, then print PCR count (in the interval) and the repetition rate, with statistics about drifts.
- If the pid carries PTS/DTS, similar data with statistics.
- If the pid contains PES packets, print the PUSI count.
- And for each pid print the pid type (PSI, PES, etc.) plust the _duplicate packets_ and the _coutinuity errors_.
- Optionally, perhaps some statistics for PSI tables (like repetition intervals) will be interesting.

This will provide a very powerful but simple TS analyzer in realtime! :wink: 
For sure, I don't want to suggest you to implement something complex, but only simple _useful_ information.
You agree?

lelegard, 2020-10-24:
@lars18th, these information are already available from `tsanalyze` and the equivalent plugin `analyze`, either in human-readable format, "grep/sed" normalized format or (recently) JSON. The timestamp drift information is not explicit but can be extracted from the output of the plugin `pcrextract`. This is CSV format which was explicitly designed for later analysis using Excel.

salvaaod, 2020-10-24:
Thanks again  @lelegard  I am going to retest, do I use just the options "--pcr-reset-backwards --accel 0 " in the merge command or do I add "--max-input-packets 16 --max-flushed-packets 16" in the tsp command as well ?

lelegard, 2020-10-24:
I recommend to keep the `$OPTS` as they are. Concerning. `--acceleration 0`, you may try with and without it (RTFM).

salvaaod, 2020-10-25:
Hi @lelegard tested again, the ouput now seems much more smooth, those are the stats for the muxed PIDs

```
              Total nb  ......Inter-packet distance.......
PID     of packets     min     max      mean   std dev
------  ----------  ------  ------  --------  --------
0x0050       5,822       5     991    113.87    128.84
0x0051       5,816       4     871    113.93    124.19
0x0052       3,272       7    1381    202.50    275.28
0x0053       5,818       5     983    113.95    127.65
0x0054       5,816       5     933    113.99    129.01
0x0055       3,268       3    1150    202.70    254.74
0x0709      79,999       1     394      8.29     32.44
0x070A       2,588       1    5914    253.82    973.37
```
The issue that I saw regarding the regular readjustment of the PCR after a few hours is still there:

```
Oct 25 01:00:59 multiplexer tsp[49098]: * 2020/10/25 01:00:59 - regulate: minimum packet burst duration is 10,000,000 nano-seconds
Oct 25 01:00:59 multiplexer tsp[49098]: * 2020/10/25 01:00:59 - null: initial input bitrate is 8,000,000 b/s
Oct 25 01:00:59 multiplexer tsp[49098]: * 2020/10/25 01:00:59 - regulate: regulated at bitrate 8,000,000 b/s
Oct 25 01:00:59 multiplexer tsp[49116]: * 2020/10/25 01:00:59 - ip: joining multicast group 238.0.0.2:2382 from local address 192.168.3.208
Oct 25 01:00:59 multiplexer tsp[49117]: * 2020/10/25 01:00:59 - ip: joining multicast group 238.0.0.1:2381 from local address 192.168.3.208
Oct 25 01:00:59 multiplexer tsp[49098]: * 2020/10/25 01:00:59 - merge: main stream bitrate reset to 8,000,000 b/s (was 0 b/s)
Oct 25 01:00:59 multiplexer tsp[49098]: * 2020/10/25 01:00:59 - merge: main stream bitrate reset to 8,000,000 b/s (was 0 b/s)
Oct 25 08:48:36 multiplexer tsp[49098]: * 2020/10/25 08:48:36 - merge: resetting PCR restamping in PID 0x0052 (82) after DTS/PTS moved backwards restamped PCR
Oct 25 08:50:23 multiplexer tsp[49098]: * 2020/10/25 08:50:23 - merge: resetting PCR restamping in PID 0x0050 (80) after DTS/PTS moved backwards restamped PCR
Oct 25 08:52:25 multiplexer tsp[49098]: * 2020/10/25 08:52:25 - merge: resetting PCR restamping in PID 0x0051 (81) after DTS/PTS moved backwards restamped PCR
Oct 25 08:55:06 multiplexer tsp[49098]: * 2020/10/25 08:55:06 - merge: resetting PCR restamping in PID 0x0055 (85) after DTS/PTS moved backwards restamped PCR
Oct 25 08:57:36 multiplexer tsp[49098]: * 2020/10/25 08:57:36 - merge: resetting PCR restamping in PID 0x0053 (83) after DTS/PTS moved backwards restamped PCR
Oct 25 08:58:35 multiplexer tsp[49098]: * 2020/10/25 08:58:35 - merge: resetting PCR restamping in PID 0x0054 (84) after DTS/PTS moved backwards restamped PCR
Oct 25 09:00:05 multiplexer tsp[49098]: * 2020/10/25 09:00:05 - merge: resetting PCR restamping in PID 0x0050 (80) after DTS/PTS moved backwards restamped PCR
Oct 25 09:01:46 multiplexer tsp[49098]: * 2020/10/25 09:01:46 - merge: resetting PCR restamping in PID 0x0052 (82) after DTS/PTS moved backwards restamped PCR
Oct 25 09:09:27 multiplexer tsp[49098]: * 2020/10/25 09:09:27 - merge: resetting PCR restamping in PID 0x0051 (81) after DTS/PTS moved backwards restamped PCR
Oct 25 09:10:52 multiplexer tsp[49098]: * 2020/10/25 09:10:52 - merge: resetting PCR restamping in PID 0x0053 (83) after DTS/PTS moved backwards restamped PCR
Oct 25 09:12:37 multiplexer tsp[49098]: * 2020/10/25 09:12:37 - merge: resetting PCR restamping in PID 0x0055 (85) after DTS/PTS moved backwards restamped PCR
Oct 25 09:16:26 multiplexer tsp[49098]: * 2020/10/25 09:16:26 - merge: resetting PCR restamping in PID 0x0050 (80) after DTS/PTS moved backwards restamped PCR
Oct 25 09:27:10 multiplexer tsp[49098]: * 2020/10/25 09:27:10 - merge: resetting PCR restamping in PID 0x0051 (81) after DTS/PTS moved backwards restamped PCR
Oct 25 09:29:50 multiplexer tsp[49098]: * 2020/10/25 09:29:50 - merge: resetting PCR restamping in PID 0x0055 (85) after DTS/PTS moved backwards restamped PCR
Oct 25 09:32:01 multiplexer tsp[49098]: * 2020/10/25 09:32:01 - merge: resetting PCR restamping in PID 0x0052 (82) after DTS/PTS moved backwards restamped PCR
```
This is the mux of the process just started:
https://www.dropbox.com/s/iy6j1rb8ko7k0cb/mux%20tsduck_last.ts?dl=0

This is the mux after a few hours after the constant PCR readjust has started:
https://www.dropbox.com/s/4ktv2s57uf3mg7j/mux%20tsduck_last_2.ts?dl=0

On another hand, seems that the merge modifications has somehow broken the remap plugin when used in a merge process, I had a remapped plugin in one of my process and it did not worked, before upgrading it worked: 
So, this  xxxx -P merge --pcr-reset-backwards  "/usr/bin/tsp   -v  --timed-log  -I ip 1234    -P  remap 256=68 272=69" -O  xxxx does not produce any output

I can provide more details if needed.

Thanks.

lars18th, 2020-10-26:
Hi @lelegard ,

Regarding the comment from @salvaaod about:

> The issue that I saw regarding the regular readjustment of the PCR after a few hours is still there:

And your previous response:

> The timestamp drift information is not explicit but can be extracted from the output of the plugin pcrextract. This is CSV format which was explicitly designed for later analysis using Excel.

I feel it will be very useful to include the PCR and PTS/DTS analysis inside the new plugin `stats`. I don't know if it's a hard work to develop it, but it will be more direct and fast to have this functionality implemented inside TSDuck than exporting CSV files for later processing. Futhermore, you can use it as a module for a testing unit, so perhaps it can helps the improvement of the `merge` plugin.

Please, think on this about a constructive suggestion. I'm not enforcing you to implement something that you think is not useful.
Thank you for your patience!
Regards.

lelegard, 2020-10-31:
Hi @salvaaod 

This is strange. The first PCR reset occurs after almost 8 hours. But afterwards, it occurs every 10 minutes or so. Looking again at the code, resetting the PCR restamping is exactly like registering the first PCR. The PCR is left unmodified in the packet, we forget everything and restart as if it was the first PCR on the PID.

> On another hand, seems that the merge modifications has somehow broken the remap plugin when used in a merge process, I had a remapped plugin in one of my process and it did not worked, before upgrading it worked:

This is different but unclear. We would need the big picture, complete commands, how plugins interact. Here, we only have unrelated command fragments.

salvaaod, 2020-11-01:
> Hi @salvaaod
>
> This is strange. The first PCR reset occurs after almost 8 hours. But afterwards, it occurs every 10 minutes or so. Looking again at the code, resetting the PCR restamping is exactly [...]

Hi @lelegard Yes, I expected the reset_pcr_backwards to behave like resetting the application and have the PCR < PST/DTS again for another 8 hours, seems that this is the time to drift for the audio  PCR, indeed that is what happens every time you restart the application.   I am unsure why then goes to this 10 minutes cicle.

How can I give you more data to debug this? , any suggestion to get less PCR drift ? I am wondering if this is due to underlying OS/CPU, initially I had this running under RPI4/Raspbian 1Gb, and later  tried with i5-4400/Ubuntu 20.04 4Gb and the results are the same.

> This is different but unclear. We would need the big picture, complete commands, how plugins interact. Here, we only have unrelated command fragments.

Forget about this last part, I retested again to give complete details and now works as expected, I am not sure what I did wrong, but now is ok.

Thanks for your support.

rezna11, 2020-11-01:
Hello...  I'm confused as to what changed in `merge` plugin recently. I'm trying simple stuff but there's nothing merged at all now... except for 1 packet.

```
C:\>tsp -v --bitrate 26390374 -I null -P regulate -P merge -t "tsp -v -I ip 239.0.0.1:10000" -O play
```
The same command works fine with TSDuck 3.21-1819.

Debug from non-working 3.24-2044:

```
C:\Users\Admin\Desktop>tsp -v -d1 --bitrate 26390374 -I null -P regulate -P merge -t "tsp -d1 -v -I ip 239.0.0.1:10000" -O file merge.tx
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -v -d1 --bitrate 26390374
* Debug: debug level set to 1
* Debug: null: debug level set to 1
* Debug: registering processor plugin "regulate", status: ok
* Debug: null: ====> tsp -I null
* Debug: registering processor plugin "merge", status: ok
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file merge.tx
* Debug: trying to load "C:\Program Files (x86)\TSDuck\bin\tsplugin_regulate.dll"
* Debug: regulate: debug level set to 1
* Debug: regulate: ====> tsp -P regulate
* Debug: trying to load "C:\Program Files (x86)\TSDuck\bin\tsplugin_merge.dll"
* Debug: merge: debug level set to 1
* Debug: merge: ====> tsp -P merge -t 'tsp -d1 -v -I ip 239.0.0.1:10000'
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* merge: main stream bitrate reset to 0 b/s (was 0 b/s)
* merge: merged stream bitrate reset to 0 b/s (was 0 b/s)
* Debug: merge: creating process "tsp -d1 -v -I ip 239.0.0.1:10000"
* regulate: minimum packet burst duration is 2,000,000 nano-seconds
* Debug: merge: receiver thread started
* Debug: null: initial buffer load: 44,620 packets, 8,388,560 bytes
* null: initial input bitrate is 26,390,374 b/s
* Debug: file: opening output file merge.tx
* Debug: control commands: found 2 packet processor plugins
* Debug: null: input thread started
* Debug: file: output thread started
* Debug: merge: packet processing thread started
* Debug: regulate: packet processing thread started
* regulate: regulated at bitrate 26,390,374 b/s
* Debug: regulate: new regulation, burst: 2,000,000 nano-seconds, 35 packets
* merge: main stream bitrate reset to 26,390,374 b/s (was 0 b/s)
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d1 -v
* Debug: debug level set to 1
* Debug: ip: debug level set to 1
* Debug: ip: ====> tsp -I ip 239.0.0.1:10000
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: ip: setting socket reuse address to 1
* Debug: ip: binding socket to 0.0.0.0:10000
* ip: joining multicast group 239.0.0.1:10000 from local address 10.0.0.11
* ip: joining multicast group 239.0.0.1:10000 from local address 192.168.138.1
* ip: joining multicast group 239.0.0.1:10000 from local address 192.168.42.1
* Debug: ip: initial buffer load: 1 packets, 188 bytes
* ip: unknown initial input bitrate
* Debug: file: opening output file
* Debug: control commands: found 0 packet processor plugins
* Debug: ip: input thread started
* Debug: file: output thread started
* merge: merged stream bitrate reset to 0 b/s (was 0 b/s)
* merge: 999 waiting packets, accelerating merged stream bitrate by factor 2
* Debug: ip: input: got bitrate 7,629,907 b/s
* Debug: ip: input: got bitrate 9,012,985 b/s
* * tsp: user interrupt, terminating...
* Debug: null: input thread aborted after 187,942 packets
* Debug: merge: stopping, last merge bitrate: 0, merged: 1, hold: 94,781, empty: 3,920
```

lelegard, 2020-11-02:
Hi @rezna11,

>  I'm confused as to what changed in merge plugin recently. I'm trying simple stuff but there's nothing merged at all now... except for 1 packet.

Previously, all incoming packets from the merged stream were inserted in the first available null packets. Since packets typically arrive from the pipe in bursts, the result was not smooth in the presence of many null packets in the main stream.

Now, the plugin `merge` tries to smoothen the insertion of packets, based on the evaluated bitrate of the merged stream. Use option `--no-smoothing` to revert back to the previous behaviour.

Based on your log, I investigated and found some issue when the bitrate from the merged stream was not evaluated fast enough, resulting in the behaviour you observed. I have fixed this.

salvaaod, 2020-11-09:
> Based on your log, I investigated and found some issue when the bitrate from the merged stream was not evaluated fast enough, resulting in the behaviour you observed. I have fixed this.

Hi actually the issue I observed:
>On another hand, seems that the merge modifications has somehow broken the remap plugin when used in a merge process, I >had a remapped plugin in one of my process and it did not worked, before upgrading it worked:

Had happened again, upgrading to the fix mentioned before cleared the issue.