tsduck/tsduck #652: Revisited: Detecting I-frames starting GOPs

SS-FF, 2020-10-14:
This is a follow-up to https://github.com/tsduck/tsduck/issues/483

So far, I can only talk about 

## H.264 IDR Frames

- The H.264 Spec (T-REC-H.264-201906 - Clause 2.4.2.4) defines a NAL unit type "Access Unit Delimiter"
- Access Unit Delimiters are optional for H.264
- But the DVB spec makes them mandatory (ETSI TS 101 154)
- And it also requires that IDR frames (RAP in 101154) start at the beginning of a PES

That  means that you should be able to detect IDR frames by doing the following:

- Wait for a PUSI of an H.264 elementary stream
- If yes, check whether the PUSI begins with a start code
- If yes, check whether the first NAL unit is of type AUD
- If yes, check the AUD's primary_pic_type to see whether it's an IDR

=>It's just a single TSPacket to check

SS-FF, 2020-10-20:
Meanwhile I found that similar is possible for MP2Video and HEVC. 

@lelegard - Thierry, you wrote:

> Sorry, I cannot see in the future ;) So, from a PUSI packet, I cannot guess if some forthcoming TS packet on the same PID will contain a start of I-frame (in the near future).

That statement is still logically correct, but it does not apply. 

In case of all three codecs, it is sufficient to investigate PUSI packets only (without re-assembling PES packets) in order to determine the start of a GOP. At least that's the result of my recent research, studying DVB specs, investigating a range of sample streams and finally a prototype implementation.

That means that there's no need for a "2-pass" approach for cutting HLS segments at GOP boundaries.

> Nowadays, with codecs such as AVC or HEVC, I am not sure. We would need the advice of a video codec specialist.

I'm not a video codec specialist, I'm technically far away from tsDuck (programming language, architecture, etc..) - sorry no PR - and I'm not requesting this to be implemented. 

It's just that I think I owe tsDuck, and when there's some interest, I'd be able to help in that matter.

lelegard, 2020-11-13:
@SS-FF, back to Access Unit Delimiter, primary_pic_type is defined as follow in H.254:

[image]

And slice_type is defined as:

[image]

An IDR is defined by slice_type being one of 2, 4, 7, 9. So, the primary_pic_type value is ambiguous.
- primary_pic_type values 0, 3, 5 : always an IDR
- values 1, 2, 4, 6, 7: maybe an IDR, maybe not

Using a modified `pes` plugin and a service from an FTA DVB-T stream, based on all AUD's over 61 seconds, I see:
- primary_pic_type = 0 : 29 times
- primary_pic_type = 1 : 116 times
- primary_pic_type = 2 : 1236 times

That makes 22 frames per second. Not much, considering that mediainfo says that the frame rate is 25 fps.

Only 29 AUD are unambiguously IDR. But one IDR every 2 seconds does not seem realistic on a live DVB stream. If we assume that primary_pic_type 1 are also IDR, that would make 2.4 IDR per second. This is bit high this time (I would have expected 2) but why not. But this is only speculations. Values 1 and 2 for primary_pic_type are ambiguous.

Any idea on how to clearly use primary_pic_type ?

lelegard, 2020-11-15:
More data on another 60 seconds snapshot of clear contents from an FTA DTTV channel, AVC encoding.

Using `-P pes --avc-access-unit --nal-unit-type 9`, the total number of frames is 1363, meaning 22.7 fps.
- Primary picture type = 0, 29 times (I frames)
- Primary picture type = 1, 115 times (I or P frames)
- Primary picture type = 2, 1219 times (I, P or B frames)

The NALunit types are:
- 0x01 (Coded slice of a non-IDR picture)
- 0x06 (Supplemental enhancement information (SEI))
- 0x07 (Sequence parameter set)
- 0x08 (Picture parameter set)
- 0x09 (Access unit delimiter)
- 0x0C (Filler data)

All video data are in NAL unit types 1 (non-IDR picture). Specifically, there is zero NAL unit type 5 (IDR picture).

AviDemux finds 60 "key frames", assuming to mean "intra frames". Many such "key frames" are displayed with missing parts or "dragged content", typical from intermediate frames. So, it is not totally convincing but is maybe only an artifact of AviDemux when skipping from key frame to key frame.

MediaInfo displays this:
~~~
Format                    : AVC
Format/Info               : Advanced Video Codec
Format profile            : High@L4
Format settings, CABAC    : Yes
Format settings, ReFrames : 4 frames
Codec ID                  : 27
Duration                  : 1 min 1 s
Bit rate                  : 4 016 kb/s
Width                     : 1 920 pixels
Height                    : 1 080 pixels
Display aspect ratio      : 16:9
Frame rate                : 25.000 FPS
Standard                  : Component
Color space               : YUV
Chroma subsampling        : 4:2:0
Bit depth                 : 8 bits
Scan type                 : MBAFF
Scan type, store method   : Interleaved fields
Scan order                : Top Field First
Bits/(Pixel*Frame)        : 0.077
Stream size               : 29.3 MiB (84%)
Color range               : Limited
Color primaries           : BT.709
Transfer characteristics  : BT.709
Matrix coefficients       : BT.709
~~~

lelegard, 2020-11-17:
I have implemented the I-frame detection inside PES packets (plugin `pes` with option `--intra-image`) and in TS packets payload (truncated start of PES) to synchronize the creation of HLS segments (output plugin `hls` with option `--intra-close`).

This works on MPEG-2 video, reliably, and on H.264 / AVC on full IDR pictures (NALunit type 5) or on Access Unit Delimiter (NALunit type 9) with primary picture types 0, 3 or 5 (non ambiguously I frames).

There are maybe other ways to detect H.264 I-frames. Nothing is done yet for H.265 / HEVC, and even less for the new H.266. Any new type of detection should be added in a [single function here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/tsPESPacket.cpp#L457).

To make sure an HLS segment is properly closed even if cannot detect the I-frame, there is a kind of timeout (option `--max-extra-duration`, 2 seconds by default). If no I-frame is detected within this timeout after the target duration of the segment, the segment is closed anyway at the next video PES packet.

Any test feedback will be appreciated.

lelegard, 2020-11-21:
Same intra-frame detection added on H.265 (HEVC) and H.266 (VVC).

But the definition of "intra-frame" is less obvious (at least to me) in AVC/HEVC/VVC as it used to be in the old MPEG-2. I am still not sure that _all_ intra-frames are detected or if some detected frames could be not really intra.

SS-FF, 2020-11-23:
> Any idea on how to clearly use primary_pic_type ?

Yes. 

In case of AVC, the easiest way is to check for AUD with primary_pic_type 0 => IDR
That's all that is required. No other condition. Forget about slice_type. 

There are also certain sequences that give a reliable indication, but this might go beyond a single packet.
(something like Nal unit 0x5 followed by SPS, iirc)

This kind of detection might not be valid for H.264 video in general but for H.264 in DVB broadcasts it is.
An 1-2s IDR interval is pretty normal for H.264 TV streams, 

I have analyzed quite a number of DVB streams from everywhere around the world and I have verified my results against ffprobe (with dumping single-frame data as json) and two other H.264 analyzers.

SS-FF, 2020-11-23:
For H,265 it's actually the same check: 

NalUnit AUD and primary_pic_type 0.
(though, nal unit id for AUD has a different value here)

In case of H.262 it's sufficient to check packets for having a start code of SEQ_START_CODE (0xB3)

lelegard, 2020-11-23:
Thanks @SS-FF 

For H.264, H.265 and H.266, this is what I do in TSDuck. With H.264, I also accept primary_pic_type 3 and 5 as intra-image. Not sure if this is necessary. I also check explicit IDR NALunit types, altrough I haven't seen any in my streams.

For H.262, I used PES start code 0xB8 (GOP start code). Trying 0xB3 (sequence header), the very same packets are found except that 0xB3 occurs before 0xB8, closer to the start of the PES payload. So I switched to 0xB3 to maximize the chance to get it in the first TS packet of the PES packet.

SS-FF, 2020-11-23:
> For H.264, H.265 and H.266, this is what I do in TSDuck. With H.264, I also accept primary_pic_type 3 and 5 as intra-image. Not sure if this is necessary. 

I don't do this because it probably wouldn't satisfy the EXT-X-INDEPENDENT-SEGMENTS requirements.

> I also check explicit IDR NALunit types, altrough I haven't seen any in my streams.

I also had that at an earlier stage but it led to doubled cut points (AUD/pictype0 and shortly after nal type 0x5) and besides the doubling, it turned out that the occurrence of and IDR NAL unit was already a too late point in time, because the AUD and SPS would land at the end of the previous segment, which would be wrong.

> For H.262, I used PES start code 0xB8 (GOP start code). Trying 0xB3 (sequence header), the very same packets are found except that 0xB3 occurs before 0xB8, closer to the start of the PES payload. So I switched to 0xB3 to maximize the chance to get it in the first TS packet of the PES packet.

As mentioned, I'm doing the same:

> In case of H.262 it's sufficient to check packets for having a start code of SEQ_START_CODE (0xB3)

SS-FF, 2020-11-23:
Another thing to note is that I ended up cutting all other streams on PES packet boundaries only.

Practically, this means that right in the moment where the video stream is cut, I'm starting a packet-countdown (like 100-500) and a hash set of  all non-video-PIDs. As long as the countdown is running, non-PUSI packets go into the previous HLS segment as long as the PID is in the list. As soon as a PUSI packet is encountered, the PID is removed and the packet is written to the new segment. The previous segment is finished once the PID-hashset is empty or the countdown is zero.

lelegard, 2020-11-23:
> The Trick: This works without any kind of buffering

Good idea. I will probably reuse it, if you don't mind.

What kind of software are you developing?

SS-FF, 2020-11-23:
It's an AI driven video toaster 😆

SS-FF, 2020-11-23:
Check your spam..

SS-FF, 2020-11-23:
> > The Trick: This works without any kind of buffering
> 
> Good idea. I will probably reuse it, if you don't mind.

How would I..?

> What kind of software are you developing?

At some point in time, I'll be able to post a link for reference.

SS-FF, 2020-11-24:
@lelegard  - Did you receive my e-mails from yesterday?

SS-FF, 2020-11-25:
Please re-open in case you got any more questions.

closed 2020-11-25 by SS-FF