tsduck/tsduck #653: tstabcomp: recompiling BAT with multiple sections results in BAT.bin with invalid section length

pilotpirxz, 2020-10-14:
### Workflow:
1. Extracted BAT (selecting 1 specific bouquet ID = 25016) from a TS capture using tstables:
  _tstables --pid 17 --tid 74 --tid-ext 25016 -b BAT25016.bin capture.ts_ 

  The BAT contains 7 sections.

2. decompiled BAT25016.bin to XML format using tstabcomp:
   _tstabcomp --decompile BAT25016.bin_

3. recompiled BAT25016.xml back to binary using tstabcomp:
   _tstabcomp --compile BAT25016.xml -output BAT25016_recompiled.bin_

### Result:
The section length of the 1st section in BAT25016_recompiled.bin is invalid, because it is larger than the max allowed section length (1021).
When comparing BAT25016.bin and BAT25016_recompiled.bin, it looks like some data fields, which are repeated in each section, have all been stacked into the first section.

See attached file:
[TSDuck_export.zip](https://github.com/tsduck/tsduck/files/5378698/TSDuck_export.zip)

lelegard, 2020-10-14:
Hi @pilotpirxz,

Thank you for reporting this. The BAT section size issue is a bug. The initial reason for the current behaviour is that ISO 13818-1 specifies that the maximum section size is 1024 bytes for MPEG-defined PSI sections and 4096 bytes for private sections (field `private_indicator` set to 1). All DVB-defined tables being "private" in the MPEG sense, the serialization uses up to 4096 bytes per section. However, I just noticed that ETSI EN 300 468 specifies that the maximum section size for the NIT, BAT and SDT is 1024 bytes (other sections such as EIT accept up to 4096 bytes). So, I fixed the (non standard) incorrect maximum size for these three types of tables. Compiling your BAT now splits sections before 1024 bytes.

Considering the second issue, I think that your initial table is incorrect according to DVB SI guidelines (ETSI TS 101 211) and that TSDuck behaviour is correct. Let me explain why.

A BAT is composed of "bouquet descriptors" (the first loop) and "transport streams entries" with one descriptor loop in each. These informations are spread over several sections and "the BAT" is the concatenation of all these informations, gathered from all sections.

Now, some definitions. In ETSI TS 101 211, section 3.1, "Definitions":

> sub-table: comprised of a number of sections with the same value of table_id, table_id_extension and version_number

> table: comprised of a number of sections with the same value of table_id

In section 4.2.2.1.1, "Bouquet name descriptor":

> This descriptor is allowed once in the first descriptor loop of each sub-table of the BAT. It is mandatory to be transmitted in the first descriptor loop of any BAT sub-table in the TS.

Note the "once in the first descriptor loop of each _sub-table_" and that "sub-table" is defined as "a number of sections with the same value of table_id, table_id_extension and version_number". So, a "sub-table" is in fact the complete BAT for the bouquet and the "table" is the set of all BAT sections, all versions, all bouquets.

So, in the BAT (the set of all sections in that BAT), the descriptor is allowed once and once only.

In your original BAT, this descriptor (and all other descriptors of the first loop) are repeated in all sections of the BAT, which is incorrect according to ETSI TS 101 211. It seems that the generator makes a confusion between "sub-table" and "section".

When the BAT is deserialized, all "bouquet descriptors" and "transport streams entries" are collected from all sections. So, the resulting "bouquet descriptors" are concatenated from all sections, resulting in cycles of identical descriptors in your case.

When a BAT is serialized, the serialization engine shall spread "bouquet descriptors" and "transport streams entries" over sections in some unspecified way (as long as the syntax is correct). In the case of TSDuck, in the BAT and all similar tables, as many "bouquet descriptors" as possible are stored in the first section. Then, if there is some space left, as many "transport streams entries" as possible are stored. In the second section, more "bouquet descriptors" are serialized (if any is remaining), same for "transport streams entries", etc. More sections are generated as long as there are "bouquet descriptors" or "transport streams entries" left.

This explains why the repeated descriptors in your original BAT are grouped in the BAT "bouquet descriptors" and why they are all serialized in the same section on output. But the original issue is that those "bouquet descriptors" shouldn't have been repeated in all sections of the original BAT.

pilotpirxz, 2020-10-15:
Hi Thierry,

Many thanks for the your quick response and the in-depth analysis of the 2nd issue.
The BAT was extracted from a broadcast transport stream (so I have no control over the content), but it is now clear to me that there is also a problem with the source.

Kind regards,
Bart

Reply to: tsduck/tsduck <reply@reply.github.com>
Date: Wednesday, 14 October 2020 at 23:17
To: tsduck/tsduck <tsduck@noreply.github.com>
Cc: Bart Lagae <blagae@synamedia.com>, Mention <mention@noreply.github.com>

lelegard, 2020-10-15:
> The BAT was extracted from a broadcast transport stream (so I have no control over the content), but it is now clear to me that there is also a problem with the source.

It happens all the time. Many signalizations are incorrect, from various operators all over the world. There are even some specific options in TSDuck commands to correctly interpret incorrect content in some well-know cases. In your case, the syntax is correct, so there is nothing to "fix". It's just duplicated information.

Back to the non-standard section size for the BAT: I checked the specification of all implemented tables in TSDuck, looking for similar non-standard maximum section sizes and I found (and fixed) a few others:

- DVB: NIT, SDT, BAT, AIT, RST
- ATSC: STT, RRT, CVCT, TVCT
- SCTE 18: cable emergency alert table
- SCTE 35 : splice information table

Thanks for drawing my attention to this non-standard oddity.

pilotpirxz, 2020-10-23:
Will the BAT section length issue be fixed in the next TSDuck release? When is this due?

lelegard, 2020-10-23:
Yes, the fix is already in place in the code. If you use Windows or Ubuntu, you will find [prebuilt binaries for the nightly builds here](https://tsduck.io/download/prerelease/). For other systems, you need to rebuild from sources, [instructions here](https://tsduck.io/doxy/building.html).

Generally speaking, TSDuck is a small project in terms of resources. I use continuous integration with small stable increments. There is no separate branches for new features and fixes. They are directly integrated and tested in the master branch.

There is no product management team which plans new features and releases schedules. However, thanks to the  continuous integration with small stable increments, nightly builds are as stable as new releases.

closed 2022-09-23 by lelegard