tsduck/tsduck #654: tsduck as a CAS for BISS-CA

nikpakar, 2020-10-23:
Dear all, sorry im been blunt to ask this question here.

Is it possible to develop a CAS for BISS-CA specification from tsduck ? Has any one done any attempt to develop any open source CAS ?

Thank you so much for any feedback.

-Nik

lelegard, 2020-10-23:
The first requirement of a Conditional Access System is security. 

Software and security are two independent dimensions.
Software is a set of features, something with an existence.
Security is a systemic characteristics of software, an attribute without existence in itself.

You could technically borrow some basic building blocks from TSDuck to build a functional software CAS. TSDuck itself can be used as a CAS using `tsecmg` (which generates fully functional ECM's with zero security). The plugin `scrambler` produces the CA-protected stream using _any_ ECMG, a real secure one or an totally insecure one such as `tsecmg`. You can even test the result with the `descrambler` plugin which understands the ECM's from `tsecmg`. Everything is functional. Because it was developed to test the _features_ of various external equipment and their capabilities and limitations.

You can implement a functionally correct open source CAS, using TSDuck or not using it. But it will have zero security and will consequently miss the first requirement of a CAS.

Security theorists will invoke the Kerckhoffs's principle which says that security shall rely only on the keys and not the implementation. This is formally very true and works well with basic blocks. An open source implementation of AES which has been reviewed and fixed by all cryptographers is much better than any other. But if you think that theory and reality always match, you need to grow up. In the real world where systems are complex, human developers are fallible and have limited resources when hackers have unlimited time and tenacity, an fully open source system is a system where vulnerabilities are exposed to hackers because all systems have vulnerabilities.

In my opinion, after 24 years in the CAS business, the simple idea of a full open source CAS is an oxymoron, a contradiction in terms. The most insane idea in this business...

nikpakar, 2020-11-02:
Thanks for the comprehensive feedback. But there are so many open source security related project isnt it ? some times been open source itself gives the validity for the security and actual implementation some thing like BISS-CA will be depending on already proven AES type of encryption and only if some one has the keys they can decrypt it, not just because the logic source code is open. Am I correct ?

From the design of BISS-CA itself, i think an un-encrypted clear ECM is permitted because the keys itself already encrypted using already accepted proven encryption methods.

In this context, wouldnt it be a good idea to develop an open-source CAS based on BISS-CA standard ?

Sorry if im been blunt, just trying to see whether my approach is right.

lelegard, 2020-11-02:
> But there are so many open source security related project isnt it ? some times been open source itself gives the validity for the security and actual implementation some thing like BISS-CA will be [...]

That was what I was saying when I was referring to the Kerckhoffs's principle.

Theory is one thing, practice is another one. Now, you do what you want...