tsduck/tsduck #655: HLS Input Plugin stop hls: HLS playlist completed

juanki182, 2020-10-23:
Hi
First of all, thank you for this amazing tool.

I have noticed that the hls input plugin stops after a while. Is this an expected behavior?, because I have seen this on 3 different servers receiving the same hls stream.
This is the stream I am testing:
tsp --debug -v -I hls --connection-timeout 3600000 --receive-timeout 3600000  https://59c3c7bda15f4.streamlock.net:444/maxtvonline/smil:maxtvonline.smil/playlist.m3u8 -P regulate --pcr-synchronous --pid-pcr 256 -O ip 239.255.1.1:1385
.
.
.
hls: HLS playlist completed

tsversion 3.24-2004
Is there a way for the plugin to restart automatically?
I checked this case: **HLS input problem #175**

Thanks! 
(Sorry for my bad English)

lelegard, 2020-10-23:
We get "403 Forbidden" when trying this link. So we cannot analyze the issue.

juanki182, 2020-10-23:
Ohh ok, could you try this: https://streamingwowza.com:1936/maxtvonline/smil:maxtvonline.smil/chunklist_w1534291786_b2108000_sleng.m3u8
This is a Ecuadorian  online channel (free)
https://maxtvonline.com/canal-online/

lelegard, 2020-10-23:
Same result:
~~~
$ curl -L 'https://streamingwowza.com:1936/maxtvonline/smil:maxtvonline.smil/chunklist_w1534291786_b2108000_sleng.m3u8' -D /dev/stdout
HTTP/1.1 403 Forbidden
Accept-Ranges: bytes
Access-Control-Expose-Headers: Date, Server, Content-Type, Content-Length
Server: WowzaStreamingEngine/4.8.5
Access-Control-Allow-Origin: *
Access-Control-Allow-Credentials: true
Access-Control-Allow-Methods: OPTIONS, GET, POST, HEAD
Connection: Keep-Alive
Access-Control-Allow-Headers: Content-Type, User-Agent, If-Modified-Since, Cache-Control, Range
Content-Length: 0
~~~

juanki182, 2020-10-23:
oh.. It works for me, maybe a geolock.. which is weird since their slogan is: De Ecuador para el mundo!!! (From Ecuador to the world !!!) :D

PS C:\Users\JuanCarlos\Desktop> tsp --debug -v -I hls --connection-timeout 3600000 --receive-timeout 3600000 https://streamingwowza.com:1936/maxtvonline/smil:maxtvonline.smil/chunklist_w1534291786_b2108000_sleng.m3u8 -P regulate --pcr-synchronous --pid-pcr 256 -O drop
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug -v
* Debug: debug level set to 1
* Debug: hls: debug level set to 1
* Debug: hls: ====> tsp -I hls --connection-timeout 3600000 --receive-timeout 3600000 https://streamingwowza.com:1936/maxtvonline/smil:maxtvonline.smil/chunklist_w1534291786_b2108000_sleng.m3u8
* Debug: drop: debug level set to 1
* Debug: drop: ====> tsp -O drop
* Debug: trying to load "C:\Program Files\TSDuck\bin\tsplugin_regulate.dll"
* Debug: registering processor plugin "regulate", status: ok
* Debug: regulate: debug level set to 1
* Debug: regulate: ====> tsp -P regulate --pcr-synchronous --pid-pcr 256
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* regulate: minimum wait: 2,000,000 nano-seconds, using 50,000,000 ns
* Debug: hls: downloading https://streamingwowza.com:1936/maxtvonline/smil:maxtvonline.smil/chunklist_w1534291786_b2108000_sleng.m3u8
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Accept-Ranges: bytes
* Debug: hls: HTTP header: Access-Control-Expose-Headers: Date, Server, Content-Type, Content-Length
* Debug: hls: HTTP header: Server: WowzaStreamingEngine/4.8.5
* Debug: hls: HTTP header: Cache-Control: no-cache
* Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* Debug: hls: HTTP header: Access-Control-Allow-Credentials: true
* Debug: hls: HTTP header: Access-Control-Allow-Methods: OPTIONS, GET, POST, HEAD
* Debug: hls: HTTP header: Connection: Keep-Alive
* Debug: hls: HTTP header: Access-Control-Allow-Headers: Content-Type, User-Agent, If-Modified-Since, Cache-Control, Range
* Debug: hls: HTTP header: Date: Fri, 23 Oct 2020 22:27:27 GMT
* Debug: hls: HTTP header: Content-Type: application/vnd.apple.mpegurl
* Debug: hls: HTTP header: Content-Length: 274
* Debug: hls: announced content size: 274 bytes
* Debug: hls: MIME type: application/vnd.apple.mpegurl
* hls: downloaded chunklist_w1534291786_b2108000_sleng.m3u8, media playlist, updatable (live), 3 segments, 11 seconds/segment
* Debug: hls: internal push-input thread started
* Debug: hls: downloading segment https://streamingwowza.com:1936/maxtvonline/smil:maxtvonline.smil/media_w1534291786_b2108000_sleng_3339.ts
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Accept-Ranges: bytes
* Debug: hls: HTTP header: Access-Control-Expose-Headers: Date, Server, Content-Type, Content-Length
* Debug: hls: HTTP header: Server: WowzaStreamingEngine/4.8.5
* Debug: hls: HTTP header: Cache-Control: no-cache
* Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* Debug: hls: HTTP header: Access-Control-Allow-Credentials: true
* Debug: hls: HTTP header: Access-Control-Allow-Methods: OPTIONS, GET, POST, HEAD
* Debug: hls: HTTP header: Connection: Keep-Alive
* Debug: hls: HTTP header: Access-Control-Allow-Headers: Content-Type, User-Agent, If-Modified-Since, Cache-Control, Range
* Debug: hls: HTTP header: Date: Fri, 23 Oct 2020 22:27:27 GMT
* Debug: hls: HTTP header: Content-Type: video/MP2T
* Debug: hls: HTTP header: Content-Length: 1916096
* Debug: hls: announced content size: 1916096 bytes
* hls: downloading from https://streamingwowza.com:1936/maxtvonline/smil:maxtvonline.smil/media_w1534291786_b2108000_sleng_3339.ts
* hls: MIME type: video/mp2t, expected size: 1916096 bytes
* Debug: hls: initial buffer load: 5 packets, 940 bytes
* hls: unknown initial input bitrate
* Debug: control commands: found 1 packet processor plugins
.
.
.

lelegard, 2020-10-24:
> maybe a geolock.. 

Most probably.

> which is weird since their slogan is: De Ecuador para el mundo!!! (From Ecuador to the world !!!) :D

This is just marketing BS. All TV channels have legal agreements with content rights holders, especially sports leagues and studios. They purchase the rights to broadcast a given content within a given geographical area, usually their home country, sometimes extended to the neighbouring countries. So, TV channels must implement technical means to limit the reception of their content to the commercially agreed area. This is very important for FTA channels. PayTV channels with expatriate subscribers abroad may have different clauses in their contracts with content owners. Only illegal pirated content can be received anywhere :)

We need more information on the technical issue. What are the last messages, debug included, at the end of the execution, just before the execution stops? Use option `--save-file` to keep the media playlists and media segments and post here the content on the last playlist, the `.m3u8` text file.

Additionally, if you want to relay a live channel, use option `--live` to avoid a useless time shift.

juanki182, 2020-10-25:
Hi @lelegard 
Thanks for your help.
I've been doing some tests over the weekend, and I was able to get the requested information on 2 different machines (Win 10 and Ubuntu Server) after the plugin stopped. The plugin stopped at the same time on both machines.
[HLS Ubuntu Server.zip](https://github.com/tsduck/tsduck/files/5436102/HLS.Ubuntu.Server.zip)
[HLS WIN 10.zip](https://github.com/tsduck/tsduck/files/5436107/HLS.WIN.10.zip)

I hope it is helpful and once again thank you very much for your help.
Best regards

lelegard, 2020-10-29:
Hi @juanki182 ,

Thanks for the logs and files. The problem is identified. At some point the server fails to return the playlists. So, this is a broadcaster's server error.

In the Window logs:
~~~
* Debug: hls: no new segment in playlist
* Debug: hls: downloading https://streamingwowza.com:1936/maxtvonline/smil:maxtvonline.smil/chunklist_w1534291786_b2108000_sleng.m3u8
* Debug: hls: HTTP header: HTTP/1.1 404 Not Found
~~~

In the Ubuntu logs:
~~~
* Debug: hls: downloading https://59c3c7bda15f4.streamlock.net:444/maxtvonline/smil:maxtvonline.smil/chunklist_w1272084643_b2108000_sleng.m3u8
* Debug: hls: HTTP header: HTTP/1.1 404 Not Found
~~~

The principle of HLS is simple. For live streams, the playlist file `.m3u8` contains a list of a few segments. The playlist file is regularly updated with new segments. Here, each segment lasts about 10 seconds and there are 3 segments in the playlist. So, the current playlist references 30 seconds of live content. The playlist is downloaded again and again to get new segments.

In your case, at some point, the server no longer serves the playlist. It returns the famous "404 Not Found" when the `hls` plugin tries to download it.

In the last successfully downloaded playlists, the "media sequence" of the segments are 20501, 20502, 20503 with the Windows client and 20502, 20503, 20504 with the Linux client. So, it is clear that this is the server which fails to serve all requests at some point and both clients fail to continue to play the stream.

Maybe, if the server error is transient, the `hls` could implement some retry strategy.

juanki182, 2020-10-30:
Hi @lelegard
Thank you very much for the analysis and explanation. I will close the issue since it is definitely an origin server problem, I have done tests with up to 4 machines and they all stop at the same time..

Best Regards
Juan.

closed 2020-10-30 by juanki182