tsduck/tsduck #656: Multicast from TS

EnoSat, 2020-10-25:
Hi, maybe help with create command for extraction or analyze this situation:
source: satellite signal using Multistream 21
stream: multicast stream on pid 515
`
root@osmio4kplus:~# dvbsnoop -s pidscan
dvbsnoop V1.4.55 -- https://github.com/OpenVisionE2/dvbsnoop

---------------------------------------------------------
Transponder PID-Scan...
---------------------------------------------------------
PID found:    0 (0x0000)  [SECTION: Program Association Table (PAT)]
PID found:   16 (0x0010)  [SECTION: Network Information Table (NIT) - actual network]
PID found:  515 (0x0203)  [SECTION: DSM-CC - private data section  // DVB datagram]
PID found:  802 (0x0322)  [SECTION: Program Map Table (PMT)]
PID found: 2097 (0x0831)  [SECTION: User private]
PID found: 2098 (0x0832)  [SECTION: User private]
PID found: 5632 (0x1600)  [unknown]
root@osmio4kplus:~# tsp -v -I dvb --device-name /dev/dvb/adapter0 --delivery-system DVB-S2 --fec-inner 8/9 --frequency 10810000 --polarity horizontal --modulation QPSK --roll-off auto -s 30000000 -P tables -a -d -p 515 -b test.zip -O drop
* tables: creating test.zip
* Error: dvb: error opening /dev/dvb/adapter0/frontend0: Device or resource busy
root@osmio4kplus:~# tslsdvb -a 0
/dev/dvb/adapter0 (Availink AVL6261, DVB-S, DVB-S2)`

lelegard, 2020-10-25:
Hi @EnoSat 

The "device busy" error seems to indicate that the tuner is already used, typically by some program receiving a TS.

Some additional remarks:

- You name the output binary file of the `tables` plugin "test.zip". The name seems to imply that you expect a zipped file. That won't be the case. The binary file contains raw sections as described in section 2.2.1 of the TSDuck user's guide.
- You do not specify any terminating condition. Your command will run forever (since a DVB reception has no "end of file") and sections accumulate forever in the binary file. You should specify some time or packet limit using a plugin `until`.

rezna11, 2020-10-26:
Judging by the command line and tuner type, he's probably trying to run TSDuck on a Linux satellite receiver running Enigma2 OS (Edision OS Mio 4K). It probably can't work because of tuner limitations in these boxes. This was discussed before: https://github.com/tsduck/tsduck/issues/173

lelegard, 2020-10-29:
@rezna11, thanks for the hint. This type of device is typically used for illegal access to PayTV content and consequently out of scope for TSDuck which is a lab tool for professional and hobbyists (needless to say that hobbyists and wannabe-hackers are distinct types of population).