tsduck/tsduck #657: Missing DLL files in latest Official build

Sefy76, 2020-10-28:
The previous version i've used TSDuck-Win64-3.18-1361 had the following plugin dll files which i've used:
tsplugin_dektec.dll
tsplugin_drop.dll
tsplugin_file.dll
tsplugin_hls.dll
tsplugin_ip.dll

When I downloaded the latest build 3.23-1965, I could not find any of those files in the Bin folder and could not figure out in the changelog what DLL file I need as a replacement. (i'm trying to have a minimum package with just TSP and the required DLL's so it will no be too big but still useable for my needs.

lelegard, 2020-10-28:
The corresponding plugins still exist (try `tsp -l`) but are no longer in separate DLL's. Most of their code was in the main DLL (tsduck.dll) and the individual DLL were just proxies. So, they were removed and the corresponding plugins are buildins in the TSDuck library.

> i'm trying to have a minimum package with just TSP and the required DLL's so it will no be too big but still useable for my needs.

There is not much to remove. Have a look at the various EXE and DLL. They are all very small. Everything is in the TSDuck library because this is a general purpose DTV library. All EXE and DLL are just small wrappers. Removing some of them will not gain much space. The TSDuck installer contains the tools and plugins, the documentation and development environment. When you select only tools & plugins during installation, there is not really any need to tune space by removing small EXE or DLL.

Sefy76, 2020-10-29:
So, if I were to only have the TSP.exe and TSduck.dll, then I could still use the previous commands that needed the DLL's?
Thank you! :+1:

lelegard, 2020-10-29:
You want to use the plugin named `IP` (uppercase) which does not exist. 
Only the plugin named `ip` (lowercase) is registered.

Sefy76, 2020-10-29:
I see, so it's now case-sensetive, with the previous build that command worked as is, but good to know!
I can confirm that the command is now working, so one final question which is related to the HLS in the CMD provided above, i keep getting this message:
Debug: hls: unexpected file extension in reference URI: index_1_647401.ts?m=1602078555

Is this an issue or just not supported?
and thank you for everything!

lelegard, 2020-10-29:
> I see, so it's now case-sensetive, with the previous build that command worked as is, but good to know!

It has always been case sensitive. It's just the Windows file system which is not case sensitive and for which files `plugin_ip.dll` and `plugin_IP.dll` are identical. You command would have never succeeded on Unix systems (Linux and the like). 

The HLS issue is different. I just tried the playlist from your command and it returns the following list of segments:
~~~
#EXTM3U
#EXT-X-VERSION:3
#EXT-X-TARGETDURATION:4
#EXT-X-MEDIA-SEQUENCE:647869
#EXT-X-PROGRAM-DATE-TIME:2020-10-29T04:38:31.596Z
#EXTINF:3.003,
index_1_647869.ts?m=1602078555
#EXTINF:3.003,
index_1_647870.ts?m=1602078555
#EXT-X-PROGRAM-DATE-TIME:2020-10-29T04:38:37.602Z
#EXTINF:3.003,
index_1_647871.ts?m=1602078555
#EXTINF:3.003,
index_1_647872.ts?m=1602078555
...
~~~

This is a limitation of the TSDuck `hls` plugin. It expects all segments to ends in `.ts` because HLS can use TS or fMP4 files while TSDuck can only process transport stream files. Here, the parsing of the segment name is incorrect and fails to take apart the arguments after the '?'. This is legal but unusual. All TS-based HLS playlists I have seen so far did not use an argument part in the relative URL of their segments. I will fix this.

lelegard, 2020-10-29:
After checking, the message " Debug: hls: unexpected file extension in reference URI: index_1_647401.ts?m=1602078555" is just a debug warning (not seen in normal or verbose mode). The URL is correctly used and the media segment is correctly downloaded anyway. So, the required fix is just aesthetics: check that the actual file part ends in `.ts` after stripping the argument part and do not report such a debug message in that case. But, in fact, it works.

Sefy76, 2020-10-29:
I will let the guys on the side of the HLS know of this.
Thank you again for all your assistance and great work on this tool!
This ticket can be closed :+1:

closed 2020-10-29 by Sefy76