tsduck/tsduck #658: PCR continuity over 26 hours period

bertothunder, 2020-10-30:
Hi,

 I've developed on my team a plugin for TSDuck that calculates pcrdrift on a given TS stream; it just keeps a record, in a similar manner than pcrextract does, of the PCR/DTS values on the stream, and for each packet with a valid PCR, it calculates:

  - local clock datetime
  - drift value
  - delta drift (in ms) between current and previous PCR values.
  - the deltas between the PCR representation and the datetime for the current vs previous and current vs first packet.

 I did this in order to research on an issue we are having with encoders / packagers warning about drifting (over 30 seconds in some cases), to point out where drifting could be coming from.

 However, just found out one expected, but bugging, issue: after 26.5 hours, deltas go crazy, due to the PCR accuracy, that gives enough values up to that time.

 I've been looking for some way from tsduck to handle this, but I can't find it, which honestly surprises me given the lot of nice features and design you @lelegard  put on it?

lelegard, 2020-10-30:
26.5 hours is exactly the time after which PCR values wrap up. The max PCR value is 2^33 * 300 at 27 MHz, exactly 26.5 hours.

bertothunder, 2020-10-30:
@lelegard yep, I know that, it's a known thing. 

My question comes if tsduck has any flag or any way to discover the wrap up already happened, or I'll have to check it myself? Open to ideas, though.

bobben1no, 2020-10-30:
I have done something similar in PHP based on a log of pcr samples and their unix time stamp when they was received.  I may use a different approach as I calculate an expected PCR value based on the difference  between unix time stamps between receiving the two pcrs. Then I calculate drift as the difference between expected pcr and the real pcr value.

Perhaps you only need to use a MOD ((2^42)-1) when calculating expected pcr value ?
 ((2^42)-1)=0x3FFFFFFFFFF;  // maximum numerical value for 42-bit pcr

```

expected_pcr=(previous_pcr + time_since_previous_pcr*27000000)  % 0x3FFFFFFFFFF;  // MOD maximum_pcr_value (42-bit) so the expected PCR wraps correctly.

```

or if you are working with 33-bit part of the pcr:
 ((2^33)-1)=0x1FFFFFFFF;  // maximum numerical value for 33-bit pcr base
```

expected_pcr=(previous_pcr + time_since_previous_pcr*90000)  % 0x1FFFFFFFF;  // MOD maximum_pcr_value (33-bit) so the expected PCR base wraps correctly
```

You will also have to make some if statements for drift calculations to handle when expected pcr is near maximum value and real pcr has just wrapped (and vice versa).

bertothunder, 2020-10-30:
> I have done something similar in PHP based on a log of pcr samples and their unix time stamp when they was received. I may use a different approach as I calculate an expected PCR value based on the [...]

Sounds like a plan, I'll work on a PoC to implement it.

bertothunder, 2020-11-23:
Hi @lelegard  I ended up using those definitions. Let me tell you how much I love your code, what a masterpiece of fine code, thank you!

Using `wrapUpPCR` and `diffPCR` is pretty easy to get it done for a single wrap up. `diffPCR` already counts such a chance, so the result is the right one after a wrapping up.

As the plugin I developed for my team is used to track PCR drifts over long periods of time (weeks, usually), we are not only calculating PCR deltas and the difference in local clock time between two PCRs, but also between current PCR and the first seen PCR value... which might be hanging around since days ago. 

In this case, to keep the absolute delta values calcualted the right way, I'm just counting the wrapUp's every time it happens, multiply PCR_SCALE by this counter, and adding it to current PCR value before comparing it to the first PCR in the PIDContext structure (I know, I know, badly defined naming). Something like this:

`
    // Check if this the first packet seen for this PID; in such a case, obtain the first PCR and the system clock time.
    // Each packet will use this as reference for the absolute drift deviation (if any)
    if ( data.first_value == ts::INVALID_PCR ) {
        data.first_value = value;
        data.first_system_time_ns = time_ns;
    }

    // The calculation for PCRDiff already counts wrapUp procedure, so no need to perform different calculations for each
    const uint64_t PCRDiff = data.last_value != ts::INVALID_PCR ? ts::DiffPCR(data.last_value, value) : 0;

    if (data.last_value != ts::INVALID_PCR && ts::WrapUpPCR(data.last_value, value)) {
        data.wrap_cnt++;
        tsp->warning(u"PCR WrapUp detected for PID %d, cycle #%d -> adjusting from old value %d to %d",
                    {ctx.pid, data.wrap_cnt, data.last_value, value});
    }

    // Calculate nominal deltas between current PCR Value (including all the wrapUp distances) and the initial PCR value.
    const uint64_t AbsPCRDiff = ( data.wrap_cnt * ts::PCR_SCALE + value ) - data.first_value;
`

bertothunder, 2020-11-23:
@lelegard this issue can now be closed.

closed 2020-11-23 by bertothunder

Anas92230, 2022-01-24:
Hi @bertothunder,
did you finish your plugin? i'm very interessting by that. My problem is on multicast with multiple port (Multicast ABR) when i analyze it with DVB Analyser i can see EBP_Timestamp is different between profiles. And i suspect a PCR drift on original stream, before abr. And i search a way to show it.
I found lot of pcr_jitter : tsp -I ip ip_src@ip:port: -l interface -P pcrverify -i -O drop the result is :
* pcrverify: PID 1000 (0x03E8), PCR jitter: 27,107 = 1,003 micro-seconds = 0 packets + 0 bytes + 0 bits (kernel time)
* pcrverify: PID 1000 (0x03E8), PCR jitter: -85,660 = 3,172 micro-seconds = 0 packets + 0 bytes + 0 bits (kernel time)
* pcrverify: PID 1000 (0x03E8), PCR jitter: 44,108 = 1,633 micro-seconds = 0 packets + 0 bytes + 0 bits (kernel time)
* pcrverify: PID 1000 (0x03E8), PCR jitter: 29,673 = 1,099 micro-seconds = 0 packets + 0 bytes + 0 bits (kernel time)
* pcrverify: PID 1000 (0x03E8), PCR jitter: -70,014 = 2,593 micro-seconds = 0 packets + 0 bytes + 0 bits (kernel time)
thanks for your help