tsduck/tsduck #659: RPM build errors: fatal error: tsTeletextCharset.h: No such file or directory

D3nis, 2020-10-31:
4.18.0-193.14.2.el8_2.x86_64 #1 SMP Sun Jul 26 03:54:29 UTC 2020 x86_64 x86_64 x86_64 
CentOS Linux release 8.2.2004 (Core) 

when I build:
make rpm NODTAPI=1 NOPCSC=1 NOTELETEXT=1 -j16

I get this errors:
```
In file included from tsduck.cpp:35:
tsduck.h:587:10: fatal error: tsTeletextCharset.h: No such file or directory
 #include "tsTeletextCharset.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[4]: *** [../../Makefile.common:1030: /root/rpmbuild/BUILD/tsduck-3.24-2044/bin/release-x86_64-centos8-rpm-build/objs-libtsduck/tsduck.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [Makefile:43: default] Error 2
make[2]: *** [Makefile:59: default] Error 2
error: Bad exit status from /var/tmp/rpm-tmp.h1c1Nq (%build)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.h1c1Nq (%build)
make[1]: *** [Makefile:115: rpm] Error 1
make: *** [Makefile:107: rpm] Error 2
```

"make rpm  -j16" works fine and RPM's will be builded fine.

lelegard, 2020-10-31:
Hi @D3nis,

Thanks for reporting this. The `NOTELETEXT` option was clearly not tested with `make rpm`. I just fixed this. 

Note that `make rpm -j16` is useless since the compilation is performed in a different environment, using the `.spec` file. The default RPM `mflags` uses the number of CPU's, which is suboptimal. To force the `-j` option in RPM builds, I add the following line in my `$HOME/.rpmmacros`:
~~~
%mflags -j10
~~~

D3nis, 2020-11-02:
thank You for your suggestion with mflags.
I can confirm 
make rpm NODTAPI=1 NOPCSC=1 NOTELETEXT=1
works now well.
thank you.

closed 2020-11-02 by D3nis