tsduck/tsduck #660: tsswitch to switch between PIDs of an MPEG-TS stream? Also is there any way to create a stream with grid of programs?

pxnky, 2020-11-01:
Hi, two things.. I have another application that is dealing with tuning (of a satellite receiver) but I wanted to receive from RTP an MPEG-TS with multiple programs, then be able to select different programs through the UDP remote interface and output a single MPEG stream (does not need to be a TS I guess).. If there is a way, would it still work even when the satellite receiver changes channels and then the stream being processed by that instance of tsduck changes? Meaning it'd have to be dynaic I guess.

The other question is, is there a way to receive a multi program MPEG-TS and output a single MPEG stream that has all of the programs in scaled down boxes, like if theres 4 streams you'd get the screen split into quarters with a stream in each quadrant.

Thanks!

lelegard, 2020-11-01:
There are several ways you can explore.

The first one is to demux the input MPTS into as many SPTS as channels in the TS. This can be done using several `-P fork "tsp -P zap ..."` inside the same `tsp` command. Then `tsswitch` can be used to select the input, including using UDP messages.

The second option is to use one single `tsp -P zap ...` and then restart the `zap` plugin on another service using `tspcontrol` remotely.

> If there is a way, would it still work even when the satellite receiver changes channels and then the stream being processed by that instance of tsduck changes?

The `zap` plugin follows the evolutions of the PAT and PMT. If the PID's of a service change in a new version of the PMT, the plugin will follow. If the service disappears, then `tsp` will stop.

> The other question is, is there a way to receive a multi program MPEG-TS and output a single MPEG stream that has all of the programs in scaled down boxes, like if theres 4 streams you'd get the screen split into quarters with a stream in each quadrant.

This is possible, I am aware of at least one user who did that. I do not have the details of the commands. TSDuck alone cannot do that because it involves video decoding and reencoding. The association between TSDuck and FFmpeg is required. The idea is to demux all channels into distinct SPTS using TSDuck (same as first option above). The four SPTS are fed into FFmpeg which decodes the 4 services, resizes and moves the video and reencodes the result. Then, TSDuck may rebuild the signalisation of the output if necessary.

But take into account that your CPU needs to run four decodes and one encode in parallel. It will work with simple video formats on a powerful CPU. Now, if you want to decode four HEVC and reencode one on a Raspberry Pi, then you will run into problems :)

pxnky, 2020-11-01:
Thanks for the reply! When you say "if the service disappears then `tsp` will stop, I wonder if this will be my pain point. I'm imagining tuning the satellite receiver to totally different transponders and I assume that the `service` that was being sent on the previous transponder would be seen as stopping by tsp? But I am a noob with these details of MPEG so unsure.

Thanks for the pointer to ffmpeg, this is a good idea but similarly I hope to be able to make it dynamic so I can tune underlying transponders and have the output change.. the 4 programs was an example, one transponder might have just one per carrier (SCPC) (and thus no 'program' I think) whereas others might have 7 or 15 or whatever .. some may be just audio.. I don't care much about audio for these purposes and would just skip those.

pxnky, 2020-11-01:
BTW.. I would have tsp receiving with the `dvb` input plugin but that seems to want to control the `frontend` rather than just the `demux` .. it would be nice to be able to bypass trying to open the frontend. Then again it is handy to be able to access the raw stream too.. so maybe I wouldn't even use this right away. I am just running `dvbstream 8192` for this (8192 tells it stream a raw TS not a specific PID)..

lelegard, 2020-11-01:
> When you say "if the service disappears then tsp will stop", I wonder if this will be my pain point.

You may use option `--ignore-absent`. Please be sure to read the reference documentation for plugins you use, at least to check if you can find what you are looking for. TSDuck is a vast toolbox with many plugins and options. You cannot use it efficiently without reading the documentation. My frustration about the lack of exhaustive documentation for ffmpeg was my primary driver to write a complete user's guide for TSDuck.

The user's guide is here: https://tsduck.io/download/docs/tsduck.pdf

Note that TSDuck is just a lab toolbox. If you want a complete, reliable and dynamic system which does everything for operation, you will have to develop your own application. TSDuck can help you because it is essentially a large C++ library for most aspects of TS processing. And the source code for TSDuck can be used as example.

The programmer's guide is here: https://tsduck.io/doxy/

> I would have tsp receiving with the dvb input plugin but that seems to want to control the frontend rather than just the demux 

TSDuck always processes the complete TS. Your vision of frontend vs. demux is based on LinuxTV. The DirectShow framework on Windows works differently. TSDuck is portable and aims at working the same way on all systems.

closed 2020-11-02 by pxnky