tsduck/tsduck #662: hides-drivers unable to build at Pi4 after upgrade

omikron88, 2020-11-06:
Hello, the dkms build fails at Raspberry Pi 4 after some update & upgrade:

**./build-dkms.sh**
build-dkms.sh: building version 16.11.10.1w
build-dkms.sh: building DKMS structure in /home/pi/hides-drivers/linux-drivers/tmp/hides-dkms-16.11.10.1w/hides-16.11.10.1w
build-dkms.sh: creating /home/pi/hides-drivers/linux-drivers/packages/hides-dkms-16.11.10.1w.tgz
build-dkms.sh: Linux distro: false, full version: false
build-dkms.sh: building DEB package
dpkg-deb: building package 'hides-dkms' in '/home/pi/hides-drivers/linux-drivers/packages/hides-dkms_16.11.10.1w_all.deb'.

 **sudo dpkg -i packages/hides-dkms_16.11.10.1w_all.deb** 
Preparing to unpack .../hides-dkms_16.11.10.1w_all.deb ...
Unpacking hides-dkms (16.11.10.1w) over (16.11.10.1w) ...
Setting up hides-dkms (16.11.10.1w) ...

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KERNELDIR=/lib/modules/5.4.72-v7l+/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.72-v7l+ (armv7l)
Consult /var/lib/dkms/hides/16.11.10.1w/build/make.log for more information.

**The log:**
DKMS make.log for hides-16.11.10.1w for kernel 5.4.72-v7l+ (armv7l)
Sat Nov  7 00:32:43 CET 2020
make -C it950x_driver EXTRA_CFLAGS=-DTSDUCK_WRITE
make[1]: Entering directory '/var/lib/dkms/hides/16.11.10.1w/build/it950x_driver'
make -C /lib/modules/5.4.72-v7l+/build SUBDIRS=/var/lib/dkms/hides/16.11.10.1w/build/it950x_driver modules
make[2]: Entering directory '/usr/src/linux-headers-5.4.72-v7l+'
make[3]: *** No rule to make target 'arch/arm/tools/syscall.tbl', needed by 'arch/arm/include/generated/uapi/asm/unistd-common.h'.  Stop.
make[2]: *** [arch/arm/Makefile:325: archheaders] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.4.72-v7l+'
make[1]: *** [Makefile:18: default] Error 2
make[1]: Leaving directory '/var/lib/dkms/hides/16.11.10.1w/build/it950x_driver'
make: *** [Makefile:5: default] Error 2

omikron88, 2020-11-07:
The pre-built deb package from TSDuck site works the same way.

lelegard, 2020-11-08:
Hi @omikron88 

I can reproduce that on my Raspberry Pi with the same version of the kernel (up-to-date Raspbian). After investigation, it was an issue in the makefile of the drivers from ITE.

I fixed this and updated the packages in https://github.com/tsduck/hides-drivers/releases/

It now builds on the latest versions of Raspbian, Debian, Ubuntu, CentOS and Fedora.

omikron88, 2020-11-09:
it compiles OK now, but when using tsduck, it produces the warning:

* Warning: hides: obsolete HiDes/it950x driver, this version uses polling, risk of performance hit, use version from https://tsduck.io/download/hides/

lelegard, 2020-11-09:
Indeed, the warning was expected. It should not make any difference. I fixed it in the TSDuck code.

FYI, the core reason is the following: The original ITE drivers for Linux (reused by HiDes) used a polling strategy: try to write, fail if internal FIFO not ready, application waits and retry until ok. This is of course awful in a driver and, BTW, the ITE Windows driver did not do that and correctly waits for the internal FIFO. So, I fixed the ITE driver to implement a correct kernel wait on the device internal FIFO. To differentiate the original ITE driver and my fixed one, I added a trailing "w" (for "wait") to the driver version. The TSDuck code checks the driver version and, without trailing "w", the TSDuck code has to implement a wait and retry strategy on failing writes.

In my fix of the driver package, I added ".1" to the version to indicate it is a updated packaging of the "w" driver (and not a new version of the driver itself). So, now the driver version ends in "w.1" which was not correctly recognized as a fixed "wait" driver.

In practice, you may safely ignore the warning since the driver implements the "wait" strategy; it will never fail on "FIFO not ready"; the TSDuck application will never need to use the wait and retry strategy even though it assumes that this driver may fail.

I fixed the TSDuck code so that driver versions ending in "w._number_" are also recognized and valid "wait" versions and TSDuck will no longer issue a warning and will not attempt the wait and retry strategy. If you prefer to use this one, just rebuild TSDuck ([details here](https://tsduck.io/doxy/building.html)).

closed 2022-09-23 by lelegard