tsduck/tsduck #663: [Q] got stuck with a very simple ffmpeg mcast->tsswitch->ffmpeg record config

andrei-ka, 2020-11-12:
hi, 

i need to switch b/t 2 ffmpeg created multicasts, so i first create mcast (~cbr) :

 ffmpeg -loglevel info -re -stream_loop -1 -i slates/x.mp4 -threads 0 -f mpegts -c:a libfdk_aac -ac 2 -af aresample=48000 \
 -c:v libx264 -g 25 -r 25 -tune film -keyint_min 25 -sc_threshold 0 -pix_fmt yuv420p -x264-params nal-hrd=cbr -b:v 800k \
 -s:v 640x360 -preset:v superfast -profile:v high -level:v 4.1 -maxrate:v 800k -bufsize:v 1600k -map 0 -b:a 160k udp://239.10.11.10:1234?ttl=2&localaddr=10.91.4.222

it runs ok 👍  : 

Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'slates/x.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.46.101
    location-eng    : +47.4833+001.8081/
    location        : +47.4833+001.8081/
  Duration: 00:01:00.83, start: 0.000000, bitrate: 1980 kb/s
    Stream #0:0(eng): Video: h264 (High) (avc1 / 0x31637661), yuv420p, 1920x1080 [SAR 1:1 DAR 16:9], 1842 kb/s, 30 fps, 30 tbr, 15360 tbn, 60 tbc (default)
    Metadata:
      handler_name    : VideoHandle
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 128 kb/s (default)
    Metadata:
      handler_name    : SoundHandle
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> h264 (libx264))
  Stream #0:1 -> #0:1 (aac (native) -> aac (libfdk_aac))
Press [q] to stop, [?] for help
[libx264 @ 0x270c940] using SAR=1/1
[libx264 @ 0x270c940] using cpu capabilities: none!
[libx264 @ 0x270c940] profile Progressive High, level 4.1, 4:2:0, 8-bit
Output #0, mpegts, to 'udp://239.10.11.10:1234?ttl=2&localaddr=10.91.4.222':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    location        : +47.4833+001.8081/
    location-eng    : +47.4833+001.8081/
    encoder         : Lavf58.27.100
    Stream #0:0(eng): Video: h264 (libx264), yuv420p, 640x360 [SAR 1:1 DAR 16:9], q=-1--1, 800 kb/s, 25 fps, 90k tbn, 25 tbc (default)
    Metadata:
      handler_name    : VideoHandle
      encoder         : Lavc58.48.101 libx264
    Side data:
      cpb: bitrate max/min/avg: 800000/0/800000 buffer size: 1600000 vbv_delay: -1
    Stream #0:1(eng): Audio: aac (libfdk_aac), 48000 Hz, stereo, s16, 160 kb/s (default)
    Metadata:
      handler_name    : SoundHandle
      encoder         : Lavc58.48.101 libfdk_aac
frame=  347 fps= 25 q=16.0 size=    1796kB time=00:00:14.05 bitrate=1046.6kbits/s dup=0 drop=68 speed=0.997x

----------------------

i can get report on it wxith tsp/analyze : 

tsp -I ip -l 10.91.4.222 239.10.11.10:1234 -P until --seconds 10 -P analyze --service-analysis -O dro
===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 192, PID's: 2 (clear: 2, scrambled: 0)                         |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C       13,060 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C       13,060 b/s  |
|  0x0000  PAT .......................................... C       10,816 b/s  |
|  0x0011  SDT/BAT ...................................... C        2,244 b/s  |
|=============================================================================|
|  Unreferenced PID's                                                         |
[... 194 lines cut ...]

br-72f7afde48c2 Link encap:Ethernet  HWaddr 02:42:79:36:11:54
          inet addr:172.18.0.1  Bcast:172.18.255.255  Mask:255.255.0.0
docker0   Link encap:Ethernet  HWaddr 02:42:12:c4:e7:75
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
em1       Link encap:Ethernet  HWaddr 54:9f:35:04:0e:92
          inet addr:11ffmpe1.111.111.111  Bcast:62.210.245.255  Mask:255.255.255.0
em2       Link encap:Ethernet  HWaddr 54:9f:35:04:0e:94
          inet addr:10.91.4.222  Bcast:10.91.4.255  Mask:255.255.255.192
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:125976913518 errors:0 dropped:1 overruns:0 frame:0
          TX packets:2704565131 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:173611259336824 (173.6 TB)  TX bytes:9570023949568 (9.5 TB)
          Interrupt:17
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.12.1.1  P-t-P:10.12.1.1  Mask:255.255.255.0

=========================

so, can anyone wise me up, what the heck makes tsswitch unhappy ?

regards, &rei

lelegard, 2020-11-12:
Your command is (raw copy/paste, just reformatted):
~~~
tsswitch --debug --remote 4444 --primary-input 0 --fast-switch --infinite \
    -I ip -l 10.91.4.222 239.10.11.10:1234 \
    -I ip -l 10.91.4.222 239.10.11.10:1234 \
    -O ip -l 10.91.4.222 239.10.11.12:1234 --packet-burst 7 --enforce-burst
~~~

So, you read the same stream in the two input plugins. If one single process (here `tsswitch`) reads the same multicast stream twice, are you sure that the same stream is duplicated and fully and identically read in the two sockets? Or are the UDP datagrams randomly distributed between the two, creating two incomplete TS? This is not a TSDuck question, this is an operating system question. I am not sure that the answer is the same on all operating systems.

More generally, multiple multicast streams are safely received when:
- All multicast IP addresses are different in their 23 least significant bits (because of IP-to-MAC mapping). This can be successive IP addresses for simplicity.
- All UDP ports are different.

Thus, if multiple receivers coexist in the same machine, they will send unique join group requests per stream and each socket is bound to a port that is unique to a stream.

When the two streams use different IP multicast addresses and different UDP ports (for instance 239.10.11.10:1234 and 239.10.11.11:1235), does `tsswitch` correctly forward a complete TS?

andrei-ka, 2020-11-12:
it's the same pb with 2 mcas ip (i've stated it : "i've tried 2 mcas sources .10 and .11 its the same")...

see the traces : 

+ echo mcast1 239.10.11.10:1234
mcast1 239.10.11.10:1234
+ tsp -I ip -l 10.91.4.222 239.10.11.10:1234 -P until --seconds 10 -P analyze --service-analysis -O drop

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 197, PID's: 2 (clear: 2, scrambled: 0)                         |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C       11,943 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C       11,943 b/s  |
|  0x0000  PAT .......................................... C       10,064 b/s  |
|  0x0011  SDT/BAT ...................................... C        1,879 b/s  |
|=============================================================================|
|  Unreferenced PID's                                                         |
|  TS packets: 1, PID's: 1 (clear: 1, scrambled: 0)                           |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Unreferenced PID's ........................... C           60 b/s  |
|  0x1A56  Unreferenced ................................. C           60 b/s  |
|=============================================================================|
|  Service: 0x0001 (1), TS: 0x0001 (1), Original Netw: 0xFF01 (65281)         |
|  Service name: Service01, provider: FFmpeg                                  |
|  Service type: 0x01 (Digital television service)                            |
|  TS packets: 6,180, PID's: 3 (clear: 3, scrambled: 0)                       |
|  PMT PID: 0x1000 (4096), PCR PID: 0x0100 (256)                              |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Digital television service ................... C      374,677 b/s  |
|  0x0100  AVC video .................................... C      303,379 b/s  |
|  0x0101  MPEG-2 AAC Audio (eng) ....................... C       61,900 b/s  |
|  0x1000  PMT .......................................... C        9,397 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
===============================================================================

+ echo mcast2 239.10.11.11:1234
mcast2 239.10.11.11:1234
+ tsp -I ip -l 10.91.4.222 239.10.11.11:1234 -P until --seconds 10 -P analyze --service-analysis -O drop

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 197, PID's: 2 (clear: 2, scrambled: 0)                         |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C       11,083 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C       11,083 b/s  |
|  0x0000  PAT .......................................... C        9,339 b/s  |
|  0x0011  SDT/BAT ...................................... C        1,744 b/s  |
|=============================================================================|
|  Unreferenced PID's                                                         |
|  TS packets: 1, PID's: 1 (clear: 1, scrambled: 0)                           |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Unreferenced PID's ........................... C           56 b/s  |
|  0x1A56  Unreferenced ................................. C           56 b/s  |
|=============================================================================|
|  Service: 0x0001 (1), TS: 0x0001 (1), Original Netw: 0xFF01 (65281)         |
|  Service name: Service01, provider: FFmpeg                                  |
|  Service type: 0x01 (Digital television service)                            |
|  TS packets: 6,150, PID's: 3 (clear: 3, scrambled: 0)                       |
|  PMT PID: 0x1000 (4096), PCR PID: 0x0100 (256)                              |
|-----------------------------------------------------------------------------|
[... 56 lines cut ...]
* Debug: ip[0]: no TS packet in message, 32 bytes
* Debug: ip[1]: no TS packet in message, 32 bytes
* Debug: ip[1]: no TS packet in message, 32 bytes
* Debug: ip[1]: no TS packet in message, 32 bytes
* Debug: ip[0]: no TS packet in message, 32 bytes
* Debug: ip[0]: no TS packet in message, 64 bytes
* Debug: ip[1]: no TS packet in message, 64 bytes

[... cut ...]
ben, si tu vx tester, mes  4 scripts (très bateau :-) sont  ici :  
https://drive.google.com/file/d/1Fsg_5tvuVUYmwnIgbH2V3h4oOdzJv_dg/view?usp=sharing

il faut redefinir source mp4 (slate/mp4) et virer local addr ou le mettre à 127.0.0 (:1) si t'as qu'un eth. 

il faut lancer avec l'arg 1 (au depart c'était utilisé ds docker-compose)...

./slate.sh 1
./slate2.sh 1
./tsswitch.sh 1
et ./recoord... 1

fin, je pense pas que c'est un bug (ni de ffmpeg ni de tsduck), c'est un paramétrage à trouver, mais moi je n'arrive pas... 

cdlt, &rei

andrei-ka, 2020-11-12:
>More generally, multiple multicast streams are safely received when:
>All multicast IP addresses are different in their 23 least significant bits (because of IP-to-MAC mapping). This can be successive IP addresses for simplicity. All UDP ports are different.

a titre d'info, j'ai +70 multicasts de pixagility et arkena qui arrivent sur le routeur, leurs ip changent (+1), les ports sont les mêmes. changer juste l'ip ca suffit...

lelegard, 2020-11-12:
> it's the same pb with 2 mcas ip (i've stated it : "i've tried 2 mcas sources .10 and .11 its the same")...

It is unclear if you used twice .10 and then twice .11 or if you used .10 and .11 in the same command. The problem with your command is that you listen to .10 _twice in the same command_, which means that the two plugins compete to read the same stream. Please try `tsswitch  -I ip ....10:1234 -I ip .....11:1234 ...`.

> a titre d'info, j'ai +70 multicasts de pixagility et arkena qui arrivent sur le routeur, leurs ip changent (+1), les ports sont les mêmes. changer juste l'ip ca suffit...

Yes, the IP addresses are more important because they generate distinct join requests. Using the same port works but the performances are poor if the same receiver machine receives more than one stream because each socket being bound to the same common port, it receives all streams for which a join request was issued. The application has to manually filter each packet to select the ones from the appropriate source (and this is exactly what TSDuck does). This is why it is better to use distinct ports as well, even though it stills works with the same port.

lelegard, 2020-11-12:
The other important point is the messages "Debug: ip[0]: no TS packet in message, 64 bytes". This means that some input datagrams do not contain any TS packet. In fact, 64 bytes is too short. In general, IP TV packets contain 7 TS packets (1316 bytes) plus headers and optional RTP layer. Such short packets have nothing to do in an IP TV stream. It looks like your IP multicast stream does not contain TS packets, or not only TS packets. Could you check with Wireshark? Any other unrelated multicast on the same address in the network?

The `tsp` command with `-P analyze` seems to analyse the stream correctly (or at least receives TS packets, which does not mean that the stream is fully consistent). Does it display the same "no TS packet in message" errors? If this is the same stream, it should because the exact same `ip` plugin is used in `tsp` and `tsswitch`.

andrei-ka, 2020-11-12:
> The problem with your command is that you listen to .10 twice in the same command, which means that the two plugins >compete to read the same stream. Please try tsswitch -I ip ....10:1234 -I ip .....11:1234 ....

you can see in the 2nd log 2 different mcast ips (this 2nd log is *another run*, and yes ffmpeg is still bitching about virtyally 50% of the output stream (.12 ip^) courrupt): 

tsswitch --debug --remote 4444 --primary-input 0 --fast-switch --infinite -I ip -l 10.91.4.222 **239.10.11.10:**1234 -I ip -l 10.91.4.222 **239.10.11.11**:1234 -O ip --packet-burst 7 --enforce-burst -l 10.91.4.222 239.10.11.12:1234
tsswitch: Debug: debug level set to 1
tsswitch: Debug: ====> tsswitch --debug --remote 4444 --primary-input 0 --fast-switch --infinite
Debug: debug level set to 1
Debug: ip: debug level set to 1
Debug: ip: ====> tsswitch -O ip --packet-burst 7 --enforce-burst -l 10.91.4.222 239.10.11.12:1234
Debug: ip: debug level set to 1
Debug: ip: ====> tsswitch -I ip -l 10.91.4.222 **239.10.11.10**:1234
Debug: ip[0]: debug level set to 1
Debug: ip: debug level set to 1
Debug: ip: ====> tsswitch -I ip -l 10.91.4.222 **239.10.11.11**:1234

> The tsp command with -P analyze seems to analyse the stream correctly (or at least receives TS packets, which does not mean that the stream is fully consistent).

well, somehow ffmpeg considers these ts streams pretty consistent and can transcode 'em before i make 'em pass through tsswitch...

lelegard, 2020-11-13:
There is no difference in the code between `tsp` and `tsswitch` in the way UDP packets are analyzed, this is the same `ip` plugin. The difference with `tsswitch` _in your case_, is that the two intances of the `ip` plugin open two distinct sockets _on the same UDP port_ because you multicast all your streams on the same port. As I wrote twice, this works fine as long as each receiver machine listens to one single stream, meaning that each receiver machine has one single socket on port 1234. 
You mention that all reception tests using `ffmpeg` work fine. But in all those cases, there is only one receiver per UDP port, not two as in the case of `tsswitch`.

> yep, tsp with debug is also printing too short packets lengths warnings, but ends up decoding pat/sdt

> somehow ffmpeg considers these ts streams pretty consistent and can transcode 'em 

It is absolutely not normal that an IP TV stream contains short datagrams without TS packets. So, there is definitely something wrong in this stream. The fact that `tsp` can extract TS packets to analyze the stream does not mean that there is no garbage in addition to the TS stream. The `ip` plugin is designed to extract TS packets from UDP datagrams regardless of any encapsulation. So, it is pretty robust. If no TS packet can be extract, then there is that debug message.

You must analyze the _actual content of the stream_, not only the result of what ffmpeg is doing with that stream. In other words, analyze the cause, not the symptom. This is why I advised to use Wireshark. This is the only way to know what the UDP packets contain exactly. I suggest to capture a part of the stream using Wireshark and make sure that `tsp` or  `tsswitch` reports the "no TS packets" during this recording. Then, please post the corresponding pcap file here. We will see what the UDP datagrams contain.

If you want to have a look at how TSDuck receives and processes UDP datagrams, see [here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/base/network/tsUDPReceiver.cpp) for the UDP reception and [here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/plugin/tsAbstractDatagramInputPlugin.cpp) for the TS packet extraction. This is surprisingly not as trivial as it seems.

andrei-ka, 2020-11-13:
ok, it was stupid me, forgot to to set output packet size :-) it's all clean now. 

output should be set as udp://IP:PORT?ttl=TTL&localaddr=L_IP&pkt_size=1316

and when it works just great.  tnx for bearing w/ dumme me !

ffmpeg -loglevel info -re -stream_loop -1 -i slates/x.mp4 -threads 0 -f mpegts -c:a libfdk_aac -ac 2 -af aresample=48000 -c:v libx264 -g 25 -r 25 -tune film -keyint_min 25 -sc_threshold 0 -pix_fmt yuv420p -x264-params nal-hrd=cbr -b:v 800k -s:v 640x360 -preset:v superfast -profile:v high -level:v 4.1 -maxrate:v 800k -bufsize:v 1600k  -muxrate 1400k -map 0 -b:a 160k udp://239.10.11.11:1234?ttl=2&localaddr=10.91.4.222&pkt_size=1316&ttl=2

regards, &rei

andrei-ka, 2020-11-13:
>If you want to have a look at how TSDuck receives and processes UDP datagrams, see here for the UDP reception and here for >the >TS packet extraction. This is surprisingly not as trivial as it seems.

nothing is trivial after you have to manage all kinds of wierd situations et errors... :-) 

btw, ffmpeg can mux in  'pro_mpeg' with fec sent on 2 extra ports (http://underpop.online.fr/f/ffmpeg/help/prompeg.htm.gz),
wonder if it's possible to switch b/w that kind of streams...

lelegard, 2020-11-13:
> if you're oldschool and i bit paranoid about all wierd kind of things that can happen

If "oldschool" means "robust code", then I am both indeed....

>  ffmpeg can mux in 'pro_mpeg' with fec sent on 2 extra ports (http://underpop.online.fr/f/ffmpeg/help/prompeg.htm.gz), wonder if it's possible to switch b/w that kind of streams...

This has been asked at least twice, latest in #627. This is currently not possible. This is a time consuming task and I have no time and no incentive to do it.