tsduck/tsduck #664: tsterinfo -b fails to calculate parameters due to undefined DeliverySystem

Foorack, 2020-11-17:
#### Bug description:
`tsterinfo -b` fails to calculate modulation parameters given a specific bitrate on latest master.

The [call to theoreticalBitrate()](https://github.com/tsduck/tsduck/tree/master/src/tstools/tsterinfo.cpp#L356) always returns 0 due to DeliverySystem argument not being set in tsterinfo.cpp and thereby [defaults to DS_UNDEFINED](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/tsModulationArgs.cpp#L385). This causes the switch statement to skip computation of the theoretical bitrate, and prints every possible combination as having a bitrate of zero.

#### How to reproduce:

1. git clone https://github.com/tsduck/tsduck.git && cd tsduck
2. make -j 10
3. ./bin/*/tsterinfo -b 18,096,256

#### Expected behavior:
Expected was the closest DVB-T parameters being printed.

#### Errors and logs:
None.

#### Environment:
 - OS: Linux
 - OS version: Ubuntu 18.04
 - TSDuck full version: 3.24-2067
 - Installation type: rebuilt binary from latest master

#### Additional information:

lelegard, 2020-11-17:
Hi @Foorack 

Thanks for reporting this and sorry for the inconvenience. This is of course a bug, a regression which was introduced some time ago when there was some substantial refactoring on modulation parameters when ISDB was added. Unfortunately, there was no non-regression test on this specific feature.

I fixed this is commit 71f60eead055d1ddada94b3f84971bb147d3664f and also added a non-regression test to make sure it will never reappear.

Foorack, 2020-11-18:
Verified fixed. ✅ Thank you. Closing.

closed 2020-11-18 by Foorack