tsduck/tsduck #665: merge and tsswitch

Worker35, 2020-11-19:
is-it possible to use a tsswitch just behind a merge. In fact I want to use the result of the switch to merge it in a MPTS.
Thanks

lelegard, 2020-11-20:
The `merge` plugin uses the output of the command it launches. So, to use the output of a `tsswitch` command as input of a `merge` plugin, the global command should look like this:
~~~
tsp ... -P merge "tsswitch -I ... -I ..." ...
~~~
Note the quote around the complete `tsswitch` command because the whole command is a parameter of the `merge` plugin.

Of course, if the `tsswitch` command is long, I would recommend to either write it in a shell script and run that shell script in the `merge` plugin or use the TSDuck partial command line redirection from a file (see the TSDuck user's guide, section 3.1.5).

Worker35, 2020-11-20:
Perfect! it works very well. (a quote forgotten in a quite long command )
Thanks!

closed 2020-11-20 by Worker35