tsduck/tsduck #666: [Q] Windows IP Multicast, no output? Won't terminate?

Napsterbater, 2020-11-20:
Hi, I am attempting to use your program to troubleshoot some IPTV issues.

I am running this on windows, and for example when I the following command, I never seem to get any output in the console window, and also when I attempt to crtl-c, TSP never seems to actually terminate no matter how long I wait, I have ukill it via task manager.

I can confirm the multicast group is joined via IGMP and data is coming in via wireshark capture on the interface.

I sorry if I am missing something obvious.

```
C:\util\TSDuck\bin>tsp --debug -I ip -l 10.20.64.211 239.21.10.22:50000 -P until --seconds 10 -P analyze --interval 1 --service-analysis -O drop
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug
* Debug: debug level set to 1
* Debug: ip: debug level set to 1
* Debug: ip: ====> tsp -I ip -l 10.20.64.211 239.21.10.22:50000
* Debug: registering processor plugin "until", status: ok
* Debug: drop: debug level set to 1
* Debug: drop: ====> tsp -O drop
* Debug: registering processor plugin "analyze", status: ok
* Debug: trying to load "C:\util\TSDuck\bin\tsplugin_until.dll"
* Debug: until: debug level set to 1
* Debug: until: ====> tsp -P until --seconds 10
* Debug: trying to load "C:\util\TSDuck\bin\tsplugin_analyze.dll"
* Debug: analyze: debug level set to 1
* Debug: analyze: ====> tsp -P analyze --interval 1 --service-analysis
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: ip: setting socket reuse address to 1
* Debug: ip: binding socket to 10.20.64.211:50000
* ip: joining multicast group 239.21.10.22:50000 from local address 10.20.64.211
* tsp: user interrupt, terminating...
```

lelegard, 2020-11-20:
Hi @Napsterbater 

I tried the very same command (local interface address excepted of course) on a Windows 10 system and it works. The IP stream was sent from a Linux VM.

So, there must be something on your Windows system that prevents the application from receiving the packets.

Try `--debug=2`. At this level, all incoming UDP packets are traced like this:
~~~
* Debug[2]: ip: received UDP packet, source: 192.168.56.108:42200, destination: 239.21.10.22:50000, timestamp: -1
~~~

But, because everything in `tsp` is triggered on TS packet processing, I suspect that you don't receive any UDP packet for two reasons:
- The plugin `until` should trigger a termination on the reception of a TS packet exactly 10 seconds after receiving the first one. If there is no termination, there is probably no packet flowing through the chain of plugins.
- The Ctrl-C interrupt is caught by `tsp` and synchronously processed on the next TS packet processing to trigger a graceful termination. Again, if it does not terminate, there is probably no packet. In this case, a second Ctrl-C abruptly terminates the process (regular system interrupt). This works on Linux and macOS but not on Windows for some usual mysterious Windowish reason. But this specific Windows issue is of course unrelated to your UDP reception.

Napsterbater, 2020-11-20:
Looks like you were right on both accounts.

Windows firewall, tsp was allowed for the Domain profile,, but not for Public and Private, which the 2nd NIC that I am using to the data with was attached.

Sorry for the noise on a simple problem, But thanks for the quick response.

Napsterbater, 2021-01-24:
Forgot to close this.

closed 2021-01-24 by Napsterbater