tsduck/tsduck #667: ISDB-Tb time fields are UTC-3 and not UTC

nicst35, 2020-11-23:
As @lelegard said in #602, ISDB chose to reuse parts of DVB while adding modifications like using JST instead UTC. During its first years it served for their purpose as it was only used at Japan.
When Brasil decided to use ISDB as the base for their digital terrestrial TV system that later was standardized as ISDB-Tb, they chose to deviate even further by not using UTC but... UTC-3! (Brasilia Time). All South American ISDB-Tb deployments are based in this standard (ABNT NBR-15603-2, see references)

With the current version of TSDuck, using `tsanalyze` or `tsdate` on ISDB-Tb streams TS1globo.ts or TS2band.ts available on website will report semantically invalid time fields. For example:

    $ tsdate TS1globo.ts
    * TOT UTC time: 2015/12/07 10:02:46.000
      Country: BRA, region: 3
      Local time:   2015/12/07 10:02:46.000, local time offset: 00:00
      Next change:  2012/02/18 00:00:00.000, next time offset:  00:00
    ...
    $ tsdate -v TS1globo.ts
    * TOT, TID 115 (0x73), PID 20 (0x0014)
      Short section, total size: 29 bytes
      - Section 0:
        UTC time: 2015/12/07 10:02:46
        - Descriptor 0: Local Time Offset (0x58, 88), 13 bytes
          Country code: BRA
          Region id: 3 (0x03), polarity: east of Greenwich
          Local time offset: 00:00
          Next change: 2012/02/18 00:00:00
          Next time offset: 00:00
        CRC32: 0x72901546 (OK)
    ...
If the TS is played at 01:30, one can see that the original local time was 10:04 = 13:04 UTC.

I see two alternatives to fix this issue:
* Internally convert UTC-3 to UTC
    * Displayed UTC is the correct UTC
    * External programs expecting UTC will get the correct time
    * Local time offsets will require conversion to keep local times correct.
* Keep UTC-3 as-is internally but change output texts (UTC/UTC-3/JST) depending of the detected standard or what is provided as an assumption via command line
    * Local time conversions are straightforward
    * Processors and external programs expecting real UTC may misbehave.

According to DiBEG, ISDB-T deployments outside Japan (using JST) and South America (using UTC-3) like the ones made in Botswana, Philippines and Maldives chose to use their local times as the time reference, complicating this situation even further.

## References
http://www.telemidia.puc-rio.br/~rafaeldiniz/public_files/normas/SBTVD/pt_BR/ABNTNBR15603-2_2007Vc_2008Ing.pdf - 2008 version
https://www.abntcatalogo.com.br/norma.aspx?ID=329893 - 2015 version (requires registration)
(Both documents are English versions.)

lelegard, 2020-11-25:
Hi @nicst35 

Thanks for reporting this. It's good to have a contact in Brazil for ISDB-Tb specificities. Currently, TSDuck testers for ISDB  were all in Japan. With their help, TSDuck implemented Japanese specificities. Character set support was the most difficult part. Let's now do the same thing with ISDB-Tb, in Brazil and elsewhere.

It is clear that ISDB people had very little consideration for the concept of "international standard" and interoperability in a connected world. They invented their own world, reusing parts of DVB while changing the semantics of SI (and even the syntax sometimes!)

Concerning the time reference, there are two different contexts: TOT/TDT and EIT.

In TOT/TDT, the field is named _UTC_time_ and should be consequently understood as UTC. Note that the ABNT standard dares to write _"UTC_time: [...] current time and date in Brazil (UTC-3)"_ (respectively JST in the ARIB version). So, TSDuck tries to adjust the time to UTC to get a consistent _UTC_time_ field everywhere.

In several commands, the option `--japan` was added to indicate some presets for Japan. First, the options `--isdb --default-charset ARIB-STD-B24 --hf-band-region japan` are automatically set. Second, the time in TDT/TOT is interpreted as JST. The field `utc_time` in `EIT` C++ objects and the corresponding attribute in `<EIT>` from XML files were automatically updated back and forth. So, a _UTC_time_ is always UTC in C++ and XML but the binary serialization and deserialization are affected by option `--japan`.

Note that there were two restrictions with option `--japan`. First, it was missing from the `tsdate` command. Second, the UTC adjustment was not done when displaying TDT/TOT with `tstables` and similar commands. I just fixed those two bugs.

I understand your concern about ISDB-Tb. So, I generalized the UTC adjustment using a new option `--time-reference` which specifies the type of adjustement from UTC to apply to TDT/TOT. The default is none, as specified by DVB. The help text is:
~~~
--time-reference name
    Use a non-standard (non-UTC) time reference in TDT/TOT. This is
    typically used in ARIB ISDB and ABNT ISDB-Tb standards. The specified
    name can be either 'UTC', 'JST' (Japan Standard Time) or 'UTC+|-hh[:mm]'.
    Examples: 'UTC+9' (same as 'JST' for ARIB ISDB), 'UTC-3' (for ABNT
    ISDB-Tb in Brazil).
~~~

As a convenience, the field `default.time` has been added in the user's configuration file to allow a global default for users always working on TS with a specific non-standard time reference.

So, in an ISDB context, three possibilities:
- Option `--japan` now means `--isdb --default-charset ARIB-STD-B24 --hf-band-region japan --time-reference JST`.
- The new option `--brazil` means `--isdb --time-reference UTC-3`.
- A specific `--time-reference` can be used anywhere else.

In EIT, the time field is named _start_time_. The description of the semantics differs in the various standards but the field name does not enforce a specific one. Event information is just informational, typically for an EPG. Because this is a very "local" concept without impact on the stucture of the transport stream, TSDuck does not interpret it. The stored value is identically used, without particular semantics, in the `EIT` C++ objects, in the `<EIT>` in XML files and displayed by `tstables` and similar commands. The options `--japan`, `--brazil` and `--time-refernce` do not affect the interpretation of the events start time.

Please provide any useful feedback on TSDuck for ISDB in South America. Did you have the opportunity to test ISDB tuners or modulators with TSDuck?

nicst35, 2020-11-27:
Thanks. Your commit looks more complete than the hack I was going to send as a PR.

>Note that the ABNT standard dares to write _"UTC_time: [...] current time and date in Brazil (UTC-3)"_

Note that the Spanish and Portuguese(the original one) versions put that field as `UTC-3_time`. There are slight differences between them (along with PDF rendering errors).
(http://www.telemidia.puc-rio.br/~rafaeldiniz/public_files/normas/SBTVD/)

> So, I generalized the UTC adjustment using a new option --time-reference which specifies the type of adjustement from UTC to apply to TDT/TOT

This creates a problem with `ts::TOT::localTime()`. Since it acts on UTC time, should it apply the inverse offset along with the local offset? `tsanalyze` is now reporting an incorrect TOT local time 

>The new option `--brazil` means `--isdb --time-reference UTC-3`

It should also imply `--default-charset ISO-8859-15`.

>In EIT, the time field is named start_time. The description of the semantics differs in the various standards but the field name does not enforce a specific one.

_Semantics of information for event information section shall be according to EN 300 468:2007, subclause 6.2.4; except for the fields below:
⎯ table_id: see Table 6;
⎯ start_time: 40-bit field that shall contain the start time of the event in the UTC-3 and the date in “modified julian date” (MJD) (see Annex A)…_

- My country (CL) is on DST right now so our local time is UTC-3, so EIT times would match the real ones.
- The only two broadcasters in my zone that are transmitting digital TV rushed the deployment so no EIT for now.

A good way to check it would be getting a TS from a peruvian broadcaster, since they use UTC-5 (and -4 during summer) so they should always have an offset, and I am assuming they didn't change the standard.

lelegard, 2020-11-27:
Hi @nicst35 

Thanks for your feedback. With your help, I will try to make TSDuck as compliant as possible with ISDB-Tb.

> Note that the Spanish and Portuguese(the original one) versions put that field as UTC-3_time. There are slight differences between them (along with PDF rendering errors).

Inconsistencies between translated versions is a pain. I can't speak Portuguese and I lost all Spanish I learnt at school, so I can only use the English versions. It was worse with ARIB since English versions are outdated and don't even exist for some documents. I had to rely on Japanese users for help (as well as Google Translation for a few specific differences in the most recent and Japanese-only versions of the standard). All technical international standards should be written in English, full stop, regardless of little national susceptibilities. English is the de-facto reference technical international language, the only language which is understood by all engineers all over the world, just as Latin was in the middle age. And I say that as a Frenchman, not a native English speaker ;) But I am so often pissed off by those little French susceptibilities on linguistic pride...

> This creates a problem with ts::TOT::localTime(). Since it acts on UTC time, should it apply the inverse offset along with the local offset?

Yes, that makes sense. The _local_time_offset_descriptor_ is defined as "offset from UTC" by DVB but should be intepreted here as "offset from whatever is in the field misnamed UTC_time".

Just for the fun, look at what is written in the ABNT English version for _local_time_offset_descriptor_:

> local_time_offset_polarity: [...] When set to "0", the local time shall advance from UTC time
> local_time_offset: [...] current offset time from UTC-3

UTC in one case, UTC-3 the other one...

Anyway, I fixed `ts::TOT::localTime()` as you suggested.

> It should also imply --default-charset ISO-8859-15.

We should discuss this more in details. I do not see any precise description of how strings should be handled in the ABNT documents. In the ARIB documents, it is clear that everything should be ARIB STD-B24. But this useful for the Japanese language only.

Searching for "8859" in ABNT NBR 15603-1, -2 and -3, I can find references to ISO-8859-15 in the various _ISO_639_language_code_ fields, _country_code_ fields and the text fields in the _short_event_descriptor_ and _extended_event_descriptor_

In the case of _ISO_639_language_code_ and _country_code_, this is useless. Quote:

> ISO_639_language_code: [...] specified by ISO 639-2. Each character is encoded [..] in accordance with ISO 8859-15
> country_code: [...] specified in ISO 3166-1. Each character will be coded [...] according to ISO 8859-15

Both ISO 639-2 and ISO 3166-1 specify simple short strings with ASCII characters only. The original standards which define them, ISO 15818-1 and ETSI EN 300 468 specify the use of ISO 8859-1. Since ISO 8859-1 and 8859-15 are supersets of ASCII and all language or country codes use ASCII characters only, we can safely ignore this.

Concerning text fields in the two event descriptors, this is understandable. However, there are other types of strings in ISDB SI. Just a few examples: _service_descriptor_, _network_name_descriptor_, _bouquet_name_descriptor_, etc. Those strings are obviously very "local" names with possibly specific characters. I assume that they use ISO 8859-15 by default as well. However, this is not clearly specified.

The question should be then raised for other uses of ISDB, outside Japan and South America. ISO 8859-15 is the preferred encoding for western european languages (Spanish, Portuguese, French). So it makes sense as default for South America. Some French satellite providers even violate DVB standards and use it as default, without leading charset code (see option `--europe` in TSDuck). But what about Asian or African countries using ISDB?

Finally, with `--brazil`, shall we use charset `ISO-8859-15` or `RAW-ISO-8859-15`? The difference is subtle but important. With DVB methods, there are various ISO-xxx charsets (plus UTF-8 and -16). The ISO-xxx charset is selected by the first one or two bytes (if lower than 0x20). Without such leading byte, the default is ISO-6937 (or, more precisely, a slightly modified version of it). The names `ISO-xxx` in option `--default-charset` designate the _default_ charset but it does not prohibit the usage of other charsets.

So, with `--default-charset ISO-8859-15`, if the first byte is greater than 0x1F, then ISO-8859-15 is used as default instead of ISO-6937. But if the string starts with a binary sequence to select another charset, then the regular DVB rules apply.

This is for decoding strings. Now, for encoding strings, TSDuck tries a few charsets to see which one can fit. In fact, not all characters can be encoded in each ISO-xxx. You have to try until one works. With `--default-charset ISO-8859-15`, the option indicates which charset to try first. But if it does not work (because some characters in the string cannot be encoded with that charset), it will try others. An in all cases, the proper binary selection sequences are correctly generated. If all charsets fail, then TSDuck falls back to UTF-8 which can encode everything.

Now come the `RAW-ISO-xxx` charsets. Here, we leave the DVB scheme with its optional initial selection sequences. The complete binary sequence in the string is unconditionally interpreted (deserialization) or genera [... cut ...]
To summarize, ARIB and ABNT standards are very hard to understand when you are used to the strictness and precision of ISO, ITU and ETSI (DVB) documents because of the uncertainty, contradictions and lack of strictness.

> Semantics of information for event information section 

About EIT, the _start_time_ remains as it is, without specific semantics. So, the displayed time (as well as values in XML files and C++ objects) shall be interpreted according to the local standard. This was already the case with ARIB. So, I did not change it for ABNT. Do you think we should change anything?

lelegard, 2020-11-27:
To illustrate the various character sets, we can use the command `tscharset`, a useful debug tool I added while working on ARIB STD-B24 support.

By default, DVB uses ISO-6937. Let's start with pure ASCII codes:
~~~
$ tscharset --encode "-c-a-e-i-e-"
2D 63 2D 61 2D 65 2D 69 2D 65 2D
~~~
This is encoded as default ISO-6937, using pure ASCII codes.

Let's add a special character 'µ' which is available in ISO-6937:
~~~
$ tscharset --encode "-c-a-e-i-µ-"
2D 63 2D 61 2D 65 2D 69 2D B5 2D
~~~
We are still in the default character set ISO-6937 with 'µ' being encoded as B5.

Now let's use western european characters:
~~~
$ tscharset --encode "-ç-à-é-ï-è-"
0B 2D E7 2D E0 2D E9 2D EF 2D E8 2D
~~~
This time, some characters are not representable as ISO-6937 and TSDuck automatically switched to ISO-8859-15 (leading binary sequence 0B as specified in section A.2 of ETSI EN 300 468).

Now let's try to use ISO-8859-1 by default.
~~~
$ tscharset --encode "-ç-à-é-ï-è-" --default-charset ISO-8859-1
10 00 01 2D E7 2D E0 2D E9 2D EF 2D E8 2D
~~~
This works because all characters are representable in ISO-8859-1 but in a more complicated way (the leading sequence for ISO-8859-1 is 10 00 01).

We can verify that this alternate ISO-8859-1 encoding is correctly decoded:
~~~
$ tscharset --decode "10 00 01 2D E7 2D E0 2D E9 2D EF 2D E8 2D"
-ç-à-é-ï-è-
~~~

Now let's try to use ISO-8859-2 by default or forcing the default ISO-6937.
~~~
$ tscharset --encode "-ç-à-é-ï-è-" --default-charset ISO-8859-2
0B 2D E7 2D E0 2D E9 2D EF 2D E8 2D
$ tscharset --encode "-ç-à-é-ï-è-" --default-charset ISO-6937
0B 2D E7 2D E0 2D E9 2D EF 2D E8 2D
~~~
Nope, some characters are not representable and TSDuck selected ISO-8859-15 (leading 0B).

Now, if we use the RAW- versions instead, we force to use a specific charset, without adding any DVB-compliant leading binary selection sequence and omit characters which cannot be represented in the forced charset: 
~~~
$ tscharset --encode "-ç-à-é-ï-è-" --default-charset RAW-ISO-8859-2
2D E7 2D 2D E9 2D 2D 2D
$ tscharset --encode "-ç-à-é-ï-è-" --default-charset RAW-ISO-6937
2D 2D 2D 2D 2D 2D
~~~
In the case of RAW-ISO-8859-2, the characters 'à', 'ï' and 'è' are not representable and deleted. In the case of RAW-ISO-6937, no special character can be represented and only the dashes remain.

nicst35, 2020-11-28:
I think `RAW-ISO-8859-15` should be the charset to use for, because of what I read from the standard (and the operational guidelines), other charsets are not considered.

>But what about Asian or African countries using ISDB?

They simply "patched" the standard and put their preferred encoding.
https://ntc.gov.ph/wp-content/uploads/2018/MC/MC-07-12-2014-Attachment.pdf, page 15. They chose UTF-8 for SI tables.
https://www.bocra.org.bw/sites/default/files/documents/ISDB-T%20Standard%20for%20Botswana%20-%20ISDB-Tbw.pdf, not as radical as the first one but they use UTF-8 for captions/subtitles and data broadcasting (BML).

lelegard, 2020-11-28:
> I think RAW-ISO-8859-15 should be the charset to use

Alright.

I started to have a look at the Philippines document. I noticed the UHF frequency mapping and updated the TSDuck HF bands file.

Do you know what is the UHF frequency mapping over South America, country by country? Any difference between countries? The required information is summarised here in the new entry for the Philippines:
~~~
  <hfband type="UHF">
    <region name="philippines"/>
    <channels first_channel="14" last_channel="44" base_frequency="473,000,000" channel_width="6,000,000"/>
  </hfband>
~~~
For countries or regions with identical mapping, several names can be specified.

nicst35, 2020-11-30:
Hi @lelegard.

I think the ABNT frequency table should be used, at least for South America, that means:

    <hfband type="VHF">
        <region name="abnt"/> 
        <channels first_channel="7" last_channel="13" base_frequency="177,142,857" channel_width="6,000,000"/>
    </hfband>
    ...
    <hfband type="UHF">
        <region name="abnt"/>
        <channels first_channel="14" last_channel="36" base_frequency="473,142,857" channel_width="6,000,000"/>
        <channels first_channel="38" last_channel="69" base_frequency="617,142,857" channel_width="6,000,000"/>
    </hfband>

Taken from https://www.abntcolecao.com.br/normavw.aspx?ID=1162, pages 58-59 (PDF, not the printed ones)
Now for some country-specific allocations.

- Chile: https://www.cntv.cl/wp-content/uploads/2020/11/CNTV-Decreto-que-contiene-CNTV-Plan_de_radiodifusion_televisiva-167_y_95.pdf, page 11,
    - Channel number, frequency range and central carrier frequencies are described. Even if it's a subset of the ABNT channels, frequencies match perfectly.
- Argentina: https://www.enacom.gob.ar/multimedia/normativas/2013/Resolucion-7_13-Anexo%203.pdf, page 7.
    - Same situation as Chile, only a subset of the whole bands used but frequencies are the same.
    - A related document describes something called SATVD (Argentinian Digital TV System), basically provisions for 576/50i and p, 720/50p and 1080/25p and 50i broadcasting. Full text is available at https://www.enacom.gob.ar/multimedia/normativas/2013/Resolucion-7_13-Anexo%201.pdf (warning: Spanish-only).
- Ecuador: https://www.arcotel.gob.ec/wp-content/uploads/2016/03/norma-television-digital-terrestre.pdf, page 7, same frequencies, but with gaps.
- Paraguay: https://www.conatel.gov.py/conatel/wp-content/uploads/2019/10/rd-1599-2016-160907-modificacion-norma-tecnica-servicio-television.pdf, pages 1 and 3.

Note that channel 37 is reserved for radio astronomy here and can't be used. The frequencies are the same, but the lowest and highest usable channels are different. Personally I think putting every country in the frequency mapping is overkill and the general frequency plan should be included in the table.

lelegard, 2020-11-30:
Hi @nicst35 

Thank a lot for all these reference information. Spanish documents are good enough for frequency tables and some useful comments are phonetically so close to French that I can easily understand the idea.

I have updated the HF band files in TSDuck.

> Personally I think putting every country in the frequency mapping is overkill and the general frequency plan should be included in the table.

It is not a big deal to add different sections for different countries. Even if the general frequency layout is the same all over South America, I did it precisely because the number of allocated channels is different by country. This makes a huge difference when using `tsscan`. This is usually a lengthy process and avoiding unused channels in a specific country is usually appreciated.

I noted that Botwana uses 8 MHz channels (like DVB countries) while all other ISDB countries use 6 MHz channels.

I think that we now have a fairly complete logical view of ISDB outside Japan.

Do you have any ISDB-T USB tuner to try TSDuck with? It could be nice to have a snapshot of the Chilean terrestrial TV in the TSDuck stream repository.

nicst35, 2020-11-30:
Yes, a Geniatech (MyGica) S270. Works well with minor quirks: OFDM modes and Guard Intervals are reported as "Auto" (they're available via debugfs though).
I can send you two streams (~700MiB) but they're quite basic, only the minimum required SI tables are sent, meaning there's no EIT is available on my area.

If someone living in Santiago can receive a channel transmitting additional tables (EIT, SDTT, etc.), please respond.

lelegard, 2020-12-09:
> Yes, a Geniatech (MyGica) S270. Works well with minor quirks

Does it work with TSDuck on Windows?

I have one too, a generous gift from a Japanese user. But, being in Europe, I have no ISDB-T network to test it and no ISDB-T modulator to build a test bed. I can only see how it is recognized by TSDuck on the various OS.

On Linux, it is correctly reported as an ISDB-T demodulator:
~~~
$ tslsdvb
0: /dev/dvb/adapter0 ("Siano Mobile Digital MDTV Receiver, MDTV Receiver 2.00", ISDB-T)
~~~
But, on WIndows, it is seen as a DVB-T demodulator:
~~~
$ tslsdvb
0: "BDA Tuner Filter" (DVB-T, DVB-T2)
~~~
So, I wonder how it can work on Windows.

SS-FF, 2020-12-09:
Maybe there's something useful: https://github.com/tsduck/tsduck/issues/389#issuecomment-541480324

As mentioned there, so many ISDB-related interfaces exist (https://docs.microsoft.com/en-us/previous-versions/windows/desktop/api/dvbsiparser/nf-dvbsiparser-iisdbterrestrialdeliverysystemdescriptor-gettransmissionmode), that it's pretty unlikely that it's not supported in tuning.

I would start looking at the registry after having confirmed that some other Windows application is working with the tuner.

lelegard, 2020-12-09:
@SS-FF, yes, I have used many useful information you provided for ISDB tuner on Windows (without being able to test in my continent, unfortunately).

All the code is normally in TSDuck. However, on this specific tuner, after connecting the tuner filter to the network provider filter, I query the `ITunerCap` interface of the network provided and get only one network type: `DVBTNetworkProvider` while I would have expected to get `ISDB_TERRESTRIAL_TV_NETWORK_TYPE`.

SS-FF, 2020-12-09:
I don't know anything better than you in that case. 
What I wanted to suggest is to look into the registry where the tuning spaces are defined to (hopefully) get an idea of how it's handled.

SS-FF, 2020-12-09:
(maybe requires that another application has been running before)

SS-FF, 2020-12-09:
Look at the following. I never had any ISDB device installed, and the tuning space is still listed.

And what's interesting: The CLSID for the DVB-T tuning space is the exact same as for the ISDB-T tuning space:

[image]

So most likely, this is all handled just like DVB-T..?

SS-FF, 2020-12-09:
PS: All tuning spaces in that list have the same CLSID. But for DVB-S it's different.

lelegard, 2020-12-09:
> The CLSID for the DVB-T tuning space is the exact same as for the ISDB-T tuning space:

Yes, but the problem is slightly different.

The method `ITunerCap::get_SupportedNetworkTypes` returns network types.

From uuids.h:
~~~
// {216C62DF-6D7F-4e9a-8571-05F14EDB766A}
OUR_GUID_ENTRY(CLSID_DVBTNetworkProvider,
0x216c62df, 0x6d7f, 0x4e9a, 0x85, 0x71, 0x5, 0xf1, 0x4e, 0xdb, 0x76, 0x6a)
~~~
From bdamedia.h:
~~~
#define STATIC_ISDB_TERRESTRIAL_TV_NETWORK_TYPE \
    0x95037f6f, 0x3ac7, 0x4452, 0xb6, 0xc4, 0x45, 0xa9, 0xce, 0x92, 0x92, 0xa2
DEFINE_GUIDSTRUCT("95037F6F-3AC7-4452-B6C4-45A9CE9292A2", ISDB_TERRESTRIAL_TV_NETWORK_TYPE); 
#define ISDB_TERRESTRIAL_TV_NETWORK_TYPE DEFINE_GUIDNAMED(ISDB_TERRESTRIAL_TV_NETWORK_TYPE) 

#define STATIC_ISDB_T_NETWORK_TYPE \
    0xfc3855a6, 0xc901, 0x4f2e, 0xab, 0xa8, 0x90, 0x81, 0x5a, 0xfc, 0x6c, 0x83
DEFINE_GUIDSTRUCT("fc3855a6-c901-4f2e-aba8-90815afc6c83", ISDB_T_NETWORK_TYPE);
#define ISDB_T_NETWORK_TYPE   DEFINE_GUIDNAMED(ISDB_T_NETWORK_TYPE)
~~~
There is one value for DVB-T and two values for ISDB-T (why two?). All three values are different.

And for the MyGica S270, the only returned value is `CLSID_DVBTNetworkProvider`. Maybe we could force some usage for ISDB-T tuning spaces. But, at some point, TSDuck needs to know which type of tuner it is. If DirectShow defines distinct values for DVB-T and ISDB-T and nonetheless an ISDB-T tuner identifies itself as supporting DVB-T only, we cannot do much.

Note: On Linux, this tuner is correctly identified as ISDB-T only, right after one single ioctl(), without running thousands of lines of code, building complex graphs and praying the magic bat at sunset...

SS-FF, 2020-12-09:
You know I haven't gone through this myself, but I think I got a few leads to the right solution for this.
I think part of the solution for this is in the code that you had dropped a while ago (which was right, though).

But let's start somewhere else. Look at the following enum:

[image]

What this tells us, it that ISDB is handed just like another variant of DVB (unlike ATSC for example).

And then, do you remember the following code?

[image]

One step earlier, we have been enumerating the system-defined (existing in the registry) tuning spaces and tried one after another whether it's compatible (via `tuner->put_TuningSpace`).

Then we had changed that (as I had advised at that time) to constructing our own tuning spaces (`GetTuningSpaceFromNetworkType`). 

In the code above, when `tuner->put_TuningSpace` succeeds, we were querying for the `IID_IDVBTuningSpace` interface.
And the `IDVBTuningSpace` interface has a method `get_SystemType`. And what does that method return?

=> A value from the enum at the top of this post - which can also be ISDB-X.

### Conclusion

My guess would be that ISDB-T is implemented using

- The `DVBTuningSpace`
- A `DVBTLocator`
- Where
  - DVBTuningSpace::SystemType== ISDB_Terrestrial
  - DVBTuningSpace::_NetworkType == STATIC_ISDB_TERRESTRIAL_TV_NETWORK_TYPE 
    (or the other one, I really don't know why there are two of them)

To test what kind of tuner we have, we will probably need to construct such tuning spaces and use `tuner->put_TuningSpace` to check which one the tuner will accept.

--- 

As a kind of proof why I'm quite confident that it will work like that, let's go back to the screenshot from GraphStudioNext that I have posted above: 

The first property page is showing various tuning spaces (precisely, they are all `DVBTuningSpace` inheritors), including an ISDB-T tuning space.

But when you go to the second property page, there is just one or two tuning spaces remaining 

**=> those that are compatible with the tuner** 

And that's the mechanism you're looking for.

(now that I'm done, I start to remember that I've written almost the same text a while before...)

nicst35, 2020-12-13:
Hi, given that the most basic issues with ISDB-Tb handling were solved. Can we mark this issue as closed?

lelegard, 2020-12-14:
> Hi, given that the most basic issues with ISDB-Tb handling were solved. Can we mark this issue as closed?

Yes, I think so.

closed 2020-12-14 by nicst35

SS-FF, 2020-12-14:
@lelegard - Have you been able to try whether my suggested procedure works?

lelegard, 2020-12-15:
>  Have you been able to try whether my suggested procedure works?

Honestly, no, I didn't. First, I have no ISDB network to test and anything I could see on a system won't be tested up to the packet reception. So, there will still be a risk that all the work is incomplete.

I must say that I am positively fed up with DirectShow programming.

I just checked the source code:
- On Linux, the overall code to handle tuners is 2500 loc, including comments, and verbose code to set/get parameters. Apart from enhancements for new modulations or new forms of tuner identification, the code was written once and worked.
- On Windows, the same code uses 6700 loc. Most of it was developed the hard way, without precise documentation, using extremely long trial and error cycles. A large part of the code was written more than once, especially when DirectShow-experienced people like you provided advices which were not otherwise available in the documentation. Each time a new type of tuner appears, this is not just a matter of a bunch of new parameters to add. We must reconsider again and again new approaches, new hypotheses, new tests, maybe, let's try, etc. And the result is still less reliable, with less performance than on Linux.

So, I give up with DirectShow. The code is good enough to capture a stream from a Windows laptop when you move to some regional location. But for stable reception operations, I have been recommended Linux for 15 years now and every new issue just confirms this position again and again.

I would be happy to merge PR's which prove to improve the tuner implementation on Windows. But, as far as I am concerned, I am simply disgusted by DirectShow once more. TSDuck is a free and open source project which is only powered by enthusiasm and DirectShow clearly kills any form of enthusiasm. So, I give up, full stop.

SS-FF, 2020-12-15:
>  But, as far as I am concerned, I am simply disgusted by DirectShow once more.

I totally understand that and I acknowledge that you still got that far with tsDuck.

> > Have you been able to try whether my suggested procedure works?
> 
> Honestly, no, I didn't. First, I have no ISDB network to test and anything I could see on a system won't be tested up to the packet reception. So, there will still be a risk that all the work is incomplete.

I thought you'd be working on fixing the detection of the user's MyGica tuner, that's why I had responded to your question in that comprehensive way.

(also hoping it would confirm my theory which I had posted here about a year ago ;-)