tsduck/tsduck #668: [FEAT] PES tools

lars18th, 2020-11-25:
#### Is your feature request related to a problem?
At time the number of PES tools is very limited. In fact we can only extract the PES content, and do some checks based on the PES content (GOP start, for example).

The implementation of some more PES tools will be desirable. Good examples are some tools from the venerable and dead Opencaster (https://github.com/nexgenta/opencaster).

#### Proposed solution:
For example, I suggest to implement almost these tools:

- PES inject: A tool to take a video PES and create a TS. And with the option to include audio, data and raw PES streams.
- PES clock: This tool is to change the PTS/DTS timestamps of a PES stream based on the content of another.

#### Alternatives:
See as examples of the above tools the `pesvideo2ts`, `pesdata2ts` and `pesclock` in the Opencaster package.

#### Additional information:
It's true that compiling the Opencaster package it's possible to obtain some of the functionality. But this is a dead project, and I prefer to use the TSDuck package to do the entire processing. Futhermore, the FFmpeg tool can manipulate the PES streams at a more low-level, but when you do only simple tasks the overhead created is huge.

I hope you want to implement some of these tools and include them inside the TSDuck.
Thank you!

lars18th, 2020-11-27:
Hi @lelegard ,

First, I'm sorry if this request doesn't sound right to you. But, please, can you comment, at some point, on what you think about it?

Thank you.

lars18th, 2020-12-03:
Hi @lelegard ,

As I don't receive any feedback from you, I'll try to describe the use case for that I need these enhancements. I hope that you can understand the reason to insist on this request:

- For a service for impaired persons I want to add an additional "audio description stream". In a regular way this will require to demux the service, add the new track, remux it, and then reinsert it inside the MPTS. However, this requires professional and reliable equipment to do it without troubles. Then I opted to try this in a different way: use the free space inside the MPTS (that's very easy to do it using TSDuck) to include the PES data of the new track (as the track has a very low bitrate). After that only two actions are necessary: update the PMT table (easy again with TSDuck) and synchronize the PTS/DTS timestamps of the new track (that's can be done calculating one time the offset of this track with the video stream). The advantage of this is based in the basis that the MPTS muxing is done previously with professional equipment, and this "modification" can be completed without touching the structure of the MPTS. So, if the process fails, the TS stream can continue be played without the only exception of the missing audio track (I don't comment here the problem caused by the old PMT of the service). Then using comodity hardware is possible to do this processing if we can garantee a switch-back to the original TS in case of fail.

What you think about this? I'm aware about the difficults of this approach. But I want to crate a mockup of this, and I need some PES processing not included at time in TSDuck.

Please express your point of view.
Thank you.

lelegard, 2020-12-08:
> PES inject: A tool to take a video PES and create a TS. And with the option to include audio, data and raw PES streams.

Isn't is something that ffmpeg does quite naturally?

lars18th, 2020-12-08:
Hi @lelegard 

> > PES inject: A tool to take a video PES and create a TS. And with the option to include audio, data and raw PES streams.
> 
> Isn't is something that ffmpeg does quite naturally?

Yes, but... At time, it's quite complex to leave all of this to FFmpeg. I have two reasons for this:
- First, is that FFmepg can fail at some point, so I want to overcome it... if I can. And using TSDuck with FFmpeg, the _entire_ service needs to be processed. So if the FFmpeg process fails I don't have any (easy) failback solution. However, using TSDuck alone, and _not touching_ the original service (except for the PMT table), if the process fails, then I can continue using the original TS.
- And the second reason is that: Based on the exemple 5.2.28 in the documentation, it's possible to encapsulate additional data for FFmpeg processing. But, the example only uses _non timed data_. Yes the EIT pid doesn't use timestamps, and then you can use the A/V timestamps _created_ by the FFmpeg process without problems. However, this is not the case with subtitles (that if they are TXT subtitles is more complex than with DVB subtitles), or other timestamped streams. The main problem is that FFmpeg requires to regenerate the PTS/DTS and PCR values. If not, and if you require to use the `--copyts` option with FFmpeg then you will be in troubles... I doesn't work as expected and is prone to errors. So, when you add the FFmpeg to the equation then you'll get new clock marks _after_ the FFmpeg processing. In this case, then is not possible to do a simple decapsulation of the data, as you can't resynchronize the encapsulated streams with the new timestamps.

I'm sure you can see the complexity of the problem. However, with some minimal tools to manipulate PES data in TSDuck is feasible to obtain the same results using only TSDuck. For example, you can enconde the additional audio with any encoder (and with FFmpeg if you want), and then you can "merge" it with the original TS if you provide a tool for PES encapsulation with time synchronization. So the last processing required will be only to adjust the PMT table. And that's all.

Please let me know what you think about this approach.
Thank you!

lars18th, 2020-12-14:
Hi @lelegard ,

> Please let me know what you think about this approach.

Please, can you write about your point of view?
Thank you for your effort with this amazing project!

premultiply, 2021-01-07:
I am looking for a quick solution to remove (wrong) PTS values from PES headers at all.

SonnyWalkman, 2021-09-27:
I have some certainly what @lars18th has commented has merit.

https://github.com/tsduck/tsduck/issues/668#:~:text=For%20example%2C%20I,content%20of%20another.

Ffmpeg is a great tool and just works for most of everyone's needs. Us DVB aficionado's and DATV enthusiasts shall find ffmpeg's tsmuxer is no yet but close to being fully DVB compliant.  Correct me if I'm wrong.

Furthermore, with Opencaster now dead, TSduck is our only hope obi one kenobi.

Legegard, 
Can we reconsider developing a tool to realigning problematic PTS/DTS timestamp issues?

I know you have indicated many times TSDuck is NOT a muxer!  although it does most of the combining and back end tasks and many tools to analyse, filter and substitute.

Seems by fortune TSduck is heading to be come a TSmuxer.?? Opencaster's tscbrmux was the only thing close however, as indicated by many.. why develop a project when you have to cobble gstreamer, ffmpeg, TSduck and opencaster to get a final goal?

There seems to be no broadcast grade opensource tsmuxer available and I feel TSDuck is our only closest fit.
Most TV broadcasters remained devoted to $K hardware muxers to do the muscle work. Maybe TSDuck could be the next best thing?

Sonny..

moving-digital, 2023-08-31:
Hello @lelegard 

I have indeed the same need here: I got a TS file that has contains one video and one audio PID.
The video PID is alright in term of PTS/DTS, but the audio one is not.

In order to fix the audio PID, I would like to be able to edit the PTS/DTS in order to copy the video PID's PTS/DTS values, but unfortunately, neither the `pcradjust` nor `pcrcopy` nor `pcredit` plugin allow to do so.

Would it be possible to make one of the above plugin evolve for the same?

lelegard, 2023-08-31:
Hi @moving-digital 

> In order to fix the audio PID, I would like to be able to edit the PTS/DTS in order to copy the video PID's PTS/DTS values

Just to be sure, you would like to copy the video PTS/DTS into the audio PTS/DTS, is that right?

Assuming yes, this is something which is impossible by design because it would give awful results. This is the job of the encoder to set the audio and video PTS so that the decoder can resynchronize them. In the transmission, audio and video are always slightly desynchronized regarding their placement in the stream. The PTS information is precisely here to inform the decoder of the time for presenting the respective audio and video frames.

What would be the "copy" operation you'd like? Follow the evolution of the PTS and DTS video clocks in the stream and interpolate the corresponding PTS/DTS when an audio PES packet arrives? This will almost never be the right value for the audio PTS/DTS and you will experience awful audio desynchronization in the receiver.

Also remember that video PTS are out of order, especially for B frames, while audio PTS are always (I think) in order.

If I am wrong in my interpretation of your request, please explain what you have in mind, how and why it would give the correct rendering result.