tsduck/tsduck #669: [Q] handleTable() never executed in a TableHandlerInterface children class

bertothunder, 2020-11-27:
Hi,

  I've found a weird case where I'm developing a specific, small plugin that would keep a record of program IDs, numbers and some other information, kinda analyzer but quite a shorter one.

 Thing is, I've created this plugin as a child of ProcessorPlugin parent class, and also TableHandlerInterface, with a third parent class that allows data to be written to CSV-style text files, and to keep them rotating etc. It's not the first plugin I write this way, but the first one that would implement TableHandlerInterface.

 According to the source code and the wonderful documentation, a TableHandlerInterface-implementation class must override handleTable(ts::SectionDemux&, const ts::BinaryTable&) virtual method, which I did:

```
class MyPlugin: public ts::ProcessorPlugin, private TableHandlerInterface, public FileRotation
{
     private:
           <some properties>
           virtual void handleTable(ts::SectionDemux&, const ts::BinaryTable&) override;
     public:
         MyPlugin(ts::TSP* tsp);
         <blah>
}
```

I did implement handleTable() following along the existing code I could see from other plugins, which is nothing else that a switch....case implementation. However, it seems this method is NEVER called, a simple std::cout or tsp->info() calls is never displayed, and debugging in some way confirms it.

 Not really sure what I could be doing wrong, I've checked several plugins and they are doing nothing different, anything that you @lelegard might think could be causing it?

 Regards

lelegard, 2020-11-27:
Did you register your handler into a `SectionDemux`?

A `TableHandlerInterface` is just an interface (in the Java sense). It won't be magically called if there is no demux to call it. And the demux must be fed with packets.

Let's have a look at the `svrename` plugin (it could any other of the same kind), just keeping the interesting lines :
~~~
    class SVRenamePlugin: public ProcessorPlugin, private TableHandlerInterface
    {

    private:
        SectionDemux      _demux;        <======= here

        virtual void handleTable(SectionDemux&, const BinaryTable&) override;
~~~

Then, in the constructor, the demux is initialized with the `TableHandlerInterface` :
~~~
ts::SVRenamePlugin::SVRenamePlugin(TSP* tsp_) :
    ProcessorPlugin(tsp_, u"Rename a service, assign a new service name and/or new service id", u"[options] [service]"),
    _demux(duck, this),    <======= here
{
~~~
There, `this` is the instance of the plugin, which is a subclass of `TableHandlerInterface`. Now, the demux knows which handler to call on new tables.

Based on the context, more and more PID's are added to the demux:
~~~
    _demux.reset();
    _demux.addPID(_old_service.hasName() ? PID_SDT : PID_PAT);  <=== on start()

        _demux.addPID(PID_PAT);  <===== when the service name is found in SDT

        _demux.addPID(it->second);  <====== when the service id is found in the PAT, add the PMT PID
~~~

Finally, in the packet processing method:
~~~
ts::ProcessorPlugin::Status ts::SVRenamePlugin::processPacket(TSPacket& pkt, TSPacketMetadata& pkt_data)
{
    _demux.feedPacket(pkt);    <======= here
~~~

So, the demux receives all TS packets of the TS. It only considers packets from PID's it is supposed to work on. And when a new complete table is complete, it calls the registered `TableHandlerInterface`. But if you don't do that, no-one will call your handler.

In a different programming style, this is exactly the scenario of what a set-top box or TV set is doing when you switch to a new TV channel.

bertothunder, 2020-11-28:
@lelegard no, I didn't. I noticed the SectionDemux objects in some of the plugins, but could not associate them in such a way. Maybe this should be noted in the programmer's guide for the `TableHandlerInterface`?

bertothunder, 2020-11-30:
Thierry, you were absolutely right, it's working now as expected. I'll close the question, much appreciated.

closed 2020-11-30 by bertothunder