tsduck/tsduck #670: [BUG] spectral inversion -13298 is not supported by Windows BDA/DirectShow

oviano, 2020-11-28:
Hello there

I just tried running my code through the TSDuck SDK for the first time in a while; I did a new DVB-T scan, UK transmitter, and the XML contains entries like this:

<network id="0x3033" type="DVB-T">
    <ts id="0x1044" onid="0x233A">
      <dvbt frequency="490,000,000" modulation="64-QAM" transmission="8K" guard="1/32" inversion="-13298"/>
      <service id="0x1044" name="BBC ONE Lon" PMTPID="0x0064" type="0x01" cas="false"/>
 
etc...

All looks similar to when I last tried, except now when I reach my m_tuner->tune(parameters, report) line of my code, it produces the above error.

So the scan seems to have found a value of spectral inversion that won't actually tune.

I'm not sure if that is a problem with the scan, or something else, but I can see that looking at my old scan from an older version of TSDuck, it doesn't include those XML fields for modulation, transmission, guard or inversion.

I guess I will try and see what happens if I remove those extra fields from the XML or something...

lelegard, 2020-11-28:
I see from your message that you are on Windows. DirectShow and BDA drivers are an endless source of surprises and miseries... 

During scan, for each frequency with a signal, `tsscan` queries the tuner driver for the current tuning parameters. It seems that the Windows driver of your tuner returns this weird value (instead of the expected on/off/auto). It looks like an issue with this driver. I just tried to scan my local DTTV network on Windows using a DVB-T Avermedia tuner and I do not get invalid values for spectral inversion. Did you previously manage to scan with the same tuner on Windows? 

Spectral inversion is an optional and rarely used option. It is possible that this driver does not correctly report it. You can safely remove the inversion attributes from the XML file.

As a precaution, I will filter invalid inversion values returned from the drivers and reset this property to "unset" if invalid. It will be a protection against invalid drivers.

oviano, 2020-11-28:
Hi Thierry - yes it’s the same tuner I’ve used ever since I first posted on here (you may remember I helped a little bit with ATSC?) anyway last time I scanned I assume the code wasn’t setup to add those extra fields, including inversion but everything worked.

Actually if I add a line to my code to reset that field to AUTO if it’s not OFF, ON or AUTO I am successfully able to tune. So yes perhaps duck can do that itself. Maybe that’s an approach you could take for all those parameters on Windows, just in case?

It’s a Hauppauge Win SOLOHD, I previously reported it working in your thread devoted for this.

lelegard, 2020-11-28:
> (you may remember I helped a little bit with ATSC?)

Yes, I do !

> Actually if I add a line to my code to reset that field to AUTO if it’s not OFF, ON or AUTO I am successfully able to tune. So yes perhaps duck can do that itself. 

Good to know. I am going to do that in TSDuck too. It seems safe.

> Maybe that’s an approach you could take for all those parameters on Windows, just in case?

Maybe. But it could backfire when new  modulation parameters appear (FEC or guard interval values for instance). TSDuck implements a strategy where all enumeration values for modulation parameters are directly mapped to the system-defined values. The TSDuck-defined enumeration value `FEC_3_4`, for instance, does not have the same binary value on Windows and Linux, it has the same binary value as its DirectShow or LinuxTV counterparts.

This has two advantages:
- It avoids long, tedious and useless conversions.
- If new modulation values appear with a new version of the system, the new values, although not yet defined as proper enum values in TSDuck, still work. You can specify them as integers on the command line. The values which are reported as "current" by the driver can be reused later to tune the same way, etc. Of course, TSDuck will later define the additional enum values but you do not need to wait for it.

If I test all possible known values of a parameter at a given time and exclude others, I will lose the benefit of this native implementation.

Spectral inversion is different. First, it is binary, on/off/auto only so we can safely test those values. No new value will appear. And since this is a rather old, obscure and seldom used parameter, there is a higher risk that a driver does not implement it correctly. So, for now, I will limit the check to spectral inversion.

If you (or others) encounter similar issues with other parameters, I may reconsider this position.